zwave-js/zwave-js #3932: [question] Support for room thermostat Danfoss DRS11

stomko11, 2021-12-20:
I am wondering if it would be possible to somehow add support for Danfoss DRS11 room thermostat. I am able to pair it, but sits at unknown product 

  "hexId": "0x0002-0x8003-0x8001",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x0002:0x8003:0x8001:0.0",
  "productLabel": "Unknown product 0x8001",
  "productDescription": "0x8003",
  "manufacturer": "Danfoss",

It is not fully paired, because thermostat won't show temperature (that's done after it is paired successfully). Also, after adding it, it looks like it is woken up, but after some time goes to sleep, and neither of buttons will wake it up.

Full debug info below:

```
{
  "id": 90,
  "name": "",
  "loc": "",
  "values": [
    {
      "id": "90-114-0-manufacturerId",
      "nodeId": 90,
      "commandClass": 114,
      "commandClassName": "Manufacturer Specific",
      "endpoint": 0,
      "property": "manufacturerId",
      "propertyName": "manufacturerId",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Manufacturer ID",
      "stateless": false,
      "min": 0,
      "max": 65535,
      "list": false,
      "value": 2,
      "lastUpdate": 1639643507122,
      "newValue": 2
    },
    {
      "id": "90-114-0-productType",
      "nodeId": 90,
      "commandClass": 114,
      "commandClassName": "Manufacturer Specific",
      "endpoint": 0,
      "property": "productType",
      "propertyName": "productType",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Product type",
      "stateless": false,
      "min": 0,
      "max": 65535,
      "list": false,
      "value": 32771,
      "lastUpdate": 1639643507123,
      "newValue": 32771
    },
    {
      "id": "90-114-0-productId",
      "nodeId": 90,
      "commandClass": 114,
      "commandClassName": "Manufacturer Specific",
      "endpoint": 0,
      "property": "productId",
      "propertyName": "productId",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Product ID",
      "stateless": false,
      "min": 0,
      "max": 65535,
      "list": false,
      "value": 32769,
      "lastUpdate": 1639643507123,
      "newValue": 32769
    }
  ],
  "groups": [],
  "neighbors": [],
  "ready": true,
  "available": true,
  "hassDevices": {},
  "failed": false,
  "inited": true,
  "hexId": "0x0002-0x8003-0x8001",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x0002:0x8003:0x8001:0.0",
  "productLabel": "Unknown product 0x8001",
  "productDescription": "0x8003",
  "manufacturer": "Danfoss",
  "endpointsCount": 0,
  "endpointIndizes": [],
  "isSecure": false,
  "isControllerNode": false,
  "keepAwake": false,
  "deviceClass": {
    "basic": 4,
    "generic": 255,
    "specific": 0
  },
  "deviceId": "2-32769-32771",
  "status": "Asleep",
  "interviewStage": "ProtocolInfo",
  "_name": "NodeID_90",
  "statistics": {
    "commandsTX": 1,
    "commandsRX": 1,
    "commandsDroppedRX": 0,
    "commandsDroppedTX": 0,
    "timeoutResponse": 0
  },
  "manufacturerId": 2,
  "productId": 32769,
  "productType": 32771,
  "protocolVersion": 1,
  "nodeType": 1,
  "security": "None",
  "supportsSecurity": false,
  "supportsBeaming": false,
  "isListening": false,
  "isFrequentListening": false,
  "isRouting": true,
  "maxDataRate": 40000,
  "lastActive": 1639643507123,
  "isBatteryPowered": false
}
```

AlCalzone, 2021-12-20:
> interviewStage": "Protocol Info"

You have to wake it up. The interview isn't finished.

stomko11, 2021-12-20:
It actually did not work like that.
After factory resetting the sensor, I paired it. ZwaveJS says node was added, and when I look it up, it is all green (means protocol info and node info already completed), device is added, but the only thing it shows is Manufacturer (Danfoss). Rest is unknown. After a while, sleep icon goes to "zzz" indicating device went to sleep.
However it is not paired, because thermostat itself shows temperature on display only when it is paired successfully. So I hit "reinterview node" and once I do that, it never passes "protocol info" again. Neither of buttons will wake it up. It just shows "zzz" and "protocol info" spins forever.

AlCalzone, 2021-12-20:
> Neither of buttons will wake it up

If the wakeup procedure isn't described in the device manual, try these:
- re-insert battery
- triple click the buttons instead of single-clicking
- use a fresh battery

If nothing helps, try excluding and including it again, this time make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly` and attach it here as a file.

stomko11, 2021-12-20:
Log is attached. It is Node 91.
After I factory reset it, I tried to add it. Here is how it looks right after addition, all green like it has already been interviewed, and is awake (that itself is fishy, because other battery operated devices usually take some time to do protocol and node info after addition). 

[image]

It sits like that until I do re-interview, then it shows as asleep.

[image]

I did try new batteries, no change. Single click, triple click, neither of buttons make any difference. I ended up removing it, factory resetting it and I can pair it back to danfoss on first attempt and it works.
I also do not see DRS11 listed in database, only DRS21.
[zwavejs_2021-12-20.log](https://github.com/zwave-js/node-zwave-js/files/7745395/zwavejs_2021-12-20.log)

PR #3947 (2021-12-22): feat(config): support adding comments to devices
With this PR, config files can have one or more comments in the metadata. These have a text and 3 different levels:
- `info`: general information, like hints for configuration
- `warning`: Should be displayed with a ⚠ to warn the user of potential issues
- `error`: Meant to warn the user of unusable devices or other severe issues. Should be shown in red with an appropriate icon, e.g. ❌.

closes: #3932
closed 2021-12-22 by AlCalzone