zwave-js/zwave-js #3933: [question] BE469ZP undefined values, refresh doesn't seem to do anything.

rct, 2021-12-20:
Interviewing 24 out of 25 devices other than 1 out of 2 Schlage (Allegion) BE469ZP deadbolts completed successfully. Since one of the BE460ZPs seems more reliable than the other, I'm assuming it's somewhat communications related. Statistics shows a number of `timeout response`. 

[image]

The deadbolt I'm having problems with shows many undefined values, including version.  Not seeing a version number on the dashboard was my first clue that there was a problem with that node.

[image]

A few values for configuration seem to fill in then there are a number of undefineds:

[image]

Moving the controller closer to the lock didn't resolve the problem. However, trying to understand the logs makes it look like there might be other issues going on.  Namely if I try to use refresh, with Zwave logging turned up to debug, it doesn't appear to **actually try to do any communication**. but I realize I really don't understand much from the logs yet. see logs below.   

The numbers under statistics, including `Commands TX` do not change when a refresh is requested.  I don't know if this is a bug, or not.

A few things to note: 

- The logs say doing a ping is successful, but also there is a line that says a route from another node failed.  Note: doing a ping increases `Commands TX`  but `Commands RX` and `Timeout Response` don't actually increase. 
- The lock I'm having problems with has been working with the old Zwave integration for about 2 years. Sending lock/unlock commands generally worked. Notification of lock/unlock events generally worked.
- Unsolicited notifications from the lock seem to be received just fine. 

Here is the log from doing a ping and clicking a refresh on the version CC values:

```
2021-12-17 11:00:24.648 INFO ZWAVE: Calling api pingNode with args: [ 7, [length]: 1 ]
2021-12-17T16:00:24.649Z CNTRLR » [Node 007] pinging the node...
2021-12-17T16:00:24.662Z SERIAL » 0x0108001307010025aa6d                                              (10 bytes)
2021-12-17T16:00:24.663Z DRIVER » [Node 007] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      170
                                  └─[NoOperationCC]
2021-12-17T16:00:24.667Z SERIAL « [ACK]                                                                   (0x06)
2021-12-17T16:00:24.672Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-12-17T16:00:24.673Z SERIAL » [ACK]                                                                   (0x06)
2021-12-17T16:00:24.674Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-12-17T16:00:26.069Z SERIAL « 0x01180013aa00008c01c4b87f7f7f0101031300000002021307d4              (26 bytes)
2021-12-17T16:00:26.073Z SERIAL » [ACK]                                                                   (0x06)
2021-12-17T16:00:26.076Z DRIVER « [REQ] [SendData]
                                    callback id:            170
                                    transmit status:        OK, took 1400 ms
                                    repeater node IDs:      19
                                    routing attempts:       2
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -60 dBm
                                    ACK RSSI on repeaters:  -72 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
                                    route failed here:      19 -> 7
2021-12-17T16:00:26.103Z CNTRLR « [Node 007] ping successful
2021-12-17 11:00:26.104 INFO ZWAVE: Success zwave api call pingNode true
2021-12-17 11:03:50.341 INFO ZWAVE: Calling api refreshCCValues with args: [ 7, 134, [length]: 2 ]
2021-12-17 11:03:50.344 INFO ZWAVE: Success zwave api call refreshCCValues undefined
```

Nothing else shows up after that even if I wait for a number of minutes.

Versions:

- Zwavejs2MQTT 6.1.1.
- Zwavejs 8.9.0-beta.1
- Installation is docker container (via Home Assistant add-on)

Debug info from the lock:

```
{
  "id": 7,
  "name": "",
[... 2631 lines cut ...]
  "maxDataRate": 100000,
  "deviceClass": {
    "basic": 4,
    "generic": 64,
    "specific": 3
  },
  "deviceId": "59-1129-1",
  "status": "Alive",
  "interviewStage": "Complete",
  "statistics": {
    "commandsTX": 120,
    "commandsRX": 80,
    "commandsDroppedRX": 0,
    "commandsDroppedTX": 0,
    "timeoutResponse": 34
  },
  "lastActive": 1639707989695,
  "minBatteryLevel": 35,
  "batteryLevels": [
    35
  ],
  "isBatteryPowered": true,
  "_name": "NodeID_7"
}
```

AlCalzone, 2021-12-27:
The log is a bit too short to say exactly, but this looks like you might be having RF connectivity issues.

```
                                    transmit status:        OK, took 1430 ms
                                    repeater node IDs:      19
                                    routing attempts:       2
```
might be indicative of that (or it might just look that way because the lock is a FLiRS device). I suggest you take a look at https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues

scyto, 2022-01-04:
looks like the sort of timeout issues i was getting a year ago on the same locks

placing the zwave dongle in the center of the house, using a 1m usb extension cable, making sure my locks were a direct connection to the controller (not multi hop) and changing that RSSI parameter on the key solved my issues.  my thesis to this say is the lock could hear the zwave controller but was having issues in opposite direction.  It has (cross fingers been stable for nealy a year now)

closed 2022-01-22 by zwave-js-assistant[bot]