zwave-js/zwave-js #3934: [bug] Failed to send the command after 3 attempts (Status Fail) (ZW0202)

Snapperud, 2021-12-20:
Hi
I have a problem for a few months, running Z-Wave JS to MQTT 0.32.1 and HA 2021.12.3.
Zwave Controller Aeotec Z-Stick 7 (FW 7.15) Have tested with extension cable and both USB-2 and USB-3.
The only solution is to restart Z-Wave JS to MQTT Add-on, or resart the docker container(zwavejs/zwavejs2mqtt) , then everything is ok, to the next crash. it happens 2-4 times a day. 

Also have tested with a separate docker container outside HA, same problem. (zwavejs/zwavejs2mqtt:latest)
Tested a new Zwave Controller Aeotec Z-Stick 7 (FW 7.15) still the same problem. There are different nodes that fails, from time to time.
Installed an Aeotec Range Extender 7 and run a "Heal network", but it did not get any better

Multiple logs in the zip file.

2021-12-16 12:20:14.348 INFO ZWAVE: Node 7: value updated: 49-0-Power 2.2 => 2.2
2021-12-16 12:20:15.475 INFO ZWAVE: Node 7 is now dead
2021-12-16 12:20:15.585 ERROR ZWAVE-SERVER: Failed to send the command after 3 attempts (Status Fail) (ZW0202)
ZWaveError: Failed to send the command after 3 attempts (Status Fail) (ZW0202)
at Driver.sendMessage (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:3345:23)
at Driver.sendCommand (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:3504:28)
at BinarySwitchCCAPI.set (/opt/node_modules/zwave-js/src/lib/commandclass/BinarySwitchCC.ts:114:21)
at Proxy.BinarySwitchCCAPI. (/opt/node_modules/zwave-js/src/lib/commandclass/BinarySwitchCC.ts:129:14)
at ZWaveNode.setValue (/opt/node_modules/zwave-js/src/lib/node/Node.ts:849:14)
at Function.handle (/opt/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:20:44)
at Object.node (/opt/node_modules/@zwave-js/server/dist/lib/server.js:39:91)
at Client.receiveMessage (/opt/node_modules/@zwave-js/server/dist/lib/server.js:95:99)
at WebSocket. (/opt/node_modules/@zwave-js/server/dist/lib/server.js:48:45)
at WebSocket.emit (node:events:390:28)
2021-12-16 12:20:15.625 INFO ZWAVE: Node 7: metadata updated: 49-0-Power
2021-12-16 12:20:15.626 INFO ZWAVE: Node 7: value updated: 49-0-Power 2.2 => 2.2
2021-12-16 12:20:15.627 INFO ZWAVE: Node 7: has returned from the dead

[(ZW0202).zip](https://github.com/zwave-js/zwavejs2mqtt/files/7744366/ZW0202.zip)

jmgiaever, 2021-12-21:
Try to upgrade to the latest, using the master tag. Remember to include driver logs with the debug level.

Snapperud, 2021-12-21:
Upgraded to image: zwavejs / zwavejs2mqtt: master
Here are the logs.
[Failed to send the command after 3 attempts (Status Fail) (ZW0202).zip](https://github.com/zwave-js/node-zwave-js/files/7755211/Failed.to.send.the.command.after.3.attempts.Status.Fail.ZW0202.zip)

AlCalzone, 2021-12-21:
Looks like yet another instance of https://github.com/zwave-js/node-zwave-js/issues/3906

A lot of repeated messages are received in a short timeframe, then the controller fails to send. (Maybe the repeated messages are already a symptom).

Snapperud, 2021-12-21:
To provoke the error, I run off & on some devices. 
It is the same error that occurs in the middle of the night, when there is less traffic.
Even if there is a lot of traffic, the program should not crash, it should just go on,when this happens I have to restart zwave-js.

Snapperud, 2021-12-21:
Good that they are working on a new FW that will solve the problem, mabye a FW as Christmas present :-)
Thanks for the info.

AlCalzone, 2021-12-21:
Don't get your hopes up. I'm in contact with them, but they'll have to reproduce this first so they can investigate.

fisch55, 2021-12-27:
got the same Problem with FW 7.15. 
I only know this FW. Does ist works with older FW? Maybe downgrade possible?

AlCalzone, 2021-12-27:
@fisch55 if you have a spare 500 series stick lying around, you can try this:
https://github.com/zwave-js/node-zwave-js/issues/3906#issuecomment-997484466

Otherwise, no. There is no 700 series FW that doesn't have these issues.

fisch55, 2021-12-27:
THX for reply.
No, only stick 7. 

But great that you are in contact with them. I hope that they find something.

AlCalzone, 2021-12-27:
Closing in favor of the pinned issue. Please follow that one for updates.

closed 2021-12-27 by AlCalzone

geonoemi, 2023-09-11:
I still have this issue on ZMEE-UZB1 stick (500 series controller) with zwave version 8.11.7. Restarting the driver solves it, but until that, neither of my zwave products are working.

AlCalzone, 2023-09-11:
You may want to update. There were almost a hundred releases since that version.