zwave-js/zwave-js #3935: [feat] Support for Ultraloq zwave locks

mbbush, 2021-12-20:
**Is your feature request related to a problem? Please describe.**
When I connect my ultraloq u-bolt pro zwave lock to zwavejs, it shows up as an unknown device from an unknown manufacturer. The interview process doesn't result in a working device, probably because this is the first zwave device this manufacturer has made and it doesn't seem to completely follow the spec.

https://products.z-wavealliance.org/regions/2/categories/6/products?company=810

**Describe the solution you'd like**
A working driver. I'd be happy to write some/all of it, although I won't have time to work on it until January at the earliest.

**Describe alternatives you've considered**

**Additional context**

Manufacturer specific report and device specific report received from the u-bolt pro zwave: 
72 05 04 52 00 04 00 01
72 07 01 28 0C 43 14 FF FE 53 4C E6

I documented a few of the quirks of this device here https://community.hubitat.com/t/ultraloq-u-bolt-zwave-700-lock/72810/26

AlCalzone, 2021-12-20:
In general we don't have device specific drivers. Z-Wave JS tries to stay as close to the specifications as possible. There are some quirks we work around but not to the extent other drivers do.

You mentioned that the device isn't working properly. 
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly` and attach it here as a file.

While making that log, re-interview the device and let it finish, then operate the device in a way that outlines what you mean by "Not working".

mbbush, 2022-01-03:
I'm back from vacation, and starting to look at this again. I watched your bit in the [State of the Open Home Video](https://www.youtube.com/watch?v=6ZMXE5PXPqU&t=513s) and liked what you said about your philosophy of sticking to the zwave spec as a means of trying to encourage manufacturers to "do it the right way". I don't know if you or I will be able to get any positive engagement from ultraloq, but I'm willing to give it a try.

Some of my previous posts in this thread contained incorrect information about supported zwave command class versions, which I've edited to correct. Next I'll start working on creating a `silly` driver log. 

I'm still just getting started with zwavejs. Just so I understand a little more about how zwavejs works, is the idea that you simultaneously attempt to have a complete library of device files, and also software that can "figure out" what is supposed to happen via an interview when there is no such device file? I'd like to be able to contribute constructively to this community project, and honestly I rank that as a higher priority than getting my own lock working quickly. So maybe the best path forward here is figure out if there are any changes that are worth making to improve the "automatic" support for this device, and then second I can work on writing a device file?

BTW: I'm not sure if having a log level named `silly` is a normal javascript thing or something you made up (the far more boring/sober name I'm accustomed to from the enterprise java world is `TRACE`), but I like it. It reminds me of android's [`log.wtf`](https://developer.android.com/reference/android/util/Log#wtf(java.lang.String,%20java.lang.String)) method, although obviously at the opposite end of the spectrum.

mbbush, 2022-01-03:
Ok, I think I know why this isn't working at all for me.

I started by pairing the device to a zooz 700 series usb stick controller, using the silicon labs SimplicityStudio software. I then unplugged the usb stick and moved it to the computer running zwaveJS. I copied the zwave security keys from SimplicityStudio into zwaveJS, but it seems to be running into problems related to security.

When I look at my zwavejs control panel, I see the node id, but it shows that it is not included securely. 
[image]

When I do things on the physical lock that cause it to send (secure) zwave messages, I get logs like this:
```
zwavejs2mqtt    | 12:14:26.301 SERIAL « 0x011e00a8000102159f032500a1363c6689a2c61d4b5e7ddccd81284a5000adbc  (32 bytes)
zwavejs2mqtt    | 12:14:26.310 DRIVER   Dropping message with invalid payload
zwavejs2mqtt    | 12:14:26.312 DRIVER « [Node 002] [REQ] [BridgeApplicationCommand]
zwavejs2mqtt    |                       │ RSSI: -83 dBm
zwavejs2mqtt    |                       └─[Security2CCMessageEncapsulation] [INVALID]
zwavejs2mqtt    |                           error: No security class granted
zwavejs2mqtt    | 12:14:26.313 SERIAL » [ACK]    
```

AlCalzone, 2022-01-03:
> Just so I understand a little more about how zwavejs works, is the idea that you simultaneously attempt to have a complete library of device files, and also software that can "figure out" what is supposed to happen via an interview when there is no such device file?

Z-Wave works by standardizing behavior in the form of Command Classes (CCs), which also offer means to discover what specifics of that CC a device supports. Z-Wave JS tries to do this (and rely on it) as much as possible, as opposed to OpenZWave which defines a lot in the config files. We only define what we cannot query (or where the queried information is often subpar), i.e. Association groups and Config params. For some devices we work around common issues by enabling compat flags.

> I'm not sure if having a log level named silly is a normal javascript thing

These loglevels are aligned with how `npm` calls them: https://github.com/winstonjs/winston#logging-levels, which is also the default for the logging library we use.
It's quite fitting though, `debug` allows for debugging and `silly` is meant for silly amounts of logging output.

> No security class granted

That is likely because the security interview wasn't done yet. I'd need to see a more complete log of the interview to tell what's happening there. Normally the driver figures out which keys the node knows.

mbbush, 2022-01-03:
I found this in the earlier logs. 
```
2022-01-03T20:08:23.447Z CNTRLR   [Node 002] Beginning interview - last completed stage: None
2022-01-03T20:08:23.447Z CNTRLR   [Node 002] new node, doing a full interview...
2022-01-03T20:08:23.448Z CNTRLR » [Node 002] querying protocol info...
2022-01-03T20:08:23.453Z SERIAL » 0x0104004102b8                                                       (6 bytes)
2022-01-03T20:08:23.453Z DRIVER » [REQ] [GetNodeProtocolInfo]
                                    payload: 0x02
2022-01-03T20:08:23.458Z SERIAL « [ACK]                                                                   (0x06)
2022-01-03T20:08:23.459Z SERIAL « 0x010a014153dc01044001007e                                          (12 bytes)
2022-01-03T20:08:23.460Z SERIAL » [ACK]                                                                   (0x06)
2022-01-03T20:08:23.460Z DRIVER « [RES] [GetNodeProtocolInfo]
                                    payload: 0x53dc0104400100
2022-01-03T20:08:23.464Z CNTRLR « [Node 002] received response for protocol info:
                                  basic device class:    Routing Slave
                                  generic device class:  Entry Control
                                  specific device class: Door Lock
                                  node type:             Routing End Node
                                  is always listening:   false
                                  is frequent listening: 1000ms
                                  can route messages:    true
                                  supports security:     false
                                  supports beaming:      true
                                  maximum data rate:     100000 kbps
                                  protocol version:      3
2022-01-03T20:08:23.465Z CNTRLR   [Node 002] Interview stage completed: ProtocolInfo
2022-01-03T20:08:23.465Z CNTRLR » [Node 002] pinging the node...
2022-01-03T20:08:23.470Z SERIAL » 0x010d00a9010201002500000000027e                                    (15 bytes)
2022-01-03T20:08:23.470Z DRIVER » [Node 002] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      2
                                  └─[NoOperationCC]
2022-01-03T20:08:23.476Z SERIAL « [ACK]                                                                   (0x06)
2022-01-03T20:08:23.477Z SERIAL « 0x010401a90152                                                       (6 bytes)
2022-01-03T20:08:23.477Z SERIAL » [ACK]                                                                   (0x06)
2022-01-03T20:08:23.478Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2022-01-03T20:08:25.739Z SERIAL « 0x011d00a9020000de00ae7f7f7f7f01010100000000420200007f7f7f7f7f07    (31 bytes)
2022-01-03T20:08:25.741Z SERIAL » [ACK]                                                                   (0x06)
2022-01-03T20:08:25.743Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            2
                                    transmit status:        OK, took 2220 ms
                                    routing attempts:       2
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -82 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
                                    beam:                   1000 ms
2022-01-03T20:08:25.751Z CNTRLR   [Node 002] The node is now alive.
2022-01-03T20:08:25.751Z CNTRLR   [Node 002] Beginning interview - last completed stage: ProtocolInfo
2022-01-03T20:08:25.752Z CNTRLR » [Node 002] querying node info...
2022-01-03T20:08:25.757Z CNTRLR « [Node 002] ping successful
2022-01-03T20:08:25.757Z CNTRLR » [Node 002] querying node info...
2022-01-03T20:08:25.761Z SERIAL » 0x010400600299                                                       (6 bytes)
2022-01-03T20:08:25.761Z DRIVER » [Node 002] [REQ] [RequestNodeInfo]
                                    payload: 0x02
2022-01-03T20:08:25.766Z SERIAL « [ACK]                                                                   (0x06)
2022-01-03T20:08:25.767Z SERIAL « 0x01040160019b                                                       (6 bytes)
2022-01-03T20:08:25.768Z SERIAL » [ACK]                                                                   (0x06)
2022-01-03T20:08:25.768Z DRIVER « [RES] [RequestNodeInfo]
                                    payload: 0x01
2022-01-03T20:09:30.781Z CNTRLR   [Node 002] Interview attempt 2/5 failed, retrying in 10000 ms...
2022-01-03T20:09:30.784Z SERIAL » 0x010400600299                                                       (6 bytes)
2022-01-03T20:09:30.784Z DRIVER » [Node 002] [REQ] [RequestNodeInfo]
                                    payload: 0x02
2022-01-03T20:09:30.789Z SERIAL « [ACK]                                                                   (0x06)
2022-01-03T20:09:30.790Z SERIAL « 0x01040160019b                                                       (6 bytes)
2022-01-03T20:09:30.791Z SERIAL » [ACK]                                                                   (0x06)
[... 130 lines cut ...]
[... cut ...]
2022-01-03T20:10:05.336Z SERIAL « 0x010401a90152                                                       (6 bytes)
2022-01-03T20:10:05.336Z SERIAL » [ACK]                                                                   (0x06)
2022-01-03T20:10:05.337Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
```
It looks like the problem is that the device is (incorrectly) advertising that it doesn't support security. Here's what I see in the node information in the SI labs PC Controller software:
[image]

AlCalzone, 2022-01-03:
> supports security:     false

is misleading, disregard that.

Support for secure CCs is advertised:
```
2022-01-03T20:09:32.130Z CNTRLR « [Node 002] node info received
                                  supported CCs:
                                  · Z-Wave Plus Info
                                  · Transport Service
                                  · Security
                                  · Security 2
                                  · Supervision
                                  controlled CCs:
```
but the interview isn't done securely. Either the keys didn't get applied, or the wrong security info is already cached. If it is the latter, its probably easiest to remove and re-include the device.

mbbush, 2022-01-03:
I'd rather not have to exclude and re-include the lock, as I don't want it to reset the fingerprints I've already set up (re-enrolling mine is no big deal, but my other family members are less enthusiastic). Is there another way to tell what security info is cached, or reset it?

The only two nodes on the network are the controller and the door lock.

AlCalzone, 2022-01-04:
> or reset it?

If you're using zwavejs2mqtt, shut it down. In the store directory, you'll find a `<homeid>.json` file you can open with a text editor.
Find the section that looks like this for node 2:
[image]
delete the lines, and save.

Start z2m again and re-interview the device.

mbbush, 2022-01-05:
Removing the `"securityClasses"` element from the `<homeid>.json` file did the trick. It still shows as unknown manufacturer and unknown product, but it shows actual codes now, not 0x0000. This is expected, as there's no device file. 

I can successfully manage the User Code command class, at least subject to the limitations of the lock itself. It treats user status 1 (enabled) and 2 (disabled) as both enabled, but I can disable a code by setting it to status 0 (available). And it shows the fingerprint slots as populated with a specific code, with status available. 

So I guess all that's needed is to write a basic device file that has the product name and id? I think I'll also write to the manufacturer encouraging them to bring their zwave implementation up a notch, although I don't have high hopes on that.

mbbush, 2022-01-05:
The only other thing that seems less than ideal is that zwavejs seems to think that this device supports both the "Door Lock" (0x62) and "Lock" (0x76) command classes. I'm not sure why it thinks that the deprecated "Lock" class is supported. It isn't advertised as supported, and the ground truth is that the device ignores both get and set commands of this class. Unfortunately, "Lock" also shows up first in the z2m ui, above "Door Lock", which initially led me to believe that I couldn't control the lock.

AlCalzone, 2022-01-05:
That might be required by the device class. Unfortunately, there is no versioning for these kinds of things on the Z-Wave specs, so we might have to disable the Lock CC through a compat flag.

Can you post another log of a re-interview, so I can double check?

mbbush, 2022-01-05:
Sure. I'm redacting secrets from it and then will post. The only explanation I can think of is that zwavejs adds it based on the device class, (which is `"deviceClass": {"basic": 4,"generic": 64,"specific": 1}`) but I don't see anything in the zwave docs that indicates that the `Lock` command class must be supported for either `GENERIC_TYPE_ENTRY_CONTROL` or `SPECIFIC_TYPE_DOOR_LOCK`. Perhaps I'm looking in the wrong place.

FWIW, the Lock command class carries this boxed warning in the SDS13781 zwave documentation:

> **THIS COMMAND CLASS HAS BEEN DEPRECATED**
> 
> A device MAY implement this command class, but it is RECOMMENDED that new implementations
use the Door Lock Command Class.
If implementing this command class, it is RECOMMENDED that the Door Lock Command Class is also
implemented.

mbbush, 2022-01-05:
[log.redacted.txt](https://github.com/zwave-js/node-zwave-js/files/7816987/log.redacted.txt)

kpine, 2022-01-05:
Lock CC was required for Door Lock device types in Z-Wave (Classic). Z-Wave Plus does not re-define the Door Lock DT, so it's a carry-over from the Classic definition.

However, Z-Wave Plus V2 removes the requirement for Lock CC. From the logs this lock is also Plus V2.

AlCalzone, 2022-01-05:
> However, Z-Wave Plus V2 removes the requirement for Lock CC. From the logs this lock is also Plus V2.

Good to know, I'll have to make the device class requirements conditional it seems

Roundaround, 2022-01-26:
What's the latest status on this? No device config yet but it (mostly) works via interview?

AlCalzone, 2022-01-26:
> What's the latest status on this? No device config yet but it (mostly) works via interview?

There is a device config, but just for the labels - config parameters don't seem to exist. The lock should work normally, but you have to ignore the `Lock CC` for now.

mbbush, 2022-01-28:
I can confirm that there are no config parameters implemented on the device.

AlCalzone, 2022-01-31:
I just found this video about the non-"Pro" version of this lock: https://youtu.be/TDHFHj9tOCg
Might be something to keep in mind when thinking of buying this 😬

guniv, 2022-04-04:
I hate to bump an old-ish issue but this is the only substantial discussion on the z-wave implementation of this lock. Is there any chance this lock will get better supported by zwavejs? I would really appreciate the ability to see when user codes are used and even set them. In Home Assistant the lock also adds a door sensor entity that does not update and is permanently set to open for me. I'm honestly a novice at this so I'm not quite sure how much of this could be Ultraloq's fault, and I do see the comment above that they may not be following the standard very well.

AlCalzone, 2022-04-05:
> In Home Assistant the lock also adds a door sensor entity that does not update and is permanently set to open for me.

This is probably the `Lock CC` value I mentioned here https://github.com/zwave-js/node-zwave-js/issues/3935#issuecomment-1021937543.

> I would really appreciate the ability to see when user codes are used and even set them.

According to https://github.com/zwave-js/node-zwave-js/issues/3935#issuecomment-997304784 I would assume that you're able to set user codes - you probably won't get any feedback that they are configured though because the lock responds with a status Available (i.e. not set).
Not sure what the lock sends when you use a user code, but I'd guess its the same as for most other locks, where a notification event is sent: https://www.home-assistant.io/integrations/zwave_js/#node-events-notification

guniv, 2022-04-05:
> This is probably the `Lock CC` value I mentioned here [#3935 (comment)](https://github.com/zwave-js/node-zwave-js/issues/3935#issuecomment-1021937543).
> 
Is there any way for me to fix this, then? At this point I assume it is an issue with the lock and Ultraloq would have to fix.
> According to [#3935 (comment)](https://github.com/zwave-js/node-zwave-js/issues/3935#issuecomment-997304784) I would assume that you're able to set user codes - you probably won't get any feedback that they are configured though because the lock responds with a status Available (i.e. not set). Not sure what the lock sends when you use a user code, but I'd guess its the same as for most other locks, where a notification event is sent: https://www.home-assistant.io/integrations/zwave_js/#node-events-notification

So I tested and was surprised to see that sending a door code to the lock did indeed work. I still, however, can find no way to see when a user code is used. Listening to zwave_js_notification or zwave_js_value_notification doesn't produce anything at all when I lock or unlock. Even listening to ALL events in Home Assistant (by putting * in the "listen to events" field) has nothing come up beyond the status of the lock entity changing from locked to unlocked and so on, as far as I can tell. I wanted to make sure there wasn't some other event it was pointed at for some reason.

Thanks for your response!

AlCalzone, 2022-04-05:
> EDIT: Actually, thinking about it again... the Lock CC would seem to me to be more likely the reason that two lock entities are added to Home Assistant. One of them never worked correctly and eventually just became completely unavailable, so I disabled it in Home Assistant. The other one works fine.

Thats what I mean. Ignore this entity and only use the one corresponding to the `Door Lock CC`.

> I still, however, can find no way to see when a user code is used

You're going to need to look at [driver logs](https://www.home-assistant.io/integrations/zwave_js/#how-to-access-the-z-wave-js-logs) then to see if there is actually anything being sent by the lock.

guniv, 2022-04-05:
> Thats what I mean. Ignore this entity and only use the one corresponding to the `Door Lock CC`.

So to be clear, there is both a 2nd lock entity and a door sensor entity. The door sensor entity is always set to "Open." [Here is a screenshot](https://i.imgur.com/52qvgi7.png) of all of the entities created by the device, for clarity.

> You're going to need to look at [driver logs](https://www.home-assistant.io/integrations/zwave_js/#how-to-access-the-z-wave-js-logs) then to see if there is actually anything being sent by the lock.

I watched the logs as I put in three codes: one I added via Z-Wave JS, one I created in their own app, and the last was using the fingerprint sensor. I THINK this log encompasses that including me locking the door between unlocks.

```
2022-04-05T14:45:25.575Z SERIAL « 0x011b0004000c159f03af00b822a2edcb0917cd51fda50110fa538dab90        (29 bytes)
2022-04-05T14:45:25.577Z CNTRLR   [Node 012] [~] [Door Lock] currentMode: 255 => 0                  [Endpoint 0]
2022-04-05T14:45:25.578Z CNTRLR   [Node 012] [~] [Door Lock] targetMode: 255 => 0                   [Endpoint 0]
2022-04-05T14:45:25.580Z CNTRLR   [Node 012] [~] [Door Lock] duration: {"value":0,"unit":"seconds"} [Endpoint 0]
                                   => {"value":0,"unit":"seconds"}
2022-04-05T14:45:25.581Z CNTRLR   [Node 012] [~] [Door Lock] outsideHandlesCanOpenDoor: false,false [Endpoint 0]
                                  ,false,false => true,false,false,false
2022-04-05T14:45:25.582Z CNTRLR   [Node 012] [~] [Door Lock] insideHandlesCanOpenDoor: false,false, [Endpoint 0]
                                  false,false => false,false,false,false
2022-04-05T14:45:25.583Z CNTRLR   [Node 012] [~] [Door Lock] latchStatus: "open" => "open"          [Endpoint 0]
2022-04-05T14:45:25.584Z CNTRLR   [Node 012] [~] [Door Lock] boltStatus: "locked" => "unlocked"     [Endpoint 0]
2022-04-05T14:45:25.585Z CNTRLR   [Node 012] [~] [Door Lock] doorStatus: "open" => "open"           [Endpoint 0]
2022-04-05T14:45:25.586Z SERIAL » [ACK]                                                                   (0x06)
2022-04-05T14:45:25.588Z DRIVER « [Node 012] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 175
                                    └─[DoorLockCCOperationReport]
                                        current mode:           Unsecured
                                        active outside handles: true, false, false, false
                                        active inside handles:  false, false, false, false
                                        latch status:           open
                                        bolt status:            unlocked
                                        door status:            open
                                        target mode:            Unsecured
                                        remaining duration:     0s
2022-04-05T14:45:27.822Z SERIAL « 0x01150004000c0f9f03b0007f405aefa51acfe8008a134a                    (23 bytes)
2022-04-05T14:45:27.825Z CNTRLR   [Node 012] [~] [Battery] level: 100 => 100                        [Endpoint 0]
2022-04-05T14:45:27.826Z CNTRLR   [Node 012] [~] [Battery] isLow: false => false                    [Endpoint 0]
2022-04-05T14:45:27.827Z SERIAL » [ACK]                                                                   (0x06)
2022-04-05T14:45:27.829Z DRIVER « [Node 012] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 176
                                    └─[BatteryCCReport]
                                        level:  100
                                        is low: false
2022-04-05T14:45:51.037Z SERIAL « 0x011b0004000c159f03b100e7b12ee302c68984aae7713611a59eb29286        (29 bytes)
2022-04-05T14:45:51.039Z CNTRLR   [Node 012] [~] [Door Lock] currentMode: 0 => 255                  [Endpoint 0]
2022-04-05T14:45:51.040Z CNTRLR   [Node 012] [~] [Door Lock] targetMode: 0 => 255                   [Endpoint 0]
2022-04-05T14:45:51.042Z CNTRLR   [Node 012] [~] [Door Lock] duration: {"value":0,"unit":"seconds"} [Endpoint 0]
                                   => {"value":0,"unit":"seconds"}
2022-04-05T14:45:51.045Z CNTRLR   [Node 012] [~] [Door Lock] outsideHandlesCanOpenDoor: true,false, [Endpoint 0]
                                  false,false => false,false,false,false
2022-04-05T14:45:51.047Z CNTRLR   [Node 012] [~] [Door Lock] insideHandlesCanOpenDoor: false,false, [Endpoint 0]
                                  false,false => false,false,false,false
2022-04-05T14:45:51.049Z CNTRLR   [Node 012] [~] [Door Lock] latchStatus: "open" => "open"          [Endpoint 0]
2022-04-05T14:45:51.050Z CNTRLR   [Node 012] [~] [Door Lock] boltStatus: "unlocked" => "locked"     [Endpoint 0]
2022-04-05T14:45:51.052Z CNTRLR   [Node 012] [~] [Door Lock] doorStatus: "open" => "open"           [Endpoint 0]
2022-04-05T14:45:51.054Z SERIAL » [ACK]                                                                   (0x06)
2022-04-05T14:45:51.056Z DRIVER « [Node 012] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 177
                                    └─[DoorLockCCOperationReport]
                                        current mode:           Secured
[... cut ...]
                                        bolt status:            locked
                                        door status:            open
                                        target mode:            Secured
                                        remaining duration:     0s
```

I'm far from an expert at interpreting these logs but I see nothing at all about codes or anything that could even correspond to codes. I do see that the "doorStatus" always goes from open to open, which may be related to the door sensor.

Again, really appreciate your help.

mbbush, 2022-04-05:
This lock doesn't support the Door Lock Logging command class, which I think is what would be required to be able to see which code is used when. 

I've been able to view and set user codes just fine, although there's the odd thing about the status I mentioned above. I've only done it by manually sending raw zwave commands; I haven't tried using any more user-friendly UI. 

I opened a support request with the manufacturer about some of these issues, and they seem open to feedback and suggestions for improvements to make in a firmware update. 

If there's something specifically that you'd like it to do, but it doesn't, maybe send the request to the ultraloq support team?

AlCalzone, 2022-04-05:
> which I think is what would be required to be able to see which code is used when.

Plenty of locks use the Notification CC to report user codes, e.g. Yale YRD220 ([link](https://devices.zwave-js.io/?jumpTo=0x0129:0x0002:0x0000:0.0), scroll down to the mapping table).

The standardized notifications define several variants to include the user codes:
https://github.com/zwave-js/node-zwave-js/blob/8256503cc18efc5e104bddda9eae6d94be64fe64/packages/config/config/notifications.json#L548-L561
https://github.com/zwave-js/node-zwave-js/blob/8256503cc18efc5e104bddda9eae6d94be64fe64/packages/config/config/notifications.json#L606-L621

IMO this would be the easiest way to implement this. Door Lock Logging is overkill for this use case, since it's meant for an audit log that captures multiple historical lock/unlock attempts.

guniv, 2022-04-05:
> I opened a support request with the manufacturer about some of these issues, and they seem open to feedback and suggestions for improvements to make in a firmware update.
>
> If there's something [...]

Is there a specific place you sent this in or just a support ticket on their website? I'd like to make sure it goes to the right place and hopefully doesn't just get ignored by customer service. If you have recommendations on how to word as well I would appreciate it because otherwise my ticket is just going to be "please let z-wave users see which door code/method was used to unlock the door"

Hopefully they plan on improving their Z-wave implementation so they can bring it to other products, and getting them feedback on an early product like this will help.

mbbush, 2022-04-05:
oh cool, I didn't realize the notifications command class could do that. That's much simpler. 

Unfortunately, (at least with the current firmware) this lock only sends the Door Lock Operation Report command when it is operated, which doesn't include any details about how/why it was operated (manually, code, fingerprint, which code, etc).

I just submitted a regular support ticket.

I'd probably also mention the Notification zwave command class as a way to accomplish this, and express your eagerness to have firmware updates available to improve the functionality of the product.

AlCalzone, 2022-04-06:
I just noticed, the lock does report that it doesn't support the door status, the driver just doesn't behave differently based on that. So we can at least fix that part...

guniv, 2022-04-07:
Just a heads up @mbbush after I sent in a support ticket I was very specifically asked to post feature requests [on their forum here.](https://community.u-tec.com/c/wishlist/8)

FirbyKirby, 2022-04-07:
Piling on to this thread. I too have the same lock, and am using it with zwave-js via Home Assistant (seems to essentially identical to @guniv setup.) My experience is identical (which is expected.)

@guniv , if you created a feature request post in their forum, can you post a link to your request? I'd like to reply specifically to it to add my weight to your request. Would recommend others to do the same so we can apply specific pressure to get what is needed for better support from the manufacturer in this toolchain.

guniv, 2022-04-07:
> if you created a feature request post in their forum, can you post a link to your request? I'd like to reply specifically to it to add my weight to your request. Would recommend others to do the same so we can apply specific pressure to get what is needed for better support from the manufacturer in this toolchain, rather then a disconnected set of feature requests from single users.

I haven't made one. If @mbbush has time and is willing to he might be the best person to write the post, since he seems to have a good understanding of Z-Wave and I do not. I can try and write something tomorrow, otherwise. I would definitely like to reply to the post as well and I'll probably reach out to people here and that I've seen on Reddit to ask them to also voice their support.

FirbyKirby, 2022-04-07:
> I haven't made one. If @mbbush has time and is willing to he might be the best person to write the post, since he seems to have a good understanding of Z-Wave and I do not.

Makes sense. I'm in the same boat. I'll standby and wait to see how it's decided to raise this issue with U-tec.

> I would definitely like to reply to the post as well and I'll probably reach out to people here and that I've seen on Reddit to ask them to also voice their support.

Roping in the Reddit thread is a good idea as well. That's actually how I routed here.

guniv, 2022-04-08:
@FirbyKirby @mbbush @Roundaround I have [made a post](https://community.u-tec.com/t/expand-and-fix-z-wave-capabilities/502) on their feature request forum. I think it would help greatly to get as many people as possible to reply to the post to say they also would like to see these features.

kpine, 2022-04-08:
> Additionally, Z-Wave currently reports two different locks to the controller (one Lock and one Door Lock),

Z-Wave JS is what is causing the two different locks, not the device.

guniv, 2022-04-08:
> Z-Wave JS is what is causing the two different locks, not the device.

Good to know, I thought based on reading this thread that it was unnecessarily reporting two types of devices. I will update my post on their forum.

kpine, 2022-04-08:
For context, see [my earlier comment](https://github.com/zwave-js/node-zwave-js/issues/3935#issuecomment-1005982754). Z-Wave JS doesn't differentiate between Z-Wave Plus and Z-Wave Plus V2 device requirements, so it mistakenly creates a "phantom" Lock CC.

FWIW, it doesn't really matter anymore. HA addressed this in the 2022.4 release and doesn't create a lock entity for the Lock CC if the Door Lock CC is also present. Only the latter is used, even for non Plus V2 devices.

FirbyKirby, 2022-04-08:
> I have [made a post](https://community.u-tec.com/t/expand-and-fix-z-wave-capabilities/502) on their feature request forum.

Thanks! I just added my response. Here's hoping we can catch the eye of a U-tec engineer cruising the forum.

guniv, 2022-04-08:
> Thanks! I just added my response. Here's hoping we can catch the eye of a U-tec engineer cruising the forum.

Thanks! I DM'ed a whole bunch of people on Reddit and asked them to chime in as well so we'll see how that goes, but two replies is already more than most posts get. 

@mbbush if you see this later, you said you are a Hubitat user. I would recommend you do the same and message people on the Hubitat forums who use this lock and ask if they'll chime in too.

closed 2022-04-09 by AlCalzone