zwave-js/zwave-js #3936: Fibaro FGBS-222 inputs state do not go into home assistant.

Herbs851, 2021-12-20:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

My module controls the opening of a gate.

The 2 outputs manage the partial opening and the total opening of the gate.

The motherboard of the gate sends 24v on the input 1 of the module when the gate is open.
The problem is that the state of input 1 does not change under home assistant with the zwavejs add-on (same thing with zwavejs2mqtt).
inputs are set as "Normally Closed alarm input"

I did several test but nothing seems to work

### Device information

Manufacturer:  Fibaro
Model name: FGBS-222

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version:
ZWAVEJS_SERVER_VERSION: 1.12.0
ZWAVEJS_VERSION: 8.8.3

### Did you change anything?

no

### If yes, what did you change?

nothing

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Yes it works with the old and deprecated zwave integration in home assistant (based on openzwave).

And it works during my first test few months ago with : 

zwave-js@8.1.1
@zwave-js/server@1.9.3

### Attach Driver Logfile

what can i see in log during opening : 

```
Niveau de journal modifié en : Debug
2021-11-22T14:23:37.139Z SERIAL » 0x010e00131407600d00062501ff250f6b                                  (16 bytes)
2021-11-22T14:23:37.162Z DRIVER » [Node 020] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      15
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 6
                                    └─[BinarySwitchCCSet]
                                        target value: true
2021-11-22T14:23:37.175Z SERIAL « [ACK]                                                                   (0x06)
2021-11-22T14:23:37.200Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-11-22T14:23:37.204Z SERIAL » [ACK]                                                                   (0x06)
2021-11-22T14:23:37.210Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-11-22T14:23:37.275Z SERIAL « 0x010700130f00000aee                                                 (9 bytes)
2021-11-22T14:23:37.279Z SERIAL » [ACK]                                                                   (0x06)
2021-11-22T14:23:37.286Z DRIVER « [REQ] [SendData]
                                    callback id:     15
                                    transmit status: OK
2021-11-22T14:23:37.315Z CNTRLR   [Node 020] [~] [Binary Switch] currentValue: false => true        [Endpoint 6]
2021-11-22T14:23:37.425Z SERIAL « 0x010d0004001407600d0600250300a8                                    (15 bytes)
2021-11-22T14:23:37.437Z CNTRLR   [Node 020] [~] [Binary Switch] currentValue: true => false        [Endpoint 6]
2021-11-22T14:23:37.441Z CNTRLR   [Node 020] Scheduled poll canceled because value was updated
2021-11-22T14:23:37.446Z SERIAL » [ACK]                                                                   (0x06)
2021-11-22T14:23:37.454Z DRIVER « [Node 020] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      6
                                    │ destination: 0
                                    └─[BinarySwitchCCReport]
                                        current value: false
2021-11-22T14:24:24.037Z SERIAL « 0x0113000400140d600d01007105000000ff07000011                        (21 bytes)
2021-11-22T14:24:24.046Z CNTRLR   [Node 020] [~] [Notification] notificationMode: "push" [Endpoint 1] [internal]
                                   => "push"
2021-11-22T14:24:24.052Z SERIAL » [ACK]                                                                   (0x06)
2021-11-22T14:24:24.063Z DRIVER « [Node 020] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      1
                                    │ destination: 0
                                    └─[NotificationCCReport]
                                        notification type:   7
                                        notification status: 255
                                        notification state:  idle
2021-11-22T14:24:24.224Z SERIAL « 0x0113000400140d600d01007105000000ff07000011                        (21 bytes)
2021-11-22T14:24:24.231Z CNTRLR   [Node 020] [~] [Notification] notificationMode: "push" [Endpoint 1] [internal]
                                   => "push"
2021-11-22T14:24:24.237Z SERIAL » [ACK]                                                                   (0x06)
2021-11-22T14:24:24.248Z DRIVER « [Node 020] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      1
                                    │ destination: 0
                                    └─[NotificationCCReport]
                                        notification type:   7
                                        notification status: 255
                                        notification state:  idle
```

AlCalzone, 2021-12-20:
Did you recently re-interview the device? If not, try it first.
Please also capture that process in a log (same settings as the one above) and if things still don't work afterwards, **attach** it here as a file (drag & drop).

Herbs851, 2021-12-20:
I move to zwavejs2mqtt since the first capture.

But i did what you propose and the result is : 

```
2021-12-20T15:26:03.276Z SERIAL « [ACK] (0x06)
2021-12-20T15:26:03.288Z SERIAL « 0x0104011301e8 (6 bytes)
2021-12-20T15:26:03.290Z SERIAL » [ACK] (0x06)
2021-12-20T15:26:03.296Z DRIVER « [RES] [SendData]
was sent: true
2021-12-20T15:26:03.307Z SERIAL « 0x01070013ac00000344 (9 bytes)
2021-12-20T15:26:03.309Z SERIAL » [ACK] (0x06)
2021-12-20T15:26:03.312Z DRIVER « [REQ] [SendData]
callback id: 172
transmit status: OK
2021-12-20T15:26:03.325Z SERIAL « 0x010c000400140670069802000009 (14 bytes)
2021-12-20T15:26:03.332Z CNTRLR [Node 020] [+] [Configuration] 152: 0
2021-12-20 16:26:03.335 INFO ZWAVE: Node 20: value added: 112-0-152 => 0
2021-12-20T15:26:03.337Z SERIAL » [ACK] (0x06)
2021-12-20T15:26:03.341Z DRIVER « [Node 020] [REQ] [ApplicationCommand]
└─[ConfigurationCCReport]
parameter #: 152
value size: 2
value: 0
2021-12-20T15:26:03.350Z CNTRLR « [Node 020] parameter #152 has value: 0
2021-12-20T15:26:03.351Z CNTRLR » [Node 020] querying parameter #153 value...
2021-12-20T15:26:03.369Z SERIAL » 0x010a0013140370059925ad95 (12 bytes)
2021-12-20T15:26:03.370Z DRIVER » [Node 020] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 173
└─[ConfigurationCCGet]
parameter #: 153
2021-12-20T15:26:03.379Z SERIAL « [ACK] (0x06)
2021-12-20T15:26:03.384Z SERIAL « 0x0104011301e8 (6 bytes)
2021-12-20T15:26:03.385Z SERIAL » [ACK] (0x06)
2021-12-20T15:26:03.390Z DRIVER « [RES] [SendData]
was sent: true
2021-12-20T15:26:03.402Z SERIAL « 0x01070013ad00000345 (9 bytes)
2021-12-20T15:26:03.404Z SERIAL » [ACK] (0x06)
2021-12-20T15:26:03.407Z DRIVER « [REQ] [SendData]
callback id: 173
transmit status: OK
2021-12-20T15:26:03.421Z SERIAL « 0x010c000400140670069902000008 (14 bytes)
2021-12-20T15:26:03.426Z CNTRLR [Node 020] [+] [Configuration] 153: 0
2021-12-20 16:26:03.428 INFO ZWAVE: Node 20: value added: 112-0-153 => 0
2021-12-20T15:26:03.430Z SERIAL » [ACK] (0x06)
2021-12-20T15:26:03.434Z DRIVER « [Node 020] [REQ] [ApplicationCommand]
└─[ConfigurationCCReport]
parameter #: 153
value size: 2
value: 0
2021-12-20T15:26:03.445Z CNTRLR « [Node 020] parameter #153 has value: 0
2021-12-20T15:26:03.446Z CNTRLR » [Node 020] querying parameter #154 value...
2021-12-20T15:26:03.466Z SERIAL » 0x010a0013140370059a25ae95 (12 bytes)
2021-12-20T15:26:03.468Z DRIVER » [Node 020] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 174
└─[ConfigurationCCGet]
parameter #: 154
2021-12-20T15:26:03.482Z SERIAL « [ACK] (0x06)
2021-12-20T15:26:03.486Z SERIAL « 0x0104011301e8 (6 bytes)
2021-12-20T15:26:03.488Z SERIAL » [ACK] (0x06)
2021-12-20T15:26:03.496Z DRIVER « [RES] [SendData]
was sent: true
2021-12-20T15:26:03.509Z SERIAL « 0x01070013ae00000346 (9 bytes)
2021-12-20T15:26:03.511Z SERIAL » [ACK] (0x06)
2021-12-20T15:26:03.514Z DRIVER « [REQ] [SendData]
callback id: 174
transmit status: OK
2021-12-20T15:26:03.530Z SERIAL « 0x010b000400140570069a01000c (13 bytes)
[... 354 lines cut ...]
2021-12-20 16:26:05.519 INFO ZWAVE: Node 20: value added 20-32-3-targetValue => undefined
2021-12-20 16:26:05.521 INFO ZWAVE: Node 20: value added 20-49-3-Voltage => 0
2021-12-20 16:26:05.874 INFO ZWAVE: Node 20: value added 20-32-4-currentValue => 29
2021-12-20 16:26:05.876 INFO ZWAVE: Node 20: value added 20-32-4-targetValue => undefined
2021-12-20 16:26:05.878 INFO ZWAVE: Node 20: value added 20-49-4-Voltage => 28.45
2021-12-20 16:26:06.187 INFO ZWAVE: Node 20: value added 20-37-5-currentValue => false
2021-12-20 16:26:06.189 INFO ZWAVE: Node 20: value added 20-37-5-targetValue => undefined
2021-12-20 16:26:06.192 INFO ZWAVE: Node 20: value added 20-117-5-local => 2
2021-12-20 16:26:06.193 INFO ZWAVE: Node 20: value added 20-117-5-rf => 0
2021-12-20 16:26:06.195 INFO ZWAVE: Node 20: value added 20-117-5-exclusiveControlNodeId => undefined
2021-12-20 16:26:06.196 INFO ZWAVE: Node 20: value added 20-117-5-timeout => undefined
2021-12-20 16:26:06.476 INFO ZWAVE: Node 20: value added 20-37-6-currentValue => false
2021-12-20 16:26:06.478 INFO ZWAVE: Node 20: value added 20-37-6-targetValue => undefined
2021-12-20 16:26:06.481 INFO ZWAVE: Node 20: value added 20-117-6-local => 2
2021-12-20 16:26:06.483 INFO ZWAVE: Node 20: value added 20-117-6-rf => 0
2021-12-20 16:26:06.484 INFO ZWAVE: Node 20: value added 20-117-6-exclusiveControlNodeId => undefined
2021-12-20 16:26:06.486 INFO ZWAVE: Node 20: value added 20-117-6-timeout => undefined
2021-12-20 16:26:06.765 INFO ZWAVE: Node 20: value added 20-32-7-currentValue => 26
2021-12-20 16:26:06.768 INFO ZWAVE: Node 20: value added 20-32-7-targetValue => undefined
2021-12-20 16:26:06.769 INFO ZWAVE: Node 20: value added 20-49-7-Air temperature => 25.6
2021-12-20 16:26:06.772 INFO ZWAVE: Node 20: value added 20-113-7-System-Hardware status => 0
2021-12-20 16:26:06.786 INFO ZWAVE: Node 20 ready: Fibargroup - FGBS222 (Smart Implant)
2021-12-20T15:26:06.787Z CNTRLR [Node 020] The node is ready to be used
2021-12-20 16:26:07.080 INFO ZWAVE: Node 20: interview COMPLETED, all values are updated
```

AlCalzone, 2021-12-20:
The entire log please on level debug, **attached** as a file.

See [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs) for details.

Herbs851, 2021-12-20:
here are the logs 

[zwavejs_2021-12-20.log](https://github.com/zwave-js/node-zwave-js/files/7747690/zwavejs_2021-12-20.log)

AlCalzone, 2021-12-20:
Ok, it does look like the commands go through but the device does not change status. There is this
```
2021-12-20T19:21:16.745Z CNTRLR « [Node 020] received protection status:
                                  local: NoOperationPossible
                                  rf     Unprotected
```
in the log which says that local control on the device is disabled, but maybe the device wrongly interprets the RF protection instead. Try changing `local` to `Unprotected` - you'll likely have to do that in zwavejs2mqtt.

Herbs851, 2021-12-21:
I tested what you suggest, but it doesn't change anything.

I put the logs here.

Could the problem be identical to : 

[](https://community.home-assistant.io/t/fibaro-universal-binary-sensor-fgbs001-not-working-when-upgrading-z-wave-js-to-0-1-16/296802)

[zwavejs_2021-12-21.log](https://github.com/zwave-js/node-zwave-js/files/7751171/zwavejs_2021-12-21.log)

closed 2022-01-07 by zwave-js-assistant[bot]