zwave-js/zwave-js #3939: Unable to set Configuration Parameter 7 for Zooz ZSE40 V2

pletch, 2021-12-21:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Attempt to set available parameter options for parameter 7 to [4] No Temperature, Flashing Motion via the Zwavejs2mqtt interface and then wake up the device to receive the parameter update.  

What do you observe?
Parameter is not updated when refreshing the Zwavejs2mqtt interface.  For example, attempting to the set the value to [4] No Temperature, Flashing Motion resulted in a value of [9] Custom being stored in parameter 7 when observing parameter value in Zwavejs2mqtt.

What did you expect to happen?
Parameter is updated to selected value.

Steps to reproduce the behavior:
  1. Go to Control panel and select ZSE40 device.
  2. Click on Configuration toggle to expose options.
  3. Scroll down to Parameter 7 and select value of [4] No Temperature, Flashing Motion.  Hit send button and then manually wake device to allow value to be updated at device.
  4. Zwavejs2mqtt response that parameter was updated but the value does not match the originally selected value and instead shows [9] Custom.

### Device information

Manufacturer: Zooz
Model name:  ZSE40 V2
Node ID in your network: 41

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [X] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 6.1.1.e82be01
zwave-js: 8.9.0-beta.3

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs2mqtt_2021-12-21.log](https://github.com/zwave-js/node-zwave-js/files/7757126/zwavejs2mqtt_2021-12-21.log)
[zwavejs_2021-12-21.log](https://github.com/zwave-js/node-zwave-js/files/7757127/zwavejs_2021-12-21.log)

AlCalzone, 2021-12-21:
After trying to change the parameter, the device first says the value is 3:
```
2021-12-21T17:42:10.928Z DRIVER « [Node 041] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -70 dBm
                                  └─[ConfigurationCCReport]
                                      parameter #: 7
                                      value size:  1
                                      value:       3
```

and shortly after it reports the value is 9:
```
2021-12-21T17:42:16.024Z DRIVER « [Node 041] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -94 dBm
                                  └─[ConfigurationCCReport]
                                      parameter #: 7
                                      value size:  1
                                      value:       9
```

@blhoward2 since you are in contact with Zooz, are you aware of an issue with the device?

blhoward2, 2021-12-21:
Haven't heard anything about this.

pletch, 2021-12-21:
I had a dumb thought on this.  I noticed the value of parameter 7 was set to 3 out of the box on a brand new device.  The zooz documentation [here](https://www.getzooz.com/downloads/zooz-z-wave-plus-4-in-1-sensor-zse40-ver2-manual.pdf) suggests the default should be mode 4 though the range of values is also stated to be 1-4.  Is it possible that the valid values on the device correspond to 0 - 3 for the 4 different mode options rather than 1-4?  It does accept seem to properly accept a value of 1 - 3 via the zwavejs2mqtt interface.

blhoward2, 2021-12-29:
It's possible, but the documentation and our files match. I'll ask Zooz.

closed 2022-01-09 by zwave-js-assistant[bot]