zwave-js/zwave-js #3943: Basic CC Exposure

Roagert, 2021-12-22:
**Is your feature request related to a problem? Please describe.**
Logic Group - Matrix ZDB5100 has updated their devices and is now providing a central scene for KeyPressedx2 and x3 but removed the x1. The x1 is now switched to a Basic CC and is now able to change from 0 to 99 instead of a central scene.

**Describe the solution you'd like**
I would like to get the Basic Switch exposed so its readable in Home Assistant

**Describe alternatives you've considered**
None, sudden firmware update when received a new device. I might be able to revert back to an older version but that kind of defeats the purpose in the future

[image]

AlCalzone, 2021-12-22:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly` and attach it here as a file.

While the log is running, make the device send such a basic command.

Also we'll need to know the ids of the device and firmware version where it changed.

Roagert, 2021-12-22:
Device has changed from firmware 1.5 to 1.6. I have contacted Logic Group for confirmation.

Here's the log file i have tried to re-interview the device in the log file.
The  node that i have tested is:
v1.6 = Node 91
v1.5 = Node 51
[zwavejs2mqtt-store_basic.zip](https://github.com/zwave-js/node-zwave-js/files/7760902/zwavejs2mqtt-store_basic.zip)

AlCalzone, 2021-12-22:
Honestly, this is just a step back. Wonder what they were thinking...

Roagert, 2021-12-22:
Okay if you think that is the case (i dont know the history) then i can see if i can revert back or maybe its a bug? 

But why wouldnt we want to share the Basic ?

AlCalzone, 2021-12-22:
@zwave-js-bot add compat flag to 0x0234:0x0003:0x0121:1.6
```
// Starting in Firmware 1.6, the device no longer uses Central Scene for Pressed / Released, only Basic CC
"disableBasicMapping": true
```

AlCalzone, 2021-12-22:
> But why wouldnt we want to share the Basic

Because most of the time it is unnecessary and it has no default meaning attached.

Roagert, 2021-12-22:
Alright, let me get in contact with them and understand whats happening

AlCalzone, 2021-12-22:
Basic CC is like the name implies basic. It can be used for simple communication, but in 99% of cases a more specific CC is available. And in this case it is even used for half the functionality.

Central Scene tells you how the device was operated even without knowing anything about it. Basic CC just gives you numbers, so UIs won't help you make sense of them.

Roagert, 2021-12-22:
Right that does make sense what you are saying because the Basic can be "anything".
I will talk to Logic Group about it.

Roagert, 2021-12-22:
@AlCalzone i just tried to go a downgrade to 1.5 and i didnt receive the scene control back. 
Strangely enough i just received like 1 button for a brief second the scene for x1 press 
2021-12-22 11:20:50.164 INFO ZWAVE: Node 93: value notification: 91-0-scene-002 0

but I cannot replicate the scene by clicking the button again.
Might be an issue with zwavejs instead ?

AlCalzone, 2021-12-22:
> Might be an issue with zwavejs instead ?

Very unlikely. Also, look at driver logs not the application logs.

Roagert, 2021-12-22:
i downgraded to 1.5 and i went back to the latest tag and i still dont see any scene notification from the device. even after a re inclusion

Roagert, 2021-12-22:
i do see in the log that the "scenes" are supported but im not seing them in the log

`2021-12-22T11:54:05.350Z DRIVER « [Node 094] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 183
                                    └─[CentralSceneCCSupportedReport]
                                        scene count:                     4
                                        supports slow refresh:           true
                                        supported attributes (scene #1):
                                        · KeyPressed
                                        · KeyReleased
                                        · KeyHeldDown
                                        · KeyPressed2x
                                        · KeyPressed3x
                                        supported attributes (scene #2):
                                        · KeyPressed
                                        · KeyReleased
                                        · KeyHeldDown
                                        · KeyPressed2x
                                        · KeyPressed3x
                                        supported attributes (scene #3):
                                        · KeyPressed
                                        · KeyReleased
                                        · KeyHeldDown
                                        · KeyPressed2x
                                        · KeyPressed3x
                                        supported attributes (scene #4):
                                        · KeyPressed
                                        · KeyReleased
                                        · KeyHeldDown
                                        · KeyPressed2x
                                        · KeyPressed3x`

AlCalzone, 2021-12-22:
The same report was there in the log for FW 1.6. They aren't used though. Maybe that's a firmware bug in 1.6.

Roagert, 2021-12-22:
Right. 
But even when i downgraded to 1.5 i cannot recreate the scene of KeyPressed

AlCalzone, 2021-12-22:
May be a config setting on the device?

Roagert, 2021-12-22:
i do have that enabled (i tried to toggle the config) no change.

I tried to implement the config in my local store/config folder. 
I should see the Basic CC in HA now or wont it get exposed like that ?

AlCalzone, 2021-12-22:
> I should see the Basic CC in HA now or wont it get exposed like that ?

Start by checking in zwavejs2mqtt after a re-interview. I'm not sure if HA exposes any Basic CC values aside from the `event`.

Roagert, 2021-12-22:
zwavejs_event doesnt show any data to me in the developer tab there for the request of the exposure of the Basic CC like other events.
i tried re-interviewing the device. 
I have now reset the device and still doesnt show the scene for all of the KeyPressed1x.

Logic Group told me that they are planning to release a 1.7 version in Q1 and there for the 1.6 is only on a specific delivery of device. They didnt support me with what to do since they are using Fibaro Home Center and there for no support outside of that.

AlCalzone, 2021-12-22:
> zwavejs_event doesnt show any data to me in the developer tab there for the request of the exposure of the Basic CC like other events.

That's different. Only `Basic Set` commands are exposed as events. Your device uses `Basic Report`s which updates the `currentValue`. The compat flag just makes it so this one isn't mapped to a more appropriate CC like Binary Switch and that the Basic CC isn't hidden.

AlCalzone, 2021-12-22:
> They didnt support me with what to do since they are using Fibaro Home Center

Ugh this makes me angry. Your problem has nothing to do with who the device is talking to. It is all about what the device itself is transmitting - clearly an issue on their end.

AlCalzone, 2021-12-22:
So I've talked to the HA devs. If the config file was applied correctly, `currentValue` should be exposed, but it may be disabled by default.

AlCalzone, 2021-12-22:
@Roagert can you post a driver log of a re-interview please?
I'd like to make sure we're not hunting ghosts here.

Roagert, 2021-12-22:
> So I've talked to the HA devs. If the config file was applied correctly, `currentValue` should be exposed, but it may be disabled by default.

And how to enable it ? 

> > They didnt support me with what to do since they are using Fibaro Home Center
> 
> Ugh this makes me angry. Your problem has nothing to do with who the device is talking to. It is all about what the device itself is transmitting - clearly an issue on their end.
Yes this is also annoying me. But i guess thats their expertise so its easier to support.

> @Roagert can you post a driver log of a re-interview please? I'd like to make sure we're not hunting ghosts here.

Yes ofc. heres a log of the re-interview with version 1.5 (i dont have the firmware of 1.6 available, the device was delivered with the 1.6 and not available for download)
[zwavejs2mqtt-store_20211222.zip](https://github.com/zwave-js/node-zwave-js/files/7763554/zwavejs2mqtt-store_20211222.zip)

AlCalzone, 2021-12-22:
```
2021-12-22T15:21:55.581Z CNTRLR   [Node 095] Embedded device config loaded
```

The custom config wasn't applied. Make sure to follow this guide: https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files

Roagert, 2021-12-22:
I know, i thought you wanted it without ? 

I can do the re interview with the local config

AlCalzone, 2021-12-22:
Nah, I wanted to see what happens when the config is applied. To figure out if you're not seeing the currentValue because of something in HA or here.

Roagert, 2021-12-22:
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/7764279/zwavejs2mqtt-store.zip)

Here ya go

AlCalzone, 2021-12-22:
Hmm, I think there's something missing. Please put this file in a subfolder `templates` in the custom config dir:
https://raw.githubusercontent.com/zwave-js/node-zwave-js/master/packages/config/config/devices/templates/master_template.json

Then try again.

Roagert, 2021-12-22:
[image]
[zwavejs2mqtt-store_2016.zip](https://github.com/zwave-js/node-zwave-js/files/7764721/zwavejs2mqtt-store_2016.zip)

AlCalzone, 2021-12-22:
Uhm... @robertsLando what's wrong here? I don't see it. The mentioned file is there.
```
2021-12-22T17:31:17.482Z CONFIG   Using external configuration dir /usr/src/app/store/.config-db
2021-12-22T17:31:17.483Z CONFIG   version 8.9.0-beta.3
2021-12-22T17:31:18.766Z CONFIG   Error parsing config file zdb5100.json: Could not parse config file /usr/src/a
                                  pp/store/config/templates/master_template.json: ENOENT: no such file or direct
                                  ory, open '/usr/src/app/store/config/templates/master_template.json'
                                  Import stack: 
                                    in #base_enable_disable
                                    in /usr/src/app/store/config/zdb5100.json (ZW0340)
```

robertsLando, 2021-12-23:
> Uhm... @robertsLando what's wrong here? I don't see it. The mentioned file is there.

I really have no clue, I checked the path multiple times and it seems correct (no typo).  I can only suggest to restart application 🤷🏼‍♂️

Roagert, 2021-12-23:
Multiple restarts tried but nothing really changed :/

So i just saw this in the log:
2021-12-23 18:45:49.232 INFO ZWAVE: Node 95: interview stage OVERWRITECONFIG completed

but still the same:
```
2021-12-23T18:42:27.715Z CONFIG   Error parsing config file zdb5100.json: Could not parse config file /usr/src/a
                                  pp/store/config/templates/logic_group_template.json: ENOENT: no such file or d
                                  irectory, open '/usr/src/app/store/config/templates/logic_group_template.json'
                                  Import stack: 
                                    in #dimming_duration
                                    in /usr/src/app/store/config/zdb5100.json (ZW0340)2021-12-23T18:42:27.715Z CONFIG   Error parsing config file zdb5100.json: Could not parse config file /usr/src/a
                                  pp/store/config/templates/logic_group_template.json: ENOENT: no such file or d
                                  irectory, open '/usr/src/app/store/config/templates/logic_group_template.json'
                                  Import stack: 
                                    in #dimming_duration
                                    in /usr/src/app/store/config/zdb5100.json (ZW0340)
```

AlCalzone, 2021-12-24:
Oh that refers to a different template file.

AlCalzone, 2021-12-27:
Let me know if this is too much trouble. I can also merge the config file change and you can test when it is available officially.

PR #3944 (2021-12-29): fix(config): disable Basic CC mapping for Logic Group ZDB5100
fixes: #3943

**TODO:**
- [x]	Change PR title to be more specific
closed 2021-12-29 by AlCalzone