zwave-js/zwave-js #3955: Crash in TransactionMachine.ts

opsnlops, 2021-12-23:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

It just happens with time.

What do you observe?

zwavejs stops running. The docker container exits.

What did you expect to happen?

zwavejs to handle this error more gracefully.

Steps to reproduce the behavior:

I can't reproduce it at will, it just happens over time.

### Device information

It's not with any one device.

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwavejs2mqtt: 6.1.1
zwave-js: 8.9.0-beta.1

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

zwavejs has been rock solid for a long time. This happened in a recent update.

### Attach Driver Logfile

[zwavejs-log-sample.log](https://github.com/zwave-js/node-zwave-js/files/7770890/zwavejs-log-sample.log)

AlCalzone, 2021-12-24:
8.9.0-beta.1 is broken. Update to the latest version including 8.9.1.

opsnlops, 2021-12-24:
Thanks, will do. I'm not sure how I even got on a beta version, I pull `latest`.

AlCalzone, 2021-12-27:
Closing for now - feel free to reopen if the issue persists.

closed 2021-12-27 by AlCalzone

opsnlops, 2021-12-28:
Yup, thanks! I pulled the latest version and it's been stable ever since.