zwave-js/zwave-js #3958: Trouble healing with 500 series

justindthomas, 2021-12-25:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I recently switched back to using an Aeotec Z-Stick 5+ (instead of the 7 I've been using for a few months). I used the migration tool that @AlCalzone has been working on to shift back.

I'm having some trouble completing a full network heal, though. About 50% of my nodes do okay, but the others fail both when triggered individually and when done as part of a network heal. Although one that was problematic just made a liar out of my and healed okay while I was trying to capture a log. I moved to another, and it failed. So the behavior seems to be inconsistent. The driver logs are attached.

I do have one node near the center of my network that is currently offline (I inadvertently unplugged its power and am not nearby to plug it back in). The interface seems to have recognized that it's offline and it is now marked to "skip." I thought I could just heal around it (honestly, I'd rather that node not be heavily relied on) - but could its existence be causing this issue?

### Device information

Manufacturer: Aeotec
Model name: Z-Stick 5+
Node ID in your network: 13

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.9.0-beta.1
`zwavejs2mqtt` branch: 0.32.1

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/7776314/zwave_js.log)

justindthomas, 2021-12-25:
Further information: doing a Soft Reset on the controller seems to help. I was able to heal 6 stubborn nodes before it started failing again with the "no response from controller" errors.

jmgiaever, 2021-12-25:
Upgrade to Z2M 6.2.0 that doesn't include beta.1. That was released by a mistake.

justindthomas, 2021-12-25:
Okay, thanks. I'll work on that. I try to update as soon as new releases are made available through the official integrations, but I'll figure out how to accelerate that locally

jmgiaever, 2021-12-26:
Try to use the master branch. I'm not using docker so not sure how it works. But I know the master tag has it. :)

AlCalzone, 2021-12-27:
Closing for now - feel free to reopen if the issue persists in v8.9.2.

closed 2021-12-27 by AlCalzone