zwave-js/zwave-js #3959: Infinite REQ AssignSUCReturnRoute

kpine, 2021-12-27:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Upgraded to z2m 6.2.0 / zwave-js 8.9.1. The problem was not occurring prior to that point, as far as I can tell.

What do you observe?

A battery powered node appears in the z2m as if it's waking and sleeping. This happens for as long as z2m has been online after an upgrade. The driver is trying to perform a `AssignSUCReturnRoute` request for some reason, and that fails. It repeats this forever. It's not obvious to me that the device is actually waking up.

What did you expect to happen?

The infinite requests to a sleeping a battery device to not occur.

Steps to reproduce the behavior:

Don't know.

### Device information

Manufacturer: BRK Brands, Inc. 
Model name: ZCOMBO
Node ID in your network: 10

I have 3 of these same devices, and only 1 of them exhibits the problem currently.

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 6.2.0
`zwavejs2mqtt` branch: 8.9.1

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Upgraded to z1m 6.2.0.

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I haven't noticed the problem in earlier releases. Not sure if it's just a coincidence. I haven't tried downgrading yet, but I might need to do that.

### Attach Driver Logfile

Here are a few days of activity. Problem starts in 12-26 log. After a restart of z2m it persists.

[zwavejs_2021-12-24.log.gz](https://github.com/zwave-js/node-zwave-js/files/7778222/zwavejs_2021-12-24.log.gz)
[zwavejs_2021-12-25.log.gz](https://github.com/zwave-js/node-zwave-js/files/7778223/zwavejs_2021-12-25.log.gz)
[zwavejs_2021-12-26.log](https://github.com/zwave-js/node-zwave-js/files/7778225/zwavejs_2021-12-26.log)

I can provide a zniffer trace during this time as well, if needed.

kpine, 2021-12-27:
Confirmed that reverting back to 8.9.0-beta.3, the problem goes away. Upgrading to 8.9.1, the problem returns.

[zwavejs_2021-12-26.log](https://github.com/zwave-js/node-zwave-js/files/7778249/zwavejs_2021-12-26.log)

AlCalzone, 2021-12-27:
most likely caused by https://github.com/zwave-js/node-zwave-js/pull/3949

AlCalzone, 2021-12-27:
I actually can't reproduce on `8.9.1`, even if I cheat by changing the nodes status:
```
18:53:48.140 DRIVER » [Node 014] [REQ] [AssignSUCReturnRoute]
                        payload: 0x0e03
18:53:48.146 SERIAL « [ACK]                                                                   (0x06)
18:53:48.147 SERIAL « 0x0104015101aa                                                       (6 bytes)
18:53:48.148 SERIAL » [ACK]                                                                   (0x06)
18:53:48.149 DRIVER « [RES] [AssignSUCReturnRoute]
                        was executed: true
18:53:48.446 SERIAL « 0x01210051030125010000007f7f7f7f7f00010600000000020900007f7f7f7f7f0 (35 bytes)
                      000a5
18:53:48.447 SERIAL » [ACK]                                                                   (0x06)
18:53:48.448 DRIVER « [REQ] [AssignSUCReturnRoute]
                        callback id:     3
                        transmit status: NoAck
18:53:48.455 CNTRLR   [Node 014] The node did not respond. It is probably asleep, moving its message
                      s to the wakeup queue.
18:53:48.456 CNTRLR   [Node 014] The node is now asleep.
```

AlCalzone, 2021-12-27:
@kpine can you share a dump for node 10?

kpine, 2021-12-27:
Here's a dump from z2m, is that what you wanted?  [10.txt](https://github.com/zwave-js/node-zwave-js/files/7781258/10.txt)

I've started of a build to use the linked PR, I'll give it a try soon.

kpine, 2021-12-27:
I think the fix is working. For good measure, I upgraded back to z2m 6.2.0, confirmed the problem, then upgraded to a z2m 6.2.0 build with the linked PR. At first glance, it seems to have fixed the problem.

[zwavejs_2021-12-27.log](https://github.com/zwave-js/node-zwave-js/files/7781450/zwavejs_2021-12-27.log)

My question is (and perhaps why you were asking for the dump?) why is zwave-js trying to assign a return route?

AlCalzone, 2021-12-27:
> why is zwave-js trying to assign a return route?

I guess because it didn't work before. Without https://github.com/zwave-js/node-zwave-js/pull/3949, the previous attempts likely failed with NoACK because the device was asleep.

PR #3962 (2021-12-27): fix: generalize transmit report handling to all node queries
fixes: #3959
closed 2021-12-27 by AlCalzone