zwave-js/zwave-js #3961: [question] Aeotec Z-Stick Gen5 not initializing

haifishtime, 2021-12-27:
Hey,
I'm trying to implement a new Zwave network and have bought an Aeotec Z-Stick Gen5 as a starting point. After hours of googling I have now completely run out of ideas to try. I hope this is the correct place to ask for help :)

### My setup:
ZwaveJs2Mqtt v6.2.0
ZwaveJS 8.9.1
installed via Truenas Community Plugin
OS Version 12.2-RELEASE-p9
Aeotec Z-Stick Gen5 (non +)

### What I'm trying to do:
Have ZwaveJs2Mqtt connect to the Z-Stick and have it displayed in the Control Panel. Later I would like to connect this to homeassistant but besides disabling MQTT I have not done anything in that direction yet.

### What I have done:
Passed /dev/cua* through to the jail. If I connect the stick to my server the port /dev/cuaU0 appears so I assume this is the correct port. When I run `ls -l /dev/cu*` within the jail I can see the port so I assume this part is working correctly.

I have typed the port into ZwaveJs2Mqtt and generated all 4 of the keys with the UI.

MQTT Gateway is as I said disabled while I have WS Server in homeassistant settings enabled.

Thought about permission problems or drivers or similar issues but ultimatively did nothing in those directions since I usually didn't know how to do it or what I saw looked correct to me.

### Settings
[image]

### Driver log
This output reappears every couple of seconds always with the same output.

`2021-12-25T20:14:00.589Z DRIVER   serial port opened`
`2021-12-25T20:14:00.587Z DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗`
`                                  ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝`
`                                    ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗`
`                                   ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║`
`                                  ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║`
`                                  ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝`
`2021-12-25T20:14:00.589Z SERIAL » [NAK]                                                                   (0x15)`
`2021-12-25T20:14:00.587Z DRIVER   version 8.9.1`
`2021-12-25T20:14:00.587Z DRIVER   `
`2021-12-25T20:14:00.587Z DRIVER   starting driver...`
`2021-12-25T20:14:00.588Z DRIVER   opening serial port /dev/cuaU0`
`2021-12-25T20:14:00.590Z DRIVER   loading configuration...`
`2021-12-25T20:14:00.590Z CONFIG   version 8.9.1`
`2021-12-25T20:14:00.803Z CONFIG   Priority device configuration directory /usr/local/etc/zwavejs2mqtt/config not found`
`2021-12-25T20:14:00.813Z DRIVER   beginning interview...`
`2021-12-25T20:14:00.813Z DRIVER   added request handler for AddNodeToNetwork (0x4a)...`
`                                  1 registered`
`2021-12-25T20:14:00.813Z DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...`
`                                  1 registered`
`2021-12-25T20:14:00.813Z DRIVER   added request handler for ReplaceFailedNode (0x63)...`
`                                  1 registered`
`2021-12-25T20:14:00.813Z CNTRLR   querying controller IDs...`
`2021-12-25T20:14:00.815Z SERIAL » 0x01030020dc                                                         (5 bytes)`
`2021-12-25T20:14:00.816Z DRIVER » [REQ] [GetControllerId]`
`2021-12-25T20:14:01.821Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try in 100 ms.`
`2021-12-25T20:14:01.929Z SERIAL » 0x01030020dc                                                         (5 bytes)`
`2021-12-25T20:14:01.929Z DRIVER » [REQ] [GetControllerId]`
`2021-12-25T20:14:02.933Z CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try in 1100 ms.`
`2021-12-25T20:14:04.040Z SERIAL » 0x01030020dc                                                         (5 bytes)`
`2021-12-25T20:14:04.040Z DRIVER » [REQ] [GetControllerId]`
`2021-12-25T20:14:05.063Z DRIVER   Failed to initialize the driver: ZWaveError: Timeout while waiting for an ACK `
`                                  from the controller (ZW0200)`
`                                      at Driver.sendMessage (/usr/local/share/zwavejs2mqtt/node_modules/zwave-js`
`                                  /src/lib/driver/Driver.ts:3350:23)`
`                                      at ZWaveController.identify (/usr/local/share/zwavejs2mqtt/node_modules/zw`
`                                  ave-js/src/lib/controller/Controller.ts:677:33)`
`                                      at Driver.initializeControllerAndNodes (/usr/local/share/zwavejs2mqtt/node`
`                                  _modules/zwave-js/src/lib/driver/Driver.ts:977:26)`
`                                      at Immediate.<anonymous> (/usr/local/share/zwavejs2mqtt/node_modules/zwave`
`                                  -js/src/lib/driver/Driver.ts:857:16)`
`2021-12-25T20:14:05.064Z DRIVER   destroying driver instance...`
`2021-12-25T20:14:12.879Z DRIVER   driver instance destroyed`

### Z-Wave log
Reappears every couple of seconds as well
`2021-12-25 22:17:08.024 INFO ZWAVE: Connecting to /dev/cuaU0`
`2021-12-25 22:17:20.185 INFO ZWAVE: Client closed`

I have completely run out of ideas on what to try or what to search for so I'm happy for any help I can get.
Thanks for any help. I'm happy to provide any additional information that might be helpful. Sorry again if this is the wrong place for this problem.

AlCalzone, 2021-12-27:
Honestly, I have no idea here. The stick doesn't respond at all, so the problem must lie somewhere in the OS or VM setup.

haifishtime, 2021-12-27:
Okay. Thanks for answering. Will see what I can do :/

closed 2021-12-27 by haifishtime

jmgiaever, 2021-12-27:
Hi. Are you using this stick with a Raspberry Pi 4? 

If so, then you need a hub to get it working. 

There's an upgraded stick (5gen +) that is working with the Pi 4 without a hub.

haifishtime, 2022-01-01:
> 

Hey,
I know about the problem with pi4, but I'm not using a pi nor do I have one in my network even.

haifishtime, 2022-01-01:
Managed to get it working. Read something about the umodem driver being used for it so I tried making sure I was using that driver. I tried understanding the logic behind it, but I don't think I did understand it in the end. 
As I see it the /dev/cuaU0 port is a port of the umodem driver so I must have used it before as well.

I did activate umodem driver by editing /boot/loader.conf on the host system and adding `umodem_load="YES"`.
After a reboot and fixing the ruleset again the stick shows up in the UI.

Thanks for trying to help me:)