zwave-js/zwave-js #3970: [bug] One device refuses OTA,

LordMike, 2021-12-27:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**
I'm trying to downgrade a firmware in order to troubleshoot an issue that has popped up, but on one of my devices (all same make, model, firmware, ..), firmware updates fail with "user action required".

**What do you observe?**
I've updated all my Logic Group ZDB5100 devices to the same firmware (1.5) from an older one (1.4). This all worked fine over a few days, and did not once show signs of requiring user interaction. It all worked - they updated, restarted, they were reinterviewed and so on.

Now, trying to downgrade one of them, I'm seeing an error: `Error_AuthenticationExpected` - which does not occur on other devices I've just tried out. These other devices (same make, model, ..) immediately begin the OTA with no further intervention.

**What did you expect to happen?**
I expected this one device to function the same as the others (and before). I've tested all 11 devices, and only this one device is failing.

### Device information

Manufacturer: Logic Group
Model name: ZDB5100
Node ID in your network: 

- Failing: 76
- Working: 5, 6, 9, 25, 27, 28, 30, 31, 32, 75 (all my other ZDB5100's)

All devices were updated to 1.5 within the last few weeks. I'm currently attempting a downgrade, and no. 76 was within eyesight, so I started with that.

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 6.2.0
zwave-js: 8.9.1
home id: 3815907651
home hex: 0xe3722143

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I've updated all devices to 1.5, so OTA _does_ work (without user intervention).

### Did this work before?

Yes (please describe)

### If yes, where did it work?
[... 134 lines cut ...]
2021-12-27T20:14:44.638Z SERIAL » [ACK]                                                                   (0x06)
2021-12-27T20:14:44.639Z DRIVER « [REQ] [SendData]
                                    callback id:     27
                                    transmit status: OK
2021-12-27T20:14:44.765Z SERIAL « 0x010900040005037a04ff75                                            (11 bytes)
2021-12-27T20:14:44.766Z SERIAL » [ACK]                                                                   (0x06)
2021-12-27T20:14:44.766Z DRIVER « [Node 005] [REQ] [ApplicationCommand]
                                  └─[FirmwareUpdateMetaDataCCRequestReport]
                                      status: OK
2021-12-27T20:14:44.946Z SERIAL « 0x010b00040005057a050100018f                                        (13 bytes)
2021-12-27T20:14:44.947Z SERIAL » [ACK]                                                                   (0x06)
2021-12-27T20:14:44.948Z DRIVER « [Node 005] [REQ] [ApplicationCommand]
                                  └─[FirmwareUpdateMetaDataCCGet]
                                      total # of reports: 1
                                      report number:      1
2021-12-27T20:14:44.950Z CNTRLR » [Node 005] Sending firmware fragment 1 / 2483
2021-12-27T20:14:44.963Z SERIAL » 0x01350013052e7a06000180000183b6422ed7f505d0061247a5e4fffe121ac5900 (55 bytes)
                                  3c7e017231f02184e6b866ba6ff7fb8ef909962251c10
2021-12-27T20:14:44.963Z DRIVER » [Node 005] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      28
                                  └─[FirmwareUpdateMetaDataCCReport]
                                      report #: 1
                                      is last:  false
```

closed 2021-12-27 by LordMike

AlCalzone, 2021-12-27:
Did you figure this out? `Error_AuthenticationExpected` seems like the device expects you to push a button or something.

LordMike, 2021-12-27:
I did ... The device had gone into an invalid state. I cut power to the house and rebooted it.. It could successfully begin a firmware update after that.. But it had me worried for a moment.

That being said - there are a number of latent bugs popping up on these devices all over.. So I'm a bit worried ... But I'll figure it out. :)