zwave-js/zwave-js #3971: Missing device configuration: Zooz ZSE43

yieldhog, 2021-12-27:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Zooz ZSE43

### Manufacturer ID

634

### Product Type

28672

### Product ID

0x7000

### Firmware Version

1.10

### Is the device listed on the Z-Wave Alliance website?

Not yet

### Do you have a manual?

[zooz-700-series-tilt-shock-xs-sensor-zse43-manual.pdf](https://github.com/zwave-js/node-zwave-js/files/7781816/zooz-700-series-tilt-shock-xs-sensor-zse43-manual.pdf)

AlCalzone, 2021-12-27:
I've converted the IDs to hex.

> Product Type 0x7000
> Product ID 0x7000

seems a bit suspicous. Are you sure? Where is this info from?

yieldhog, 2021-12-27:
> 

Hi, coming from zwave2mqtt -- sorry if incorrect, first time I'm adding a device:

```
Product: 0x7000
Product code: Unknown product 0xe003
Manufacturer: Showing correct as Zooz

Device ID: 634-57347-28672 (0x027a-0x7000-0xe003)
```

AlCalzone, 2021-12-27:
That looks better 👍🏻

PR #3979 (2021-12-29): feat(config): add Zooz ZSE43 and ZSE44 device config files

Add config file for Zooz zse43 tilt/shock sensor

Closes #3971
closed 2021-12-29 by AlCalzone