zwave-js/zwave-js #3992: Restoring NVM from 500 to 700 series controller marks all nodes as dead

Gunsmithy, 2021-12-31:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Restoring the NVM backup from a 500 series stick (Aeotec ZW090) to a 700 series stick (Zooz ZST10 700). 

What do you observe?
All nodes failing their interviews, stuck on "Protocol Info" in z2m, all mains-powered nodes being marked as dead.

What did you expect to happen?
I expected all nodes to connect happily to their new controller. (Not really, that's why I was testing this, to break things. 😉)

Steps to reproduce the behavior:
1. Backed up old 500 stick with z2m UI and then shut down addon.
2. Replaced sticks and started addon.
3. Changed USB serial ID in z2m config to new stick and restarted addon, saw no nodes in z2m list.
4. Restored NVM to 700 stick using custom driver function (below) in z2m UI with restoreNVM so it would convert then restarted addon, saw all nodes in list but all interviewing.

I was able to stop the addon, swap sticks, restart, change the USB path and restart again to restore previous network state fine. No attempt at re-interview in this case, only a full re-interview for all devices with the new stick it seems.

```
const fs = this.require("fs-extra");  
const { storeDir } = this.require('../config/app');  
const nvm = fs.readFileSync(storeDir + "/NVM_RAW_2021-12-30.bin");  
await driver.controller.restoreNVM(nvm);
```

### Device information

Manufacturer: Zooz
Model name: ZST10 700
Node ID in your network: All

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 8.9.1
`node-zwave-js` branch: master
`zwavejs2mqtt` branch: master

Using the latest HaOS z2m addon version 0.32.2, with z2m 6.2.0 with zjs 8.9.1

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwavejs_2021-12-30.log](https://github.com/zwave-js/node-zwave-js/files/7795433/zwavejs_2021-12-30.log)

Controller backup starts at 2021-12-31T02:51:22.750Z on line 25313

AlCalzone, 2021-12-31:
Looks like the home ID and SUC Node ID didn't transfer properly. Can you send me backups of the 500 series, and the 700 series (before conversion)? Preferably to the email in my profile.

Gunsmithy, 2021-12-31:
The 700 was factory new then restored with the converted 500 NVM backup. I can provide the 500 series raw backup easy, but not sure what you want for the 700. Would you like me to factory reset it or just do a raw backup of it now in its messed state?

Gunsmithy, 2021-12-31:
Emailed you the 500 series backup and the 700 series in its current state.

AlCalzone, 2022-01-11:
Does this still reproduce with the latest version of zwavejs2mqtt and the built-in restore function?

Gunsmithy, 2022-01-11:
Still appears to be the case. I did the following running the latest z2m 6.3.1

1. Backed up NVM of 500 stick using the UI to NVM_2022-01-11.bin
2. Shut down addon and swapped to 700 stick.
3. Started addon and changed USB serial, saved, driver restarted, everything dead like before.
4. Restored NVM to 700 stick using the UI, after completion, everything still dead.
5. Restarted the addon, everything still dead.

6. Backed up NVM of 700 stick using the UI to NVM_2022-01-11_Dead700.bin
7. Factory reset 700 stick and restarted addon, then backed up using the UI to NVM_2022-01-11_Fresh700.bin
8. Restored NVM backup NVM_2022-01-11.bin again using the UI, everything still dead.
9. Backed up NVM of 700 stick using the UI to NVM_2022-01-11_Dead700_2.bin

I've attached the logs for today during all of this, and have emailed you the above named files.
Let me know if there are any other steps I can try!

[zwavejs_2022-01-11.log](https://github.com/zwave-js/node-zwave-js/files/7849033/zwavejs_2022-01-11.log)

AlCalzone, 2022-01-11:
Ok these logs are confusing me like hell. The restored 700 stick has a different home ID than the 500 series one and a different one than before the restore.
Lets see if the backups can shed some light onto this.

c96pv, 2022-01-29:
Thanks to @kpine for referencing this. You might see more threads opened on this topic with the recent 700 firmware release (people, like me, trying to migrate back to their original 700 after converting to a 500).

Botched1, 2022-01-29:
I see the same thing. Working Aeotec 500, backup nvm, swap out sticj for UZB7, restart container, restore NVM. All nodes dead, nothing will ping.

Tried the conversion from the UI twice and command line once - same result.

Went back to the Aeotec 500, back working again.

mundschenk-at, 2022-01-30:
Unfortunately, I just experienced the same thing (Aeotec 5gen+ to Aeotec 700). I'm not sure if additional logfiles will help at this point, but I can provide some if required (via mail).

rice1204, 2022-02-01:
tagging for same issue (Aeotec 5 to ZST10 700). Anything I can do to help troubleshoot?

AlCalzone, 2022-02-01:
I'll need to find some time to investigate this. I should have all the logs/files I need for now and I now also have a 500-series stick to test with.

Danny2100, 2022-02-03:
I tried going from an aeotec Gen5+ to a zooz 700 series as well and had all nodes die and then the integration lost the node IDs.

Danny2100, 2022-02-05:
> I'll need to find some time to investigate this. I should have all the logs/files I need for now and I now also have a 500-series stick to test with.

Any luck yet on figuring out what may be causing the issue? I have an aeotec 700 series stick coming today is it worth testing that one too and seeing if I get dead nodes just as I did with the zooz 700 series stick?

AlCalzone, 2022-02-05:
@Danny2100 having a second data point could be interesting. AFAIK these two 700 series sticks have different NVMs, so maybe that's an influence (or not). I'll try to figure this out over the weekend.

closed 2022-02-06 by AlCalzone

Kyoshi67, 2023-01-18:
Hello,
I'm not used to GitHub. Sorry, if my message is not in the good place.
I have a similar problem. I am currently using an Aeotec Gen5 controller on a French home automation solution (Jeedom). This solution uses zwavejs.

I made a backup with Z-Wave JS UI of my Aeotec Gen5 ZStick (BackupAeotecGen5.bin), an other of my Aeotec Gen7 ZStick (BackupAeotecGen7.bin).
Then I converted the backup with this command : npx @zwave-js/nvmedit convert --source BackupAeotecGen5.bin --target BackupAeotecGen7.bin --out NewGen7.bin

I then restored my NewGen7.bin backup on my Gen7.
But all nodes as dead. I retested, rebooted, nothing...

I am not the only one in this situation. Several users of the solution are in the same situation as me.

Here are the versions of my SDK :
- Gen5 : SDK 6.7
- Gen7 : SDK 7.18.1

Kyoshi67, 2023-02-13:
> And yes a new issue should be created, if the issue persists after trying what I said as this isn't the right place.

Hello,
Thank you.
I created a issue few weeks ago (#5347)
I hope it is complete and that I have not forgotten anything because there is no feedback.