zwave-js/zwave-js #3994: SupervisionReport no longer sent in response to a SupervisionGet

IvanBrazza, 2021-12-31:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Sending a Supervision-encapsulated command (like NotificationReport - only example I have).

What do you observe?
The node doesn't receive a SupervisionReport in response within its defined timeout period, causing it to think the command wasn't processed and resending the original command again. There are no issues on 8.8.2 and issues start from 8.9.0-beta.0.

Log snippet from 8.9.2:

```
2021-12-31T14:39:58.846Z SERIAL « 0x012200a8000133199f03c700c10880ed30819b4873895ba9e2f5faf61eeaf6ad2 (36 bytes)
                                  500b1eb
2021-12-31T14:39:58.847Z CNTRLR   [Node 051] [~] [Notification] notificationMode: "push" [Endpoint 0] [internal]
                                   => "push"
2021-12-31T14:39:58.848Z SERIAL » [ACK]                                                                   (0x06)
2021-12-31T14:39:58.849Z DRIVER « [Node 051] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -79 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 199
                                    └─[SupervisionCCGet]
                                      │ session id:      1
                                      │ request updates: false
                                      └─[NotificationCCReport]
                                          notification type:   Home Security
                                          notification status: 255
                                          notification state:  Intrusion
2021-12-31T14:39:58.850Z CNTRLR   [Node 051] [~] [Notification] Home Security[Sensor status]: 2 =>  [Endpoint 0]
                                  2
2021-12-31T14:40:20.780Z SERIAL « 0x012200a8000133199f03c800e4339d298cc601c041617468185950166674dd845 (36 bytes)
                                  000b173
2021-12-31T14:40:20.781Z CNTRLR   [Node 051] [~] [Notification] notificationMode: "push" [Endpoint 0] [internal]
                                   => "push"
2021-12-31T14:40:20.781Z SERIAL » [ACK]                                                                   (0x06)
2021-12-31T14:40:20.782Z DRIVER « [Node 051] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -79 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 200
                                    └─[SupervisionCCGet]
                                      │ session id:      2
                                      │ request updates: false
                                      └─[NotificationCCReport]
                                          notification type:   Home Security
                                          notification status: 255
                                          notification state:  Intrusion
2021-12-31T14:40:20.783Z CNTRLR   [Node 051] [~] [Notification] Home Security[Sensor status]: 2 =>  [Endpoint 0]
                                  2
```

Snippet from 8.8.2:
```
2021-12-31T14:43:12.785Z SERIAL « 0x012200a8000133199f03d0003a00b879170752afe9fa1cfbc3aa09c2e9899b9c2 (36 bytes)
                                  100b5a1
2021-12-31T14:43:12.786Z CNTRLR   [Node 051] [~] [Notification] notificationMode: "push" [Endpoint 0] [internal]
                                   => "push"
2021-12-31T14:43:12.786Z SERIAL » [ACK]                                                                   (0x06)
2021-12-31T14:43:12.787Z DRIVER « [Node 051] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -75 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 208
                                    └─[SupervisionCCGet]
                                      │ session id:      9
                                      │ request updates: false
                                      └─[NotificationCCReport]
                                          notification type:   Home Security
                                          notification status: 255
                                          notification state:  Intrusion
2021-12-31T14:43:12.788Z CNTRLR   [Node 051] [~] [Notification] Home Security[Sensor status]: 0 =>  [Endpoint 0]
                                  2
2021-12-31T14:43:12.796Z SERIAL » 0x011d00a90133119f03f400e8f7446af88bd2116020c3851525000000000eb9    (31 bytes)
2021-12-31T14:43:12.796Z DRIVER » [Node 051] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      14
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 244
                                    └─[SupervisionCCReport]
                                        session id:          9
                                        more updates follow: false
                                        status:              Success
                                        duration:            0s
2021-12-31T14:43:12.805Z SERIAL « [ACK]                                                                   (0x06)
2021-12-31T14:43:12.805Z SERIAL « 0x010401a90152                                                       (6 bytes)
2021-12-31T14:43:12.806Z SERIAL » [ACK]                                                                   (0x06)
2021-12-31T14:43:12.806Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2021-12-31T14:43:12.818Z SERIAL « 0x011800a90e00000100b57f7f7f7f0000030000000003010000f5              (26 bytes)
2021-12-31T14:43:12.819Z SERIAL » [ACK]                                                                   (0x06)
2021-12-31T14:43:12.819Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            14
                                    transmit status:        OK, took 10 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 100 kbit/s
                                    ACK RSSI:               -75 dBm
                                    ACK channel no.:        0
                                    TX channel no.:         0
```

What did you expect to happen?
The node should receive a SupervisionReport to confirm that the command was successfully processed.
I suspect this is due to the changes made in https://github.com/zwave-js/node-zwave-js/pull/3858

Steps to reproduce the behavior:
  1. Have node send Supervision-encapsulated command

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.9.2
`zwavejs2mqtt` branch: 6.2.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Version 8.8.2 (via zwavejs2mqtt 6.0.2) works fine

### Attach Driver Logfile

[zwavejs_2021-12-31_8.9.2.log](https://github.com/zwave-js/node-zwave-js/files/7796477/zwavejs_2021-12-31_8.9.2.log)
[zwavejs_2021-12-31_8.8.2.log](https://github.com/zwave-js/node-zwave-js/files/7796478/zwavejs_2021-12-31_8.8.2.log)

AlCalzone, 2022-01-01:
Can you post a dump of the node or the json cache file please?

IvanBrazza, 2022-01-01:
Sure: 
[node_51.json.txt](https://github.com/zwave-js/node-zwave-js/files/7797760/node_51.json.txt)

AlCalzone, 2022-01-01:
Probably has to do with the node being a sleeping device.

IvanBrazza, 2022-01-01:
> Probably has to do with the node being a sleeping device.

I was just thinking that - I remember when we implemented SupervisionGet that unless the priority of the Report was set to the highest it wouldn't be sent to a sleeping node until it woke up - which is too late.

IvanBrazza, 2022-01-01:
For what it's worth, I changed `MessagePriority.Supervision` to `MessagePriority.Nonce` in [SupervisionCC.ts#L112](https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/commandclass/SupervisionCC.ts#L112) just to test and the Reports are sent back as expected

PR #4012 (2022-01-04): fix: reply to Supervision requests when the node is asleep
fixes: #3994
closed 2022-01-04 by AlCalzone