zwave-js/zwave-js #3995: Missing configuration parameters for 0x0208/hkzw_so03

klyoku, 2021-12-31:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Missing configurations for parameters# 151, 152, 171, 172, 173, 174. 
Details for the parameters can be found in the device manual here: 
http://manuals-backend.z-wave.info/make.php?lang=en&sku=HKZW-SO03&cert=ZC10-16065127

### Device information

Manufacturer: Hank
Model name: HKZW-SO03

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.1.51
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

N/A

AlCalzone, 2022-01-03:
Let's see if this works...
@zwave-js-bot import config 1854 from zwa

AlCalzone, 2022-01-03:
The imported file in that PR needs to be cleaned up. If someone wants to help out, please raise a PR with `import-config-1641217424` as the target branch.

PR #4084 (2022-01-16): fix(config): add missing parameters for HANK Electronics HKZW-SO03
Fixes #3995

Sorry, not sure how I can do this without duplicating the existing PR (#4020)
PR #4020 (2022-01-18): fix(config): add missing parameters for HANK Electronics HKZW-SO03
fixes: #3995

**TODO:**
- [x]	Change PR title to be more specific
closed 2022-01-18 by AlCalzone