zwave-js/zwave-js #4008: Jasco 45609 missing some manual switch presses

drytoastman, 2022-01-01:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**

Manual presses at the switch will sometimes send updates, other times not.

**What do you observe?**

Some presses show log entries for CNTRLR like:

```
Received updated node info
Node does not send unsolicited updates; refreshing actuator and sensor values...
querying Binary Switch state...

```
Other times, nothing is received, _**even at the SERIAL/DRIVER level**_.   I'm not sure if nothing is being sent from the switch or the controller is eating it.   No reports of dropped RX packets but there were two dropped tx packets somewhere during the day.

The missed updates are random too.   Sometimes it works for 5-6 presses in a row, then none for a couple, working again for 2-3 then off for another 5 or more.   No relation to requests to change state from the controller either.   Zwave operation always works as expected.

Its actually happening on 3 different GE/Jasco 45609 switches as well as one GE/Jasco 12722.   These are all old ZWave switches from almost a decade ago.   None of the newer switches (all Zwave+) have any issues.

**What did you expect to happen?**

I expect some update for each manual press.

**Steps to reproduce the behavior:**
  1. Manual press switch on/off until an update isn't received

### Device information

Manufacturer: GE/Jasco 0x0063
Model name: 45609
Node ID in your network:  was originally 3, now its 32 after playing with removing and re-including

Z-Wave Controller:   HUSBZB-1 (firmware 4.32)

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 8.9.1

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

It did work with a previous HA/openzwave install.   Its been an issue since moving to zwavejs about a year ago but I was waiting to see if any updates fixed it.   Maybe its time to retire the old hardware.

### Attach Driver Logfile

[zwavejs_2021-12-31.log](https://github.com/zwave-js/node-zwave-js/files/7797388/zwavejs_2021-12-31.log)

AlCalzone, 2022-01-03:
The communication looks mostly fine. There was one instance at the beginning where the node didn't acknowledge a command, so this might be caused by temporary RF issues.

If nothing shows up in the logs at all, either the node doesn't send anything or the commands don't reach the controller due to RF issues. Both are things the host software (OZW or ZJS) has no influence on and both can only be diagnosed with a Zniffer (Z-Wave RF sniffer).

You might want to check out https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues first before you retire the switches though.

blhoward2, 2022-01-06:
Old switches frequently don't call in their state at all. It's not surprising the implementation is subpar.

AlCalzone, 2022-01-06:
Actually, they are getting polled when the device calls in - this is a legacy Z-Wave trick to get around a patent at the time:
```
2022-01-01T04:30:20.895Z DRIVER « [Node 032] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x200a04100125277370867277
2022-01-01T04:30:20.900Z CNTRLR « [Node 032] Received updated node info
2022-01-01T04:30:20.901Z CNTRLR   [Node 032] Node does not send unsolicited updates; refreshing actuator and sen
                                  sor values...
2022-01-01T04:30:20.904Z CNTRLR » [Node 032] querying Binary Switch state...
2022-01-01T04:30:20.912Z SERIAL » 0x0109001320022502256ca9                                            (11 bytes)
2022-01-01T04:30:20.913Z DRIVER » [Node 032] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      108
                                  └─[BinarySwitchCCGet]
2022-01-01T04:30:20.915Z SERIAL « [ACK]                                                                   (0x06)
2022-01-01T04:30:20.919Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-01-01T04:30:20.920Z SERIAL » [ACK]                                                                   (0x06)
2022-01-01T04:30:20.923Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-01-01T04:30:20.964Z SERIAL « 0x010500136c0085                                                     (7 bytes)
2022-01-01T04:30:20.976Z SERIAL » [ACK]                                                                   (0x06)
2022-01-01T04:30:20.977Z DRIVER « [REQ] [SendData]
                                    callback id:     108
                                    transmit status: OK
2022-01-01T04:30:21.013Z SERIAL « 0x010900040020032503ff08                                            (11 bytes)
2022-01-01T04:30:21.015Z CNTRLR   [Node 032] [~] [Binary Switch] currentValue: false => true        [Endpoint 0]
2022-01-01T04:30:21.018Z SERIAL » [ACK]                                                                   (0x06)
2022-01-01T04:30:21.020Z DRIVER « [Node 032] [REQ] [ApplicationCommand]
                                  └─[BinarySwitchCCReport]
                                      current value: true
2022-01-01T04:30:21.024Z CNTRLR « [Node 032] received Binary Switch state:
                                  current value:      true
```

But if the first message doesn't reach the controller, it can't know it has to poll.

drytoastman, 2022-01-18:
I've always had a USB extender for the controller and the switch in question is a first hop node that is about 5-6 feet away.   I don't have the old setup but I'm starting to guess that OZW was setup to poll some of these.   I know it polled an old Zwave Thermostat for current temperature.   I'm just replacing the ones where the current state is important for some automation tasks.  

Out of curiosity, from the zwavejs level, would debug logging show all the packets received by the controller.

closed 2022-01-18 by drytoastman

AlCalzone, 2022-01-18:
> Out of curiosity, from the zwavejs level, would debug logging show all the packets received by the controller.

Yes