zwave-js/zwave-js #4013: Backup fails with unknown nvm size

robsee, 2022-01-02:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Using the NVM Backup feature, it fails with "Error while calling api backupNVMRaw: Unknown NVM size - cannot backup! (ZW0207)"

### Device information

Manufacturer: Nortek
Model name: HUSBZB-1 
Node ID in your network:1

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 6.2.0
zwave-js: 8.9.1

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs2mqtt_2022-01-02.log](https://github.com/zwave-js/node-zwave-js/files/7799652/zwavejs2mqtt_2022-01-02.log)

AlCalzone, 2022-01-03:
That is not a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs).

robsee, 2022-01-03:
Looking at another ticket, I believe it is due to the software on the stick not supporting NVM backups. Closing this issue.

closed 2022-01-03 by robsee