zwave-js/zwave-js #4016: Aeotec Dual Nano Switch only reports total meters, not individual meters

ntompson, 2022-01-03:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

#### Intro
The Aeotec ZW132 Dual Nano Switch contains two separately controlled switches in the one package. Like OZW, zwave-js correctly controls both switches (individually or together). The ZW132 also includes three meters to observe power, energy, current and voltage for:

- total (loads 1 and 2 together)
- load 1
- load 2

#### What causes the bug?
I don't know what causes the bug.

#### What do you observe?
zwave-js only captures reports for the total meters, not for the individual loads. Looking at the logs, the same `[Meter] value[xx]` number is always reported, for example `66049` in the case of the power meter:

```
2022-01-03T04:48:09.282Z CNTRLR   [Node 023] [~] [Meter] value[66049]: 54.876 => 68.526             [Endpoint 0]
```

Corresponding sensors in HA only update for the total meters, not the individual loads.

#### What did you expect to happen?
I expected to see meter reports for loads 1 and 2, as occurred for OZW. Similarly, I'd expect to see the HA sensors update for the individual loads. I can see that this used to work prior to migrating from OZW by looking at the sensor history in HA for the individual loads.

*Very occasionally* (once per day-ish) the individual load meters update. I can't figure out what's triggering it, so I haven't been able to capture logs.

Steps to reproduce the behavior:
  1. In HA, activate / deactivate one of the two switches
  2. Observe HA power sensors

### Device information

Manufacturer: Aeotec
Model name: ZW132 Dual Nano Switch
Node ID in your network: 23

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: no idea, sorry. Whatever comes with the standard HA integration. I'm up to date with everything (HA is at 2021.12.7, Supervised, non Development)

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Prior to migrating from the deprecated OZW integration, all of this worked just fine.

### Attach Driver Logfile

Example log file in which the first load is activated, ~10 seconds pause, then the second load is activated.

[zwave_js_20220103.log](https://github.com/zwave-js/node-zwave-js/files/7800499/zwave_js_20220103.log)

AlCalzone, 2022-01-03:
Please make another driver log and re-interview the device while doing so. This is usually an issue with how the lifeline association is set up.

ntompson, 2022-01-04:
Thanks so much for the quick response. Here is the driver log from a re-interview.

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/7806375/zwave_js.log)

AlCalzone, 2022-01-04:
Hmm. According to our config files, we force a "normal" association instead of a Multi Channel Assocation. The latter would make sense to get individual endpoint reports, but apparently it has caused problems in the past.

Could you reach out to the Aeotec support and have them tell you how the Lifeline association(s) for this device is supposed to be set up, so all reports from all endpoints are sent?

ntompson, 2022-01-05:
Happy to. Can I check that I am asking the right question:

> Hi,
> 
> I am hoping you can help advise the correct configuration for the Aeotec ZW132 Dual Nano Switch product. I am controlling this product using Home Assistant with the zwave-js integration via the Aoetec Z-stick Gen5. Currently, I have found that the regular reporting of meters (current, energy, power, voltage) is only working for the "total" meters and not for the individual loads. The total meters update regularly (with the frequency determined by the parameter 90 / 91 / 92 / 101 / 102 / 103 / 111 / 112 / 113 settings).
> 
> I've been advised to ask for the correct Lifeline association(s) for the ZW132 to ensure that all reports from all endpoints are sent?

Let me know if you think this is sufficiently accurate (as I am uncertain of the technical detail here).

Thanks

AlCalzone, 2022-01-05:
Looks good.

ntompson, 2022-01-07:
Thanks again for your help @AlCalzone. Aeotec came back to me quickly, and I think I can see what is going on. Firstly, their response:

> By default, the Dual Nano witch is set so that only the total values are transmitted.
> 
> If you want to receive the individual values, deactivate the parameter 90.
> Change the parameter 101 to 1776399 and the parameter 111 to 60. 
> 
> To determine the individual values, follow the petition description in the enclosed instructions on the last page.

I've attached the full document, but here is the key table:

[image]

Comparing this to the [current configuration for the ZW132](https://github.com/zwave-js/node-zwave-js/blob/d611f734db707e3d5bac3cd8efc38e063f257252/packages/config/config/devices/0x0086/zw132.json#L155):

```js
{
	"#": "101[0x01]",
	"$import": "templates/aeotec_template.json#auto_report_group1_kwh"
},
{
	"#": "101[0x02]",
	"$import": "templates/aeotec_template.json#auto_report_group1_watt"
},
{
	"#": "101[0x04]",
	"$import": "templates/aeotec_template.json#auto_report_group1_v"
},
{
	"#": "101[0x08]",
	"$import": "templates/aeotec_template.json#auto_report_group1_amp"
},
```

I can see that only the LSB of the 4 byte parameter `101` (and `102`/`103`) are encoded, which means zwave-js is currently only able to configure the meters for "whole Channels" (ie the totals, not the individual loads). Additional configuration would be required to configure the individual loads mapped to the (somewhat messy) configuration described in the table above.

Unfortunately, I haven't been able to validate the solution by setting parameter `101` to `1776399` as suggested by Aeotec support. I have attempted to use the `zwave_js.set_config_parameter` service to set it manually:

```yaml
service: zwave_js.set_config_parameter
data:
  parameter: 101
  value: 1776399
target:
  entity_id:
    - switch.stairs_all
```

But this fails with, the following error logged:

```
zwave_js_server.exceptions.NotFoundError: Configuration parameter with value ID 23-112-0-101 could not be found
```

I assume this is because the value `1776399` invalidates the bitmask config set for parameter `101` as above. My guess is that it would work fine: `1776399` translates to activating all of the meters.

  | 7 | 6 | 5 | 4 | 3 | 2 | 1 | 0
-- | -- | -- | -- | -- | -- | -- | -- | --
MSB | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0
  | 0 | 0 | 0 | 1 | 1 | 0 | 1 | 1
  | 0 | 0 | 0 | 1 | 1 | 0 | 1 | 1
LSB | 0 | 0 | 0 | 0 | 1 | 1 | 1 | 1

I also guess that this is how parameter `101` was configured in OZW before the migration to zwave-js, but my monkeying around in zwave-js post migration zeroed out bytes 2 and 3.

Hopefully this gives you enough to fix the bug?

I should add: the template to report the bug didn't give me the opportunity to thank you and the team for your amazing work with zwave-js. I can't believe how much better my zwave gear is working now. It's fast, it's reliable, there is less traffic on the zwave network and I don't need any of the dumb work arounds I used to have under OZW to poll for missed state changes and the like. OZW had given me the idea that zwave is just a bit crap and I'd be better off using ESP32/8266 devices (shudders). So thank you.

Reference:

[ES - Dual Nano Switch ZW132 V2-01.pdf](https://github.com/zwave-js/node-zwave-js/files/7825856/ES.-.Dual.Nano.Switch.ZW132.V2-01.pdf)

kpine, 2022-01-07:
> I have attempted to use the `zwave_js.set_config_parameter` service to set it manually

Try [`zwave_js.bulk_set_partial_config_parameters`](https://www.home-assistant.io/integrations/zwave_js/#service-zwave_jsbulk_set_partial_config_parameters) instead.

ntompson, 2022-01-07:
Thanks @kpine - that sounded like a good idea. This:

```yaml
service: zwave_js.bulk_set_partial_config_parameters
target:
  entity_id: switch.stairs_all
data:
  parameter: '101'
  value: 1776399
```

Yielded an unknown error, with the following in the logs:

```
websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Config parameter 101 failed validation on partial parameter 8
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zwave_js_server/util/node.py", line 238, in _bulk_set_validate_and_transform_new_value
    return _validate_and_transform_new_value(zwave_value, new_partial_value)
  File "/usr/local/lib/python3.9/site-packages/zwave_js_server/util/node.py", line 172, in _validate_and_transform_new_value
    raise InvalidNewValue(
zwave_js_server.exceptions.InvalidNewValue: Must provide a value for 23-112-0-101-8 that represents a valid state key or label from {"0": "Disable", "1": "Enable"}

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 381, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 584, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/services.py", line 381, in async_bulk_set_partial_config_parameters
    cmd_status = await async_bulk_set_partial_config_parameters(
  File "/usr/local/lib/python3.9/site-packages/zwave_js_server/util/node.py", line 132, in async_bulk_set_partial_config_parameters
    _validate_raw_int(partial_param_values, new_value)
  File "/usr/local/lib/python3.9/site-packages/zwave_js_server/util/node.py", line 327, in _validate_raw_int
    _bulk_set_validate_and_transform_new_value(
  File "/usr/local/lib/python3.9/site-packages/zwave_js_server/util/node.py", line 240, in _bulk_set_validate_and_transform_new_value
    raise BulkSetConfigParameterFailed(
zwave_js_server.exceptions.BulkSetConfigParameterFailed: Config parameter 101 failed validation on partial parameter 8
```

Looks similar to the previous attempt: it's failing a validation check.

kpine, 2022-01-07:
Hmm, I had thought it bypasses the checks if you set a raw value, but unfortunately it doesn't.

Besides waiting for a fix, your other option would be to switch to zwavejs2mqtt. You can set raw parameters that way.

ntompson, 2022-01-07:
Thanks @kpine. The process to switch from the official zwavejs add-on to zwavejs2mqtt looks painful. When I first did the migration, I hadn't realised there were benefits to using zwavejs2mqtt other than MQTT itself, so just went with the standard set up. But now it's done, and I don't relish having to reconfigure / rename everything just to make the switch...

kpine, 2022-01-07:
You don't have to rename anything. The [process](https://www.home-assistant.io/integrations/zwave_js/#how-do-i-switch-between-the-official-z-wave-js-add-on-and-zwavejs2mqtt) isn't that difficult. If you go through with it, just make sure all the interviews are Complete in z2m before re-enabling the integration (otherwise you _will_ have to rename due to a bug in HA).

gorillica, 2022-04-04:
I've got the same problem with the ZW132, and had struck on the `1776399` value myself following the Aeotec engineering manual (bitmask operations make my head hurt). I tried putting that in for parameter `101` using ZwaveJS2MQTT, but it doesn't seem to have taken effect. It looks like @ntompson is on top of things here, but is there anything I can do to help?

kpine, 2022-04-05:
@gorillica You have tried the custom configuration option? If so, what do the driver debug logs show?

[image]

gorillica, 2022-04-05:
Here are the logs from when I set the custom parameter, and then turning off and on the two loads, watching the updated power readings coming in. I removed anything related to any other node, hope this is OK.
[node_39_log.txt](https://github.com/zwave-js/node-zwave-js/files/8414217/node_39_log.txt)

kpine, 2022-04-05:
Need a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs).

gorillica, 2022-04-05:
Whoops, sorry. Here…
[node_39_driver_log.txt](https://github.com/zwave-js/node-zwave-js/files/8414387/node_39_driver_log.txt)

Anyway, I've just realized that the device does support power and current per-channel, but only reports at the automatic reporting interval (parameter `111`). I had this set quite high: around 30 minutes, and instead wanted to rely on power changes to trigger events (parameters `90`, `91`, `92`).

Those events evidently get triggered for the whole node (both channels), but not for the individual channels. You might see in the log that I've experimented with dropping the reporting interval to `20 seconds`, and began to see power reported on both individual channels, presumably because parameter `101` is now set to `1776399` as discussed previously. So, it can be made to work, but I hate having the reporting interval so low because of increased network traffic.

So, can we conclude that this is a hardware bug?

ntompson, 2022-04-06:
Hi @gorillica - no this is not a hardware bug. It is a problem with the current configuration for the ZW132 in zwavejs. I don't think it's hard to fix - it's "just" a matter of updating the config to pick up the extra messages. You can see what needs to be done in my [earlier comment](https://github.com/zwave-js/node-zwave-js/issues/4016#issuecomment-1007093112). 

Months ago when I raised this, I stupidly thought I might have time to do a PR myself to update the config file, which is why I was asking questions about using zwavejs2mqtt to validate the solution. But reality caught up with me and I haven't had time to do anything.

I can see that @AlCalzone has added this to the roadmap - hopefully he will get a chance to update the config at some point.

AlCalzone, 2022-04-07:
@ntompson @gorillica could one of you test the config file from the linked PR?
The general description for this process [is here](https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files?id=using-custom-device-configuration-files),
you need to create the following file structure:
- [This file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/b63e8a7aa01a2687bed41bbca01caaeaf6404ad4/packages/config/config/devices/0x0086/templates/aeotec_template.json) and [this file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/master/packages/config/config/devices/templates/master_template.json) in a folder called `templates`
- [This file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/b63e8a7aa01a2687bed41bbca01caaeaf6404ad4/packages/config/config/devices/0x0086/zw132.json) in the config root

gorillica, 2022-04-08:
Thanks @AlCalzone for your prompt action! (And, indeed for everything you've contributed on this project.)

I've just dropped in the new custom configs. I can see the additional config parameters in Z2M for the report types for the individual channels. They were all already enabled for group 1, presumably because I'd put in the custom config parameter a few days ago.

I also tried increasing the Automatic Reporting Interval to my preferred ~30 minute time from the 90s interval I set previously. I found that, as before, the individual channels don't update as the load changes, but the report for the whole device (is that channel 0?) does update as the load changes.

I have minimum power change (parameter 91) set to 2 watts and minimum percentage power change (parameter 92) set to 1%.

It was this discrepancy between the reports for the whole device vs. the individual channels that led me to wonder whether there's a hardware bug: surely the individual channels should report according to the setting for parameters 91 & 92 as the whole device does?
[Driver Log after installing custom config: the node I'm experimenting with is Node 39](https://github.com/zwave-js/node-zwave-js/files/8447625/driver_log_custom_config.txt)

I see that when I refresh the Meter CC using the button in Z2M, the individual channels’ meters are updated.

Thanks again for all your help!

AlCalzone, 2022-04-08:
I'm a bit confused that it prints
```
[Node 039] Embedded device config loaded
```
this should say "User-provided". But if the config params show up, you seem to have changed the correct one.

Anyways, it could be that the lifeline association is configured incorrectly. Can you show me how the association groups look like for this node?

gorillica, 2022-04-08:
[image]

AlCalzone, 2022-04-08:
Ok try changing that to Target Endpoint **0**. If this does the trick, I need to figure out why it doesn't happen automatically. I'm going to need a log of a re-interview then.

gorillica, 2022-04-08:
Hmm. When I tried that just now, the new Association doesn’t take. I deleted the existing Association, then tried adding as below, but it doesn’t then appear, even after clicking Refresh. But when I tried without specifying a target endpoint, it added the Association (back) successfully.

[image]

AlCalzone, 2022-04-08:
Okay, maybe Multichannel associations aren't supported.
Can you try adding the existing association (target endpoint None) to Endpoint 1 and 2 aswell?

gorillica, 2022-04-08:
When I try to add to both Endpoint 1 and 2, with no target node, it again doesn't take.

[image]

The options for Group when adding the association to endpoints 1 or 2 are `Key 1/2 via Lifeline` or `Button 1/2`.

AlCalzone, 2022-04-08:
Hmm, that's probably shared with the root endpoint then. Looks like I'm going to need that re-interview log after all.

gorillica, 2022-04-08:
No problem: thanks! Hope I've included the whole interview here.

[interview_zw132.txt](https://github.com/zwave-js/node-zwave-js/files/8451408/interview_zw132.txt)

AlCalzone, 2022-04-08:
Ok it looks like the associations are configured as expected. The device does support Multi Channel Associations, but we've explicitly configured it to use a normal association for the Lifeline. Based on previous experiments, this device requires doing this for reporting properly. 

Apparently it doesn't though. Could you make another driver log where you try to set target endpoint 0 on the existing association again? The endpoints don't have a separate lifeline, so its expected that those won't accept the command.

gorillica, 2022-04-08:
Ok, I restarted the logging and immediately went to try adding that association with target endpoint 0. For good measure, I then clicked refresh to call `getAssociations`. Hope this helps.
[Here’s the log](https://github.com/zwave-js/node-zwave-js/files/8455447/adding-association.txt)

AlCalzone, 2022-04-12:
Sorry, took me a while to get back to you. I can't see anything relevant in the log. Looks like its filtered somehow or only includes half the log.

AlCalzone, 2022-04-14:
I'm going to merge the corresponding PR. Please let me know if this issue is resolved after upgrading to the upcoming release 9.0.2.

closed 2022-04-14 by AlCalzone

gorillica, 2022-04-17:
Thanks, @AlCalzone. I did have my logs set to report only for that node (39), but it's weird that we weren't getting anything when I tried to adjust the groups on that very node.

I've found that if I set the device to notify by `Hail CC` (parameter 80) rather than `Basic CC Set`, the individual channel power report is now happening with channel 0! So that's great. I had thought that `Hail` was now deprecated in favour of `Basic`—is that right?

gorillica, 2022-04-19:
> That's not an individual report. `Hail CC` only notifies the driver that there is something new - as a result, the driver re-queries **ALL** potentially changed application CC values. In your case the 12 (!) energy meter values. So in total 1 (Hail) + 2*12 (GET and REPORT) = 25 messages - an easy way to clog up your network.

Yeesh. That's what I've been trying to avoid in my efforts to make the whole network as efficient as possible.

> > I can get you a log for trying again without the log being filtered if that's any help.
> 
> Yes, please

OK, [here's the file](https://github.com/zwave-js/node-zwave-js/files/8510154/driver_log_all_nodes_group_add.txt). This is the log set to `debug`, with no filter on individual nodes. I restarted the instance by clicking ‘Save’ in the Z2M settings, and waited for things to begin to stabilize on the log before then adding the Association Group. I think that I tried to add at about `2022-04-19T05:35:09.208Z` in the log file. It doesn't look that helpful to me, but perhaps everything is in the serial commands!

AlCalzone, 2022-04-20:
Hmm I don't even see the word `Association` appear in there 🥲

gorillica, 2022-04-22:
Yeah, exactly. I just tried with the log level set to `Silly`, and still don't get anything helpful. The Z2M log, meanwhile, points out that multi-channel associations aren't supported. Just as you observed previously!

```
2022-04-22 11:02:53.147 DEBUG SOCKET: Event ZWAVE_API emitted to fw9SHDHWGNmv0rY3AAAD
2022-04-22 11:02:53.152 INFO ZWAVE: Calling api addAssociations with args: [
  { nodeId: 39, endpoint: 0 },
  1,
  [ { nodeId: 1, endpoint: 0 }, [length]: 1 ],
  [length]: 3
]
2022-04-22 11:02:53.154 INFO ZWAVE: Assocaitions: Adding Node 1 to Group 1 of Node 39
2022-04-22 11:02:53.160 WARN ZWAVE: Error while adding associations to Node 39: Node 39, group 1 does not support multi channel associations! (ZW0302)
2022-04-22 11:02:53.162 INFO ZWAVE: Success zwave api call addAssociations undefined
2022-04-22 11:02:54.147 DEBUG SOCKET: Event ZWAVE_API emitted to fw9SHDHWGNmv0rY3AAAD
```

AlCalzone, 2022-04-22:
Ok before we spend any more time guessing in the dark here, please contact Aeotec support about this. They should be able to tell you how the device needs to be configured (Associations and config parameters) so the meter reporting works on its own.

gorillica, 2022-04-26:
The response I got was:

>Unfortunately, it is not possible to selectively trigger the measurement data of the individual channels. 
>
>The configuration of the individual channels only works via parameters 101 and 111.

After I asked, I was told that they have no intention/plan of adding this feature in a future firmware update. That's a shame. I think the Fibaro double switch (FGS-222) reports power usage live, but I've invested in more of the Aeotec devices! I've realized I can work around it, though.

AlCalzone, 2022-04-26:
I don't quite understand that answer...

I thought params 101-103 are meant to say **which measurements are sent** (which includes the individual channels) and params 111-113 control **how often** they are sent.

Isn't your issue that no matter what you configure, you don't get the individual channels' measurements? I mean why even have these options if they aren't supposed to do anything?!

[image]

gorillica, 2022-04-27:
The point is that I do get reports on individual channels, but only in the periodic reports, which are sent every _x_ seconds, depending on the setting of parameters 111–113.

What I was hoping for is to set that reporting period to a high value (to minimize unnecessary reports on the network), and instead to have the device send out reports only when the power draw changes by more than a small threshold, as specified in parameters 91 & 92.

With the configuration I've arrived at now, those threshold event reports are sent out for channel 0 (i.e. the total power draw of both loads), but not for the individual channels. The individual channels’ reports are only updated either (1) with the automatic reporting period, or (2) if I refresh the metering CCs, or (3) if I set the notification type to Hail CC (parameter 80) which, as you pointed out, isn't an efficient use of network time.

I think the point of parameters 101–103 is to set the reports that are sent with the automatic periodic reports, and not the threshold event reports. As to why they can't (or won’t) implement this for the threshold reports, I have no idea!

It's frustrating, but I can work around it.

AlCalzone, 2022-04-27:
Ohhh, ok got it.

PR #4447 (2022-04-14): fix(config): correct reporting groups for Aeotec ZW132 Dual Nano Switch
fixes: #4016