zwave-js/zwave-js #4017: HS-WX300 not configured properly

HerveB, 2022-01-03:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I am using Home Assistant with Z-Wave JS to 8.9.2.

Adding new HomeSeer HS-WX300 dimmer-switch doesn't create a device with proper configuration: no trigger are exposed in automations for the device. 

I was able to successfully add other HS-WX300 dimmer-switches with earlier version of Z-Wave JS and configure automations involving double keypressed triggers.

### Device information

Manufacturer: HomeSeer
Model name: HS-WX300 with firmware 1.12
Highest Security: S2 Authenticated
Z-Wave Plus: Version 2
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: Z-Wave JS to 8.9.2

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Two other HS-WX300 devices were added using older versions of Z-Wave JS. Both have the same firmware 1.12 and security configuration. Both are working properly, exposing various triggers in automation such as 2x keypressed.

### Attach Driver Logfile

No log file.

AlCalzone, 2022-01-03:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly` and attach it here as a file.
While the log is running, re-interview the problematic device.

HerveB, 2022-01-03:
Just added log file my initial post.

AlCalzone, 2022-01-03:
According to that log, the device has two scenes with support for these events:
· KeyPressed
· KeyReleased
· KeyHeldDown
· KeyPressed2x
· KeyPressed3x
· KeyPressed4x
· KeyPressed5x

Can you capture another log where you operate the device (and would expect to see events)?

HerveB, 2022-01-06:
The latest update for Home Assistant (2021.12.8) fixed the problem. I suspect it was an issue with the interface to configure automations. Everything is fine now.
Thanks!

closed 2022-01-07 by AlCalzone