zwave-js/zwave-js #4018: DSB45 water sensor always reports wet

mhendu, 2022-01-03:
I managed to get this device correctly interviewed after some time but it always reports wet, even when it's clearly not. I found some other articles online indicating that I needed to change the configuration so I updated to the following (non-default) settings:

Wake device for 10 minutes after power on - enable
On Trigger: Send Battery Report - enable
On Trigger: Send Binary Sensor Report - enable
On Trigger: Send Alarm - enable

I'm using the sensor to indicate if a tank of water is empty, so I also changed Invert Basic Set on Open/Close Event to Set on when closed.

Changing these settings, or really any settings, seems to have no impact on what the sensor reports. It's been in the tank for a year and I haven't tested it in a while, so it could just be that the sensor ends have corroded, but even when I changed Invert Binary Report Value to Report on when closed it didn't report dry, which seems weird. The sensor was working for a while on another system using the legacy Z-Wave integration.

Logs attached - any help would be appreciated.
[zwavejs2mqtt_2022-01-01.log](https://github.com/zwave-js/zwavejs2mqtt/files/7798312/zwavejs2mqtt_2022-01-01.log)
[zwavejs_2022-01-01.log](https://github.com/zwave-js/zwavejs2mqtt/files/7798313/zwavejs_2022-01-01.log)

AlCalzone, 2022-01-03:
And ideally leave it running for 5-10 minutes after the report. Not sure how quickly the device is supposed to reset the status.

mhendu, 2022-01-06:
Try this - I had it out of the tank for the last couple of days, started logging a few minutes ago, reset the controller, then after it loaded everything I put the sensor back in the water. I could see it immediately reported back that it was 'wet' but it has also been reporting 'wet' this whole time even though it hasn't been in the water until now for a couple of days. If I'm reading the logs correctly it does seem like it's reporting a state change, though.
[zwavejs2mqtt_2022-01-06.log](https://github.com/zwave-js/node-zwave-js/files/7819872/zwavejs2mqtt_2022-01-06.log)
[zwavejs_2022-01-06.log](https://github.com/zwave-js/node-zwave-js/files/7819873/zwavejs_2022-01-06.log)

mhendu, 2022-01-06:
Oh, I also just pulled it out of the water again, it reported back pretty much immediately and this is what I saw in the logs:

2022-01-06T06:25:42.184Z CNTRLR   [Node 007] [Binary Sensor] Any: metadata updated                  [Endpoint 0]
2022-01-06 00:25:42.185 INFO ZWAVE: Node 7: metadata updated: 48-0-Any
2022-01-06T06:25:42.186Z CNTRLR   [Node 007] [~] [Binary Sensor] Any: true => false                 [Endpoint 0]
2022-01-06 00:25:42.191 INFO ZWAVE: Node 7: value updated: 48-0-Any true => false
2022-01-06T06:25:42.193Z SERIAL » [ACK]                                                                   (0x06)
2022-01-06T06:25:42.195Z DRIVER « [Node 007] [REQ] [ApplicationCommand]
                                  └─[BinarySensorCCReport]
                                      type:  Any
                                      value: false
2022-01-06T06:25:42.233Z SERIAL « 0x010b0004000703800363bf00ab                                        (13 bytes)
2022-01-06T06:25:42.236Z CNTRLR   [Node 007] [~] [Battery] level: 86 => 99                          [Endpoint 0]
2022-01-06 00:25:42.237 DEBUG ZWAVE: Node 7: mapping value(s) of property 'level' (level) from CC 128 (Battery) to node property 'minBatteryLevel
2022-01-06 00:25:42.238 DEBUG ZWAVE: Node 7: mapping value(s) of property 'level' (level) from CC 128 (Battery) to node property 'batteryLevels
2022-01-06 00:25:42.239 INFO ZWAVE: Node 7: value updated: 128-0-level 86 => 99
2022-01-06T06:25:42.243Z CNTRLR   [Node 007] [~] [Battery] isLow: false => false                    [Endpoint 0]
2022-01-06 00:25:42.244 INFO ZWAVE: Node 7: value updated: 128-0-isLow false => false
2022-01-06T06:25:42.249Z SERIAL » [ACK]                                                                   (0x06)
2022-01-06T06:25:42.253Z DRIVER « [Node 007] [REQ] [ApplicationCommand]
                                  └─[BatteryCCReport]
                                      level:  99
                                      is low: false
2022-01-06T06:25:42.285Z SERIAL « 0x0112000400070a7105000000ff05040000bf00d1                          (20 bytes)
2022-01-06T06:25:42.289Z CNTRLR   [Node 007] [~] [Notification] notificationMode: "pull" [Endpoint 0] [internal]
                                   => "pull"
2022-01-06T06:25:42.290Z SERIAL » [ACK]                                                                   (0x06)
2022-01-06T06:25:42.292Z DRIVER « [Node 007] [REQ] [ApplicationCommand]
                                  └─[NotificationCCReport]
                                      notification type:   Water Alarm
                                      notification status: 255
                                      notification event:  Water level dropped
2022-01-06 00:25:42.302 INFO ZWAVE: Node 7 CC Notification {
  type: 5,
  event: 4,
  label: 'Water Alarm',
  eventLabel: 'Water level dropped',
  parameters: undefined
}

kpine, 2022-01-06:
Are you using HA? You will have a binary sensor that is disabled. Enable it, and it will correctly report the Wet and Dry status, via On and Off.

From the logs in the comment above, this device seems to behave like an [Ecolink leak detector](https://github.com/home-assistant/core/issues/51691#issuecomment-858300763) I ran into previously. It reports a wet condition with the `Water leak detected` notification, but instead of resetting it (`Idle (0)` notification), it sends the unrelated `Water level dropped` notification. The latter does not clear the former.

AlCalzone, 2022-01-06:
> It reports a wet condition with the `Water leak detected` notification, but instead of resetting it (`Idle (0)` notification), it sends the unrelated `Water level dropped` notification. The latter does not clear the former.

That could actually be the problem. We don't correlate all possible notifications yet - see the discussion around lock jammed etc.

mhendu, 2022-01-06:
Ah, I see it there in Home Assistant. binary_sensor.water_sensor_any reports On when wet and Off when dry. I'll use this instead. Thanks!

AlCalzone, 2022-01-06:
Oh, yeah that's even better :)

AlCalzone, 2022-01-06:
Guess we can close this then?

closed 2022-01-06 by mhendu