zwave-js/zwave-js #4030: Unknown manufacturer, 0xXXXX, Unknown product

JasFos, 2022-01-03:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Not sure what is causing it

What do you observe?
Devices keep going to "Unknown manufacturer, 0xXXXX, Unknown product 0xXXXX" ?
This is the third time it's happened, very annoying. Everything in HA fails when this happens.
I end up having to remove them and add them again to get them to work.
zWave stick info HUSBZB-1 (Silicon Labs Z-Wave 500 Series Controller ACC-UZB3)

What did you expect to happen?
Once added I expect them to stay remembered and not go to unknown

Steps to reproduce the behavior:
  1. Add zWave devices
  2. Wait for them to turn up as unknown (not specific time)
  3. I do reboot my HA device (NUC) daily
  4. Maybe rebooting will make it repro

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

snap install

store-url: https://snapcraft.io/zwavejs2mqtt
installed:          v6.2.0            (315) 137MB

snap-id:      k26XpW9UYHL1HNh20isi5MHoazqHZ8br
tracking:     latest/edge
refresh-date: yesterday at 05:25 PST

zwavejs2mqtt: 6.2.0.fa6b2ab
zwave-js: 8.9.3-0-pr-4011-940427a
home id: 4276191763
home hex: 0xfee18213

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Yes, it works for a while before this issue appears

### Attach Driver Logfile

[zwavejs_2022-01-03.log](https://github.com/zwave-js/node-zwave-js/files/7804381/zwavejs_2022-01-03.log)
[image]

jmgiaever, 2022-01-03:
Hi. Please note that the edge channel uses experimental versions. You should use the channel `latest/candidate` for rolling releases and `latest/stable` for stable release. The latter one mainly only releases the final release for each major version, while candidate for each minor and/or patch. 

The edge branch you're on got updated yesterday with a test of a certain branch of zwjs, to test a feature that was specific for a device. It can be updated x amount of times in a row and causes stability issues such as this. 

Leaving your network running should re-interview the nodes, but that can take time.

jmgiaever, 2022-01-03:
Also sometimes i helps to just restart if the interview process finishes with an unknown device. 

```
sudo snap stop zwavejs2mqtt
sudo reboot
```

If it doesn't help. Re-interview node by node, and make sure each finish before starting a new interview.

JasFos, 2022-01-03:
I saw the same behavior on the latest/stable release, which is why I moved the edge release hoping it might resolve or fix my issue

I've also tried rebooting...doesn't fix it

It's been sitting for most of the day and hasn't re-found, re-identified the devices correctly

jmgiaever, 2022-01-04:
Have you tried to solve it for just one device? Interviewing your whole network will cause a very high load on your network.

JasFos, 2022-01-04:
I've tried re-interviewing individual ones multiple times
I've tried just about everything multiple times...not sure what is causing this
.
It will be fine for weeks and then one day it just does this (nothing changed)

JasFos, 2022-01-04:
It's only happening on these are newer Eaton devices

https://devices.zwave-js.io/?jumpTo=0x001a:0x4449:0x0501:0.0
https://devices.zwave-js.io/?jumpTo=0x001a:0x534c:0x0503:0.0

AlCalzone, 2022-01-04:
Node 22 is included securely:
```
2022-01-03T22:41:42.380Z CNTRLR « [Node 022] node info received
                                  supported CCs:
                                  · Z-Wave Plus Info
                                  · Transport Service
                                  · Security
                                  · Security 2
                                  controlled CCs:
```
and the driver thinks it is NOT included with Security S2, so it only checks S0 but that yields an empty list of CCs:
```
2022-01-03T22:41:54.843Z DRIVER « [Node 022] [REQ] [ApplicationCommand]
                                  └─[SecurityCCCommandEncapsulation]
                                    │ sequenced: false
                                    └─[SecurityCCCommandsSupportedReport]
                                        reportsToFollow: 0
                                        supportedCCs:
                                        controlledCCs:
```
Its the same with Node 21.

I'm not sure how that happened, but somehow the driver must have not been able to figure out which S2 classes the nodes support... Incorrect keys, timeouts during initial interview, broken cache during upgrade, ...

I don't think you're able to reset this information via the UI yet. You can [try this](https://github.com/zwave-js/node-zwave-js/issues/3935#issuecomment-1004570860), but the procedure to get to the cache file is going to be a little different for Snap. @jmgiaever?
Make sure to only remove this section for the problematic nodes.

JasFos, 2022-01-04:
Thank you for looking into this and the "try this" info :-)

I went ahead and removed and re-added the devices
I added them this time with S2_Authenticated only...we'll see how it goes.
I'm also not going to reboot my system for awhile.

If I have problems again I will look at the "security classes"

This is how it looks when it's working
                "dsk": "47256-00133-11506-53356-41327-05514-24302-42460",
                "securityClasses": [
                    "S2_Authenticated"
                ],

Here's how they look after being re-added
[image]

Thanks again

jmgiaever, 2022-01-04:
Z2M has this "reset security classes" option when you want to re-interview anything. Guess that's doing the same as you linked to @AlCalzone ?

JasFos, 2022-01-04:
Not sure when/why the security class is resetting or changing to the "S2_Authenticated": false.
It's NOT because of a re-interview.

The only thing that might have caused it was a reboot, I reboot the machine my zwavejs2mqtt/HA is on daily
But it rebooted for weeks before the "unknown" or security class change showed up

AlCalzone, 2022-01-04:
> It's NOT because of a re-interview.

Yeah, but a re-interview with that option enabled should fix the problem without excluding and including again. I expect this to go away once I've reworked how the JSON cache file gets written.

JasFos, 2022-01-05:
That would be great if you can fix it...it sounds like you know what the problem might be
Do you need anything more from me at the moment?

AlCalzone, 2022-01-05:
Nope, thanks!

JasFos, 2022-01-07:
Once you've reworked this...will I be able to reboot my system without loosing my devices?

jmgiaever, 2022-01-07:
You should be able to already now, just make sure to shut down z2m first. 

```
sudo snap stop zwavejs2mqtt
```

closed 2022-01-25 by zwave-js-assistant[bot]