zwave-js/zwave-js #4032: [bug] Qubino never finishes his interview

YOYONL89, 2022-01-04:
**Checklist:**

- [x] I am **not** using Home Assistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem is **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

Zwavejs2Mqtt version: 6.2.0
Z-Wave JS version: 8.9.1

### Describe the bug

Node 23 a Flush 1 Relay never pulls it through his interview.
This is now for a while and though pushing its button triggers a response every now and then it seems it gets stuck.
The node has been excluded and included a few times now to try to fix the problem but it just wont do.
For a long time i hade made peace with it but i think its interfering with the network in general which causes performance issues.

### To Reproduce

Steps to reproduce the behavior:

Reinterview or reinclude the device.

### Expected behavior

The interview completing in order to see if the network stability improves.

### Additional context

Maybe not part of the same problem but a other 1 flush relay reports as unknown on Manufacturer, Product and Product code.
The interview of that component succeeds though, same goes for a 2 flush relay.
Last but not least the generation of the network graph has never been working for me?
Node 23 is next to Node 5, is almost in line of sight of  node 19 and node 2.
[zwavejs2mqtt_2022-01-03 - General Debug.log](https://github.com/zwave-js/zwavejs2mqtt/files/7804086/zwavejs2mqtt_2022-01-03.-.General.Debug.log)
[zwavejs2mqtt_2022-01-03 - General Silly.log](https://github.com/zwave-js/zwavejs2mqtt/files/7804089/zwavejs2mqtt_2022-01-03.-.General.Silly.log)
[zwavejs_2022-01-03 - Zwave Debug.log](https://github.com/zwave-js/zwavejs2mqtt/files/7804119/zwavejs_2022-01-03.-.Zwave.Debug.log)

AlCalzone, 2022-01-04:
It looks like your mesh is in a terrible state. Many commands take seconds to reach the target
```
                                    transmit status:        OK, took 4410 ms
                                    repeater node IDs:      11
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 9.6 kbit/s
```
and the responses don't come at all sometimes.

Check out and apply the first part here:
https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting

YOYONL89, 2022-01-05:
Hi Alcazone,

Oho that doesn't sound good and is strange because the thing is in the same
place for a time longer then the time Im encountering the problems.
Maybe something shifted as living near a railroad track makes things
tremble now and then.

I need to check this on the physical machine but it's not easy to reach.
I will take a look on it this Friday when I have more time.

Thanks in advance and I'll keep you updated.

Cheers,

YOYONL89, 2022-01-07:
Hi AlCalzone,

Like i promissed i went to check on the machine and i didn't see anything out of the ordinary.
The only thing i can think off its because the stick is in a USB hub and there for has some issues.
The strange thing is it has been working for quite some time without any problems.

The alternative i have at the moment is to put the stick directly on the back of the machine but that also isnt recommended.

AlCalzone, 2022-01-08:
> The strange thing is it has been working for quite some time without any problems.

You might just not have noticed. Or another USB device is now interfering that didn't previously.
Both your options are not ideal, you should really get a USB extension cable.

YOYONL89, 2022-01-08:
Hmm ok well let's get this thing sorted for once and for all then.
What is the recommendation of the extension cable?
Does it have to be long? Is it depending on the location the machine is in?

The situation is the following:
The machine running the Dockers is a shallow 1U rack which is located in a closet. 
Directly in the neighbourhood are some power lines Pasing the closet from the circuitbreakers of the house to different places.

AlCalzone, 2022-01-08:
I can't give an exact recommendation, for some users 30cm is enough, others use longer ones. I'd say make it long enough so you can move it far away from the power lines and USB ports (e.g. the edge of the closet) and still experiment with the location a bit if that particular location still has interference.

YOYONL89, 2022-01-12:
Hi AlCalzone,

I ordered a 1 meter long extension cable and i have to see it improved the network throughput around a dosen times.
Im now able to get a topoloy every time i ask for it.
Also no timeouts on the statistics board.

Unfortunally the device wich started our little hunt still is acting up and not finishing its interview.
Here is a new logfile with the improvements made.
I did a heal of the network twice and also reinterviewd all the nodes again.
This did resolve the fact of some slaves not displaying there manufacturer data.

[zwavejs_2022-01-12 - Debug.log](https://github.com/zwave-js/node-zwave-js/files/7856916/zwavejs_2022-01-12.-.Debug.log)

AlCalzone, 2022-01-12:
The device is a Qubino device and it's failing to answer to the endpoint query, which makes me suspicious.

Qubino devices have the ability to enable/disable endpoints via configuration parameters. However, the changes only show up **after excluding and re-including** the device. This is written very prominently in the manual, but I'm guessing many users don't read them if they have the nice config parameters in the UI.

Is it possible this is the case here?

YOYONL89, 2022-01-12:
If those parameters have been altered then that happend without me knowing about it as I never changed something without the UI.

What I did when it was still working was changing the behavior over S2 via the UI because there is a binary switch attached to it sending a pulse when the ventilation has to go in high gear.

Perhaps the solution could be to exclude/include the module again in order to get it back talking if that's what you're suggesting.

AlCalzone, 2022-01-12:
> Perhaps the solution could be to exclude/include the module again in order to get it back talking if that's what you're suggesting.

Try it

YOYONL89, 2022-01-21:
Hi AlCalzone,

Unfortunally i think excluding it made it worse.
Starting to think that something might be wrong with the qubino?
[zwavejs_2022-01-21.log](https://github.com/zwave-js/node-zwave-js/files/7916154/zwavejs_2022-01-21.log)

AlCalzone, 2022-01-23:
Still no connection to the device:
```
                                    transmit status:        NoAck, took 7810 ms
                                    routing attempts:       17
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    TX channel no.:         1
                                    route failed here:      15 -> 24
```
and next attempt:
```
                                    transmit status:        NoAck, took 8040 ms
                                    routing attempts:       19
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    TX channel no.:         1
                                    route failed here:      2 -> 24
```

The other sent messages to other devices seem to go through just fine. This leaves either the location of the node or the node itself.

YOYONL89, 2022-02-04:
Hmm i have been going back and fourth with it and i did get it working on one point.
But today i hade to update the container of ZwaveJS to match the Homeassistant container.
Things didnt get any better unfortunaly....

i'm now even missing a second unit as well having problems with my qubino again.
Is it still the mesh or is it something different this time?

[zwavejs_2022-02-04.log](https://github.com/zwave-js/node-zwave-js/files/8006098/zwavejs_2022-02-04.log)
[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/8006099/zwavejs_current.log)

AlCalzone, 2022-02-06:
I only see the part where the driver tries to reach Node 24, but unfortunately that just looks like previously, except it is now trying the direct range.

zwavejs2mqtt now has a health check tool included in the network map (click node, press check health). You could try using that to potentially get some more insight. Maybe another node has a bad connection too and is influencing this one.
Just note that there currently is a bug where you have to restart zwavejs2mqtt if the check seems to go on forever.

YOYONL89, 2022-02-08:
@AlCalzone What is the probability that node 24 is the old node which i removed.
Node 25 would be the new node and supposingly be the working one?

Why did node 24 return? and is it messing up node 25 because its back?

AlCalzone, 2022-02-08:
It could be, but your log is filtered somehow so I only see communication with Node 24. If that one has been removed, you need to tell the controller that it was removed using the "remove failed node" function.

YOYONL89, 2022-02-08:
Ok node 24 was confirmed a relic of the new node 25.
Still after removing node 24 and a network heal, node 25 now still doesn't want to.

If i understand the attached log correctly it reply's as expected?

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/8027326/zwavejs_current.log)

AlCalzone, 2022-02-08:
This time the device acknowledges the commands, but we're back to this:
https://github.com/zwave-js/node-zwave-js/issues/4032#issuecomment-1011449666

I think this is a case for the Qubino support.

closed 2022-02-26 by zwave-js-assistant[bot]