zwave-js/zwave-js #4038: Previous Home ID detected after NVM restore

kpine, 2022-01-05:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

NVM restore

What do you observe?

The NVM restore completes successfully, and the driver is destroyed automatically, according to the logs. The driver starts up again automatically, and when it does it still detects the old home ID (prior to the backup). After this there are a lot of failed controller commands, and the driver tries to re-interview all nodes (presumably because it loads the wrong cache file), with more commands failing. In Z2M, all the nodes are unknown and stuck in NodeInfo or ProtocolInfo. 

The system remains in this failed state, but after restarting zwavejs2mqtt, the home ID is detected properly (backed up home ID) and everything works again.

What did you expect to happen?

After the restore, everything to be working the same as before.

Steps to reproduce the behavior:
  1. Restore an NVM backup with the contents of another network
  2. Check logs for observed behavior
  3. Restart z2m, everything is OK.

This sounds similar to #3992, but there was no mention of a restart fixing things, so I've created this new issue.

### Device information

Controller source was UZB7, controller target was Zooz ZST10 700.

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [X] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.10.0
`zwavejs2mqtt` branch: master (bd8c933010ccd0e9c39d7392984c386ff25deaa7)

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

Here is the restore and restart. I cut out the debug info of the NVM restore itself, let me know if you need that contents and/or the NVM files and I can send privately.

[zwavejs_2022-01-04.log](https://github.com/zwave-js/node-zwave-js/files/7812542/zwavejs_2022-01-04.log)

There's something I noticed I thought I'd ask. Here's the restore and driver reload:

```
2022-01-05T02:17:53.613Z CNTRLR   NVM backup restored
2022-01-05T02:17:53.613Z CNTRLR   Turning RF on...
2022-01-05T02:17:53.619Z SERIAL » 0x0104001001ea                                                       (6 bytes)
2022-01-05T02:17:53.620Z DRIVER » [REQ] [SetRFReceiveMode]
                                    enabled: true
2022-01-05T02:17:53.625Z SERIAL « [ACK]                                                                   (0x06)
2022-01-05T02:17:53.627Z SERIAL « 0x0104011001eb                                                       (6 bytes)
2022-01-05T02:17:53.628Z SERIAL » [ACK]                                                                   (0x06)
2022-01-05T02:17:53.629Z DRIVER « [RES] [SetRFReceiveMode]
                                    success: true
2022-01-05T02:17:53.633Z CNTRLR   Restarting driver to activate restored NVM backup...
2022-01-05T02:17:53.640Z DRIVER   destroying driver instance...
2022-01-05T02:17:53.673Z DRIVER   driver instance destroyed
2022-01-05T02:17:53.706Z DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                                  ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                                    ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                                   ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                                  ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                                  ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
2022-01-05T02:17:53.707Z DRIVER   version 8.10.0
2022-01-05T02:17:53.707Z DRIVER   
2022-01-05T02:17:53.707Z DRIVER   starting driver...
2022-01-05T02:17:53.711Z DRIVER   opening serial port /dev/zwave
2022-01-05T02:17:53.757Z DRIVER   serial port opened
2022-01-05T02:17:53.758Z SERIAL » [NAK]                                                                   (0x15)
2022-01-05T02:17:53.764Z DRIVER   loading configuration...
2022-01-05T02:17:53.775Z CONFIG   Using external configuration dir /usr/src/app/store/.config-db
2022-01-05T02:17:53.776Z CONFIG   version 8.10.0
2022-01-05T02:17:54.953Z DRIVER   beginning interview...
2022-01-05T02:17:54.955Z DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                                  1 registered
2022-01-05T02:17:54.955Z DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                                  1 registered
2022-01-05T02:17:54.956Z DRIVER   added request handler for ReplaceFailedNode (0x63)...
                                  1 registered
2022-01-05T02:17:54.956Z CNTRLR   querying controller IDs...
2022-01-05T02:17:54.964Z SERIAL » 0x01030020dc                                                         (5 bytes)
2022-01-05T02:17:54.965Z DRIVER » [REQ] [GetControllerId]
2022-01-05T02:17:54.970Z SERIAL « [ACK]                                                                   (0x06)
2022-01-05T02:17:54.972Z SERIAL « 0x01080120c39f8d88018e                                              (10 bytes)
2022-01-05T02:17:54.973Z SERIAL » [ACK]                                                                   (0x06)
2022-01-05T02:17:54.975Z DRIVER « [RES] [GetControllerId]
                                    payload: 0xc39f8d8801
2022-01-05T02:17:54.980Z CNTRLR   received controller IDs:
                                    home ID:     0xc39f8d88
                                    own node ID: 1
```

I notice there's no soft-reset after the NVM restore and before reading the home ID (unless it wasn't logged). The restart does a soft-reset. Could it be that a soft-reset is required prior to destroying and starting the driver?

kpine, 2022-01-05:
Also, I saw this log about NVM format conversion:

```text
2022-01-05T02:17:06.553Z CNTRLR   Converting NVM to target format...
2022-01-05T02:17:06.565Z SERIAL » 0x0104002e00d5                                                       (6 bytes)
2022-01-05T02:17:06.566Z DRIVER » [REQ] [NVMOperations]
                                    command: Open
```

Was there actually any conversion done, since this was from a 7.16 controller to another 7.16 controller?

AlCalzone, 2022-01-05:
Could you cross-check if converting the NVM manually using
https://github.com/zwave-js/node-zwave-js/tree/master/packages/nvmedit#convert-one-nvm-to-be-compatible-with-another-one
and then converting the result to JSON with
https://github.com/zwave-js/node-zwave-js/tree/master/packages/nvmedit#convert-binary-nvm-file-to-json
yields the same incorrect ID?

This looks like a similar issue to https://github.com/zwave-js/node-zwave-js/issues/3992 - maybe the forced soft reset is all that's needed.
My intention was to forget if a soft-reset is supported and do it on restart, but the application configuration could be interfering here.

AlCalzone, 2022-01-05:
> Was there actually any conversion done, since this was from a 7.16 controller to another 7.16 controller?

The migration way is always NVM -> JSON -> NVM. The current routine does not yet support "continuing" the 700 series memory ring buffer and initializes a "fresh" one without leftover data.

Out of curiosity: Can you send me an NVM backup of the target controller **before and after** attempting the migration?

kpine, 2022-01-05:
> I notice there's no soft-reset after the NVM restore and before reading the home ID (unless it wasn't logged). The restart does a soft-reset. Could it be that a soft-reset is required prior to destroying and starting the driver?

Minor correction to this comment. After the driver auto-restarts, it doesn't do a soft-reset prior to reading the home ID, but after getting the capabilities it does do the usual soft-reset. Does zwave-js use the home ID for anything besides logging and locating the cache file? I wonder if the timeouts were just because it was trying to re-interview everything at once, and maybe it would have recovered had I let it finish and re-interview the battery devices? I don't normally get controller timeouts though, but these did happen after the soft-reset.

AlCalzone, 2022-01-05:
> Does zwave-js use the home ID for anything besides logging and locating the cache file?

No, but the Home ID is part of the RF frames the controller sends. With an incorrect one, the nodes never pick up the messages that were meant for them.

Looking again at your log, it looks like the correct Home ID is actually used. You're getting messages from the nodes and the outgoing messages succeed, except a few. That wasn't the case while restoring.

I don't like the softreset after restart though. I think it leads to confusion with the home ID and apparently the wrong cache (nothing should need to be re-interviewed).
I'll change it to force a soft reset after the restore and then restart. If anyone restores a backup onto a UZB-1, they'll have to deal with re-plugging it anyways.

kpine, 2022-01-06:
Testing the linked PR, but encountered something very strange. Before I even got to the restore, I did a hardware reset. The home ID changed after the reset, but somehow it found all the nodes? Then I restarted z2m and it reverted back to the original home ID.

[zwavejs_2022-01-05.log](https://github.com/zwave-js/node-zwave-js/files/7818766/zwavejs_2022-01-05.log)

kpine, 2022-01-06:
Second try doing a hard reset worked, that was strange. Makes me wonder if the original issue was a fluke as well. Anyways, the restore worked without issue this time. No re-interview was performed, not sure if you were expecting that or not.

On another note, I still see broadcast messages from devices after doing an NVM backup, as mentioned in [here](https://github.com/zwave-js/node-zwave-js/discussions/3706). Different device, but same stick.

[zwavejs_2022-01-05.log](https://github.com/zwave-js/node-zwave-js/files/7819844/zwavejs_2022-01-05.log)

[nvm.zip](https://github.com/zwave-js/node-zwave-js/files/7819851/nvm.zip)

AlCalzone, 2022-01-06:
> No re-interview was performed, not sure if you were expecting that or not.

I did, but I also see why. When you factory reset, the home ID changes, so the information is reset in the wrong cache file. After restarting the new home ID is active and the information from the other cache file is still there.
I'll have to save the request to reset all info somewhere outside the homeid specific cache files, but thats something for another day.

> I still see broadcast messages from devices after doing an NVM backup

Most likely this is because the radio is turned off for the backup. Then the nodes don't get the expected ACKs and resort to broadcasts as the last measure.

closed 2022-01-06 by AlCalzone