zwave-js/zwave-js #4047: HS-WX300 Switch/Dimmer Parameter 3 seems to not work correctly

pdobrien3, 2022-01-05:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Unknown

What do you observe? 

Parameter 3, which controls the bottom LED operation, doesn't seem to work correctly.  Here is what happens on my HS-WX300 that is setup as a switch 

**on when load is on**
	load is off == LED 1 is on
	load is on  == All 7 LEDs are on
**on when load is off**
	load is off == All 7 LEDs are off
	load is on  == All 7 LEDs are on

What did you expect to happen?

**on when load is on**
	load is off == LED 1 is off
	load is on  == LED 1 is on
**on when load is off**
	load is off == LED 1 is on
	load is on  == LED 1 is off

Steps to reproduce the behavior:
  1. Go to config parameter 3 and set it to desired setting, make sure config parameter 13 is disabled and all LEDs are switched to off.
  2. Click on/off the switch either manually or remotely
  3. Look at LED 1
 

### Device information

Manufacturer:  Homeseer
Model name: HS-WX300
Node ID in your network:  120

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

zwavejs2mqtt: 6.2.0
zwave-js: 8.9.1

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

2022-01-05 23:13:51.445 INFO ZWAVE-SERVER: ZwaveJS server listening on port 3000

AlCalzone, 2022-01-06:
Honestly, this sounds like a firmware bug in the device. The driver only sets the parameter to a value and has no control over how the device behaves.
Its not even consistent in any way that we could say the description is inverted or something.

pdobrien3, 2022-01-06:
I get your point. Although it would seem like if it was a firmware issue, there would be several issues found through a google search.  The only thing I can find is two posts related to HAss.  All that said, I have emailed HS Support although I don’t expect any resolution. They always blame it on HAss.

AlCalzone, 2022-01-06:
Don't give in to their excuses then.

The configuration CC is standardized and if you set a parameter with the correct value and size as stated in the manual, it simply cannot be the application's fault. That's not how Z-Wave works.

All the application does is tell the device to change a config parameter. It cannot control the LEDs (or anything really) on the device itself. The device firmware is in charge of the hardware.

pdobrien3, 2022-01-07:
Thank you for your last comment. I used it in my communications with HS. I seem to have gotten some traction. 

From Homeseer:

It looks like this is an error in documentation as parameter 3 explicitly says both that it will control the bottom LED and that it will control all of them. I will reach out to our engineers about this so we can get the doc updated and perhaps get a feature request in that allows for both options.

pdobrien3, 2022-01-11:
I gave homeseer support this link to report their findings.

rrauenza, 2024-04-11:
I'm confused because I don't understand how the integration works at the code level, but if the docs are wrong, can the logic not be just swapped in hass code?

Also, latest docs say this - https://docs.homeseer.com/products/hs-wx300-r2-user-guide

Param #3:

0 = Bottom LED ON if load is OFF
1 = Bottom LED OFF if load is OFF

...is this still the wrong docs?  Where in the codebase does this logic live?   I don't mind doing a PR ... but my searching has failed me so far.

Oh - this is in zwavejs not hass core.  I'll search this repo next.

rrauenza, 2024-04-11:
Looks like their docs are still wrong unless the code has already been fixed:

```
"led_indicator_two_options": {
		"label": "LED Indicator",
		"valueSize": 1,
		"defaultValue": 1,
		"allowManualEntry": false,
		"options": [
			{
				"label": "On when load is off",
				"value": 0
			},
			{
				"label": "On when load is on",
				"value": 1
			}
		]
	},
```

AlCalzone, 2024-04-12:
@rrauenza I'm fine updating our config files to match the actual device behavior.

rrauenza, 2024-04-12:
Thanks ... there is a subtle language issue in the existing templates:

```
"led_indicator_two_options": {
                "label": "LED Indicator",
                "valueSize": 1,
                "defaultValue": 1,
                "allowManualEntry": false,
                "options": [
                        {
                                "label": "On when load is off",
                                "value": 0
                        },
                        {
                                "label": "On when load is on",
                                "value": 1
                        }
                ]
        },
        "led_indicator_two_options_inverted": {
                "label": "LED Indicator",
                "valueSize": 1,
                "defaultValue": 1,
                "allowManualEntry": false,
                "options": [
                        {
                                "label": "On when load is on",
                                "value": 0
                        },
                        {
                                "label": "On when load is off",
                                "value": 1
                        }
                ]
        },
```

Neither of these apply. The indicator is configurable to be either on or off when the load is **off**.   This setting doesn't affect anything about the indicator and the load being on.

```
0 = Bottom LED ON if load is OFF
1 = Bottom LED OFF if load is OFF
``` 

I don't know if this subtle distinction is only for this particular device ... or if the template is subtly wrong for other devices as well.

I was going to proceed by just not referring to the template and fixing it locally in the wx300 json, unless you'd rather I just point to the inverted template.

AlCalzone, 2024-04-12:
> This setting doesn't affect anything about the indicator and the load being on.

I see! Usually the LED indicator either follows the load state (on when on, off when off) or inverts it (on when off, off when on).
Maybe this was the assumption when the templates were written.

> I don't know if this subtle distinction is only for this particular device ... or if the template is subtly wrong for other devices as well.

Actually let me reach out to my contact at HomeSeer - the situation has changed a bit since this issue was created. They should be able to clarify.

AlCalzone, 2024-04-15:
Response from HomeSeer:
> parameter 3 is designed to control the state of the bottom LED when the load is OFF and the switch is in NORMAL mode. If the load is ON and the switch is in NORMAL mode, the bottom LED will always be ON. This is how the WD200 worked as well.

Seeing that the `master_template` is currently referenced, this HomeSeer specific parameter should probably go into their template file and get referenced from HS-WX300 and -WD200.

closed 2024-05-24 by AlCalzone

pdobrien3, 2024-05-30:
After re-interviewing, device information is lost:

`Unknown manufacturer 0xXXXX | 0xXXXX | Unknown product 0xXXXX
`

AlCalzone, 2024-05-30:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of the re-interview, loglevel `debug` and attach it here as a file (drag & drop into the text field).

pdobrien3, 2024-06-06:
[zwavejs_2024-06-06.log](https://github.com/user-attachments/files/15613124/zwavejs_2024-06-06.log)

AlCalzone, 2024-06-06:
This looks like a device bug. Here's the relevant log snippet with comments:
```
2024-06-06T09:35:26.470Z CNTRLR   [Node 176] [+] [Node Naming and Location] name: "Kitchen Light"
2024-06-06T09:35:26.470Z CNTRLR   [Node 176] Beginning interview - last completed stage: None
2024-06-06T09:35:26.470Z CNTRLR   [Node 176] new node, doing a full interview...
2024-06-06T09:35:26.471Z CNTRLR » [Node 176] querying protocol info...
2024-06-06T09:35:26.472Z DRIVER   all queues busy
2024-06-06T09:35:26.475Z SERIAL » 0x0105004100b00b                                                     (7 bytes)
2024-06-06T09:35:26.476Z DRIVER » [REQ] [GetNodeProtocolInfo]
                                    payload: 0x00b0
2024-06-06T09:35:26.479Z SERIAL « [ACK]                                                                   (0x06)
2024-06-06T09:35:26.481Z SERIAL « 0x010a0141d39c0104100100ee                                          (12 bytes)
2024-06-06T09:35:26.482Z SERIAL » [ACK]                                                                   (0x06)
2024-06-06T09:35:26.482Z DRIVER « [RES] [GetNodeProtocolInfo]
                                    payload: 0xd39c0104100100
2024-06-06T09:35:26.483Z CNTRLR « [Node 176] received response for protocol info:
                                  basic device class:    Routing Slave
                                  generic device class:  Binary Switch
                                  specific device class: Binary Power Switch
                                  node type:             End Node
                                  is always listening:   true
                                  is frequent listening: false
                                  can route messages:    true
                                  supports security:     false
                                  supports beaming:      true
                                  maximum data rate:     100000 kbps
                                  protocol version:      3
2024-06-06T09:35:26.484Z CNTRLR   [Node 176] Interview stage completed: ProtocolInfo
2024-06-06T09:35:26.484Z CNTRLR » [Node 176] querying node info...
2024-06-06T09:35:26.486Z SERIAL » 0x0105006000b02a                                                     (7 bytes)
2024-06-06T09:35:26.486Z DRIVER » [Node 176] [REQ] [RequestNodeInfo]
                                    node id: 176
2024-06-06T09:35:26.489Z SERIAL « [ACK]                                                                   (0x06)
2024-06-06T09:35:26.491Z SERIAL « 0x01040160019b                                                       (6 bytes)
2024-06-06T09:35:26.491Z SERIAL » [ACK]                                                                   (0x06)
2024-06-06T09:35:26.492Z DRIVER « [RES] [RequestNodeInfo]
                                    was sent: true
2024-06-06T09:35:26.518Z SERIAL « 0x010d00498400b0060410015e559f08                                    (15 bytes)
2024-06-06T09:35:26.518Z SERIAL » [ACK]                                                                   (0x06)
2024-06-06T09:35:26.519Z DRIVER « [Node 176] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x00b0060410015e559f
2024-06-06T09:35:26.519Z CNTRLR « [Node 176] node info received
                                  supported CCs:
                                  · Z-Wave Plus Info
                                  · Transport Service
                                  · Security 2
2024-06-06T09:35:26.519Z CNTRLR « [Node 176] node info received
                                  supported CCs:
                                  · Z-Wave Plus Info
                                  · Transport Service
                                  · Security 2
2024-06-06T09:35:26.520Z CNTRLR   [Node 176] Interview stage completed: NodeInfo
2024-06-06T09:35:26.521Z CNTRLR » [Node 176] Querying securely supported commands (S2_Authenticated)...
2024-06-06T09:35:26.523Z SERIAL » 0x011100a9000100b0039f0197250000000037ee                            (19 bytes)
2024-06-06T09:35:26.523Z DRIVER » [Node 176] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      55
                                  └─[Security2CCNonceGet]
                                      sequence number: 151
2024-06-06T09:35:26.528Z SERIAL « [ACK]                                                                   (0x06)
2024-06-06T09:35:26.530Z SERIAL « 0x010401a90152                                                       (6 bytes)
2024-06-06T09:35:26.530Z SERIAL » [ACK]                                                                   (0x06)
2024-06-06T09:35:26.530Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2024-06-06T09:35:26.547Z SERIAL « 0x011d00a93700000100ab7f7f7f7f00000300000000030100007f7f7f7f7fa8    (31 bytes)
2024-06-06T09:35:26.547Z SERIAL » [ACK]                                                                   (0x06)
2024-06-06T09:35:26.547Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            55
                                    transmit status:        OK, took 10 ms
[... 140 lines cut ...]
2024-06-06T09:35:27.734Z SERIAL « 0x011d00a93a00000300aa7f7f7f7f00000300000000030100007f7f7f7f7fa6    (31 bytes)
2024-06-06T09:35:27.734Z SERIAL » [ACK]                                                                   (0x06)
[... cut ...]
                                    │ security class:  S2_Authenticated
                                    └─[VersionCCCommandClassReport]
                                        CC:      Version
                                        version: 2
2024-06-06T09:35:27.746Z CNTRLR   [Node 176]   supports CC Version (0x86) in version 2
```

You can try excluding, factory resetting and including the device again to see if that fixes it. If not, this is a case for HomeSeer's support - they may want to see that log snippet.

pdobrien3, 2024-06-07:
exclusion, factory reset, and inclusion fixed the issue.  Thanks !!!

PR #6868 (2024-05-24): fix(config): use HomeSeer template for LED Indicator (parameter 3) for all HomeSeer switches

Fix LED indicator (parameter 3) for all HomeSeer switches, fixes #4047
Also fix small issue with parameter 32 for WX300 which should not allow manual entry.