zwave-js/zwave-js #4053: [bug] Sensative Strips Guard 700 not responding on versions > 6.0.3

h-parks, 2022-01-07:
### Version

**Checklist:**

- [ x ] a developer has told me to come here.
- [ x ] I have checked the troubleshooting section and my problem is **not** described there.
- [ x ] I have read the changelog and my problem is **not** mentioned there.

Build/Run method

- [ x ] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

Zwavejs2Mqtt version: 6.2.0
Z-Wave JS version: 8.9.1

### Describe the bug

Sensative Strips Guard 700 window/door sensors unresponsive after installing any version above v6.0.3

### To Reproduce

Steps to reproduce the behavior:

1. Install any release > 6.0.3
2. Open window/door with sensative strips guard 700
3. Device does not respond

### Expected behavior

Device responds 

[v620__zwavejs2mqtt-store.zip](https://github.com/zwave-js/zwavejs2mqtt/files/7789062/v620__zwavejs2mqtt-store.zip)
[v603__zwavejs2mqtt-store.zip](https://github.com/zwave-js/zwavejs2mqtt/files/7789063/v603__zwavejs2mqtt-store.zip)

AlCalzone, 2022-01-07:
I did change how duplicates are detected (the last 10 seq nums are checked rather than just the last), but that was to avoid discarding the shared encryption state when a recent message was re-transmitted by the mesh.

Maybe the devices get hung up on wanting a response to the Supervision Get requests, whose duplicates aren't answered anymore.

mundschenk-at, 2022-01-07:
Might be. There is a configuration setting for activating supervision, that should be set to "only open/close reports" and not "all unsolicited reports". Pitty I didn't keep one strip un-updated for more detailled A/B analysis. I might order another one, though (but that will take some time).

mundschenk-at, 2022-02-18:
Just had noticeable issues with one of the Sensative strips again (i.e. the sensor was still showing "open" when it was already closed) and had a look at the logs: Sure, there were "duplicate command" errors again, also for other nodes and a lot earlier than I noticed.

Node 89 and Node 90 are Sensative Strips Guard 700 (with firmware 10.28): [zwavejs_2022-02-18.log](https://github.com/zwave-js/node-zwave-js/files/8099231/zwavejs_2022-02-18.log)

AlCalzone, 2022-03-02:
So, I've picked out one of these situations. Before and afterwards you can see that the sequence number increases with each message:

```
2022-02-18T14:02:05.830Z SERIAL « 0x011a00040059129f03e400e9907854383076bb1ca5e38fd0bfbc0044          (28 bytes)
2022-02-18T14:02:05.834Z DRIVER « [Node 089] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 228
                                    └─[SupervisionCCGet]
                                      │ session id:      43
                                      │ request updates: false
                                      └─[WakeUpCCWakeUpNotification]
...
2022-02-18T14:52:06.344Z SERIAL « 0x0117000400590f9f03e500771cf41658a2428d521d88bc0004                (25 bytes)
2022-02-18T14:52:06.358Z DRIVER « [Node 089] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 229
                                    └─[BatteryCCReport]
                                        level:  100
                                        is low: false
...
2022-02-18T16:02:07.006Z SERIAL « 0x0117000400590f9f03e60016ed048887c1cdb258ca13be00b1                (25 bytes)
2022-02-18T16:02:07.019Z DRIVER « [Node 089] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 230
                                    └─[BatteryCCReport]
                                        level:  100
                                        is low: false
...
2022-02-18T16:04:06.546Z SERIAL « 0x012100040059199f03e7009e4cd5449efe938037b33ef886fa905b903600849fb (35 bytes)
                                  c0025
2022-02-18T16:04:06.553Z DRIVER « [Node 089] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 231
                                    └─[SupervisionCCGet]
                                      │ session id:      44
                                      │ request updates: false
                                      └─[NotificationCCReport]
                                          notification type:   Access Control
                                          notification status: 255
                                          notification state:  Window/door is open
```

Then suddenly, it doesn't:
```
2022-02-18T16:04:27.527Z SERIAL « 0x011d00040059159f03e7008348f94def04959737d460919862a259eabc0002    (31 bytes)
2022-02-18T16:04:27.533Z DRIVER « [Node 089] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Duplicate command
```
and the serial data shows that it reuses the sequence number (the `e7` after `9f03`), although with a different payload.

The next message (which can't be decoded) is actually using sequence number 232, and after re-transmitting this on request, it uses 233.
```
2022-02-18T16:05:20.100Z DRIVER « [Node 089] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 233
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0xc5f8c56b34e3177d6465fc0209dce10e
                                    └─[SupervisionCCGet]
                                      │ session id:      45
                                      │ request updates: false
                                      └─[NotificationCCReport]
                                          notification type:   Access Control
                                          notification status: 255
                                          notification state:  Window/door is open
```

---

TL;DR: This is definitely a device bug - and none that I can work around. Allowing the re-use of sequence numbers and accepting the command would jeopardize S2 encryption.

AlCalzone, 2022-03-02:
FWIW, I'm not sure if you changed some settings here - Node 89 sometimes reports its status with Supervision
```
2022-02-18T17:01:56.471Z DRIVER « [Node 089] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 240
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0xa6883969addd18d09d00e1dba6b09ee4
                                    └─[SupervisionCCGet]
                                      │ session id:      50
                                      │ request updates: false
                                      └─[NotificationCCReport]
                                          notification type:   Access Control
                                          notification status: 255
                                          notification state:  Window/door is closed
```

and sometimes without:
```
2022-02-18T09:21:34.472Z DRIVER « [Node 089] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 219
                                    └─[NotificationCCReport]
                                        notification type:   Access Control
                                        notification status: 255
                                        notification state:  Window/door is closed
```

Not sure why that is.

mundschenk-at, 2022-03-02:
Anyway, I'm in contact with Sensative engineers on these problems (i.e. I have an open ticket on the original S2 issues), so I'll report your findings to them and maybe direct them here directly.

closed 2022-03-06 by zwave-js-assistant[bot]

thecobra666, 2022-03-09:
I'm also having this exact same issue @mundschenk-at  did you receive any feedback from them?

mundschenk-at, 2022-03-09:
> I'm also having this exact same issue @mundschenk-at did you receive any feedback from them?

Yes, I have. They have found one issue, but are still looking for the other one (the missing Supervision I think). They had some questions on how to set up HA/Z-Wave JS to create detailed log files, which I was happy to answer. I'm confident that we will get a new firmware version soon.

mundschenk-at, 2022-03-15:
Update: I've received an interim firmware which I will try out in the next few days.

mundschenk-at, 2022-03-16:
> Update: I've received an interim firmware which I will try out in the next few days.

OK, I was bit too hasty: I haven't received the firmware yet, but it should be ready in the next few days.

mundschenk-at, 2022-04-16:
Update on the new firmware: It appears to have fixed the most common problem, but Sensative have discovered a few edge cases to iron out, so the final version will take some more time (I'm estimating one or two week, but that is just an educated guess on my part). I'll post an update when the release version has "shipped".

mundschenk-at, 2022-09-10:
For anyone still reading this: I got a new interim firmware build to test but there is no final release version yet.