zwave-js/zwave-js #4060: Interview of Qubino ZMNHTD  is not finding the meter values for endpoint 1

lapo-pasqui, 2022-01-09:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

The HA integration fails to be fed with values coming from the Qubino Meter (NodeId: 7) as, during the interview phase of such nodes, not all values are reported.

Accordingly to [HA developer](https://github.com/home-assistant/core/issues/63088#), the driver fails in correctly send information about the whole Instance 1 of the CommandClass 50 during the interview phase but only send data value changes for such instances.

As HA needs to know the datatype of all entities and their presence before evolving their states to new values, all data changes coming from an unknown/partially filled entity are discharged.

Therefore all of this data reported upward, such as
```
2022-01-09T16:48:41.576Z SERIAL « 0x01180004000712600d0100320221340000103d00000000000092              (26 bytes)
2022-01-09T16:48:41.578Z DRIVER   Node 7, Endpoint 1: Trying to access endpoint instance before Multi Channel in
                                  terview
2022-01-09T16:48:41.579Z DRIVER   Node 7, Endpoint 1: Trying to access endpoint instance before Multi Channel in
                                  terview
2022-01-09T16:48:41.580Z CNTRLR   [Node 007] [~] [Meter] value[66049]: 440.5 => 415.7               [Endpoint 1]
2022-01-09T16:48:41.584Z SERIAL » [ACK]                                                                   (0x06)
2022-01-09T16:48:41.588Z DRIVER « [Node 007] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      1
                                    │ destination: 0
                                    └─[MeterCCReport]
                                        type:       Electric
                                        scale:      W
                                        rate type:  Consumed
                                        value:      415.7
                                        time delta: 0 seconds

```
are received
```
2022-01-09 16:48:41 DEBUG (MainThread) [zwave_js_server] Received message:
WSMessage(type=<WSMsgType.TEXT: 1>, data='{"type":"event","event":{"source":"node","event":"value updated","nodeId":7,"args":{"commandClassName":"Meter","commandClass":50,"endpoint":1,"property":"value","propertyKey":66049,"newValue":415.7,"prevValue":440.5,"propertyName":"value","propertyKeyName":"Electric_W_Consumed"}}}', extra='')
```

but not evaluated as the ```value_id='17-50-1-value-66049'``` is not _registered_.

Re-interviewing the node do not solve the problem

### Device information

Manufacturer: Qubino
Model name: ZMNHTD
Node ID in your network: 7

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [X] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: honestly, not idea. I think is included by zwavejs2mqtt
`zwavejs2mqtt` branch: master (Release 6.2.0)
`Home assistant zwave_js integration` Version: Home Assistant 2021.12.5

The node-zwave-js has been manually built from github from `zwavejs2mqtt` master version 6.2.0

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

Attached the [log files](https://github.com/zwave-js/node-zwave-js/files/7835561/logs.zip) taken
1. during the interview of the node 7 started from HA and
2. during the subsequent restart of HA (which you may by not interested on)

AlCalzone, 2022-01-09:
There are a bunch of crazy things going on.

The lifeline association of the node is weird:
```
2022-01-09T16:38:23.486Z DRIVER « [Node 007] [REQ] [ApplicationCommand]
                                  └─[MultiChannelAssociationCCReport]
                                      group id:           1
                                      maximum # of nodes: 1
                                      node ids:
                                      endpoints:          0:1, 0:0
```
This is wrong on many levels. First it reports that it only accepts one destination, but reports two. Second it has a target node of 0, which is invalid. The controller is 1. Third (not shown here), it isn't possible to delete those - they just come back...

---

The node doesn't report support for the Multi Channel CC, so **there is no endpoint 1**.
```
2022-01-09T16:38:15.637Z CNTRLR « [Node 007] node info received
                                  supported CCs:
                                  · Z-Wave Plus Info
                                  · Basic
                                  · All Switch
                                  · Meter
                                  · CRC-16 Encapsulation
                                  · Configuration
                                  · Version
                                  · Manufacturer Specific
                                  · Powerlevel
                                  · Association
                                  · Multi Channel Association
                                  · Association Group Information
                                  · Device Reset Locally
                                  · Firmware Update Meta Data
                                  controlled CCs:
                                  · Basic
```
Since this is a Qubino device, I'm guessing you changed a configuration parameter that enables endpoint 1, but you did not **exclude and re-include** it like the manual tells you to.

Unless you do that, it's not possible to correctly interact with the device.

lapo-pasqui, 2022-01-11:
You were absolutely right.
This is one of the first nodes I have included in my network and it has been around for years before I decided to make it fully functional. I think it was included using [z-way] (https://z-wave.me/z-way) and I don't remember how I did it. Then I switched to HA with zwave support (now legacy) and never managed to get it to work with autonomous reports but was forced to poll it to get the values from it.

Now, I gave it another chance and wanted to refine it using the new integration which finally gave me access to the entire node configuration.

And yes, ** the exclusion and re-inclusion ** of the node make endpoint 1 disappear and all the necessary data is present consistently on instance 0.

Thanks for the exclude-include trick that I had no idea of and that I barely found in the manual and thanks again for the support and the wonderful project.

It's okay for me to close this ticket

closed 2022-01-11 by AlCalzone