zwave-js/zwave-js #4061: Aeotec  ZWA024-C (EU version)  0x0371:0x0002:0x00018.  Must be cloned from 0x0371/0x0102/0x0018 which is the US version (ZWA024-A/product 4088). Two files or more to be created to manage region specifics

paifre, 2022-01-09:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Aeotec ZWA024-C (EU version)

### Manufacturer ID

0x0371

### Product Type

0x0002

### Product ID

0x0018

### Firmware Version

1.1

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4061

AlCalzone, 2022-01-09:
@zwave-js-bot add fingerprint 0x0371:0x0002:0x00018 to 0x0371:0x0102:0x00018

AlCalzone, 2022-01-10:
@zwave-js-bot add fingerprint 0x0371:0x0002:0x0018 to 0x0371:0x0102:0x0018

paifre, 2022-01-15:
Hello,
Many thanks for the PR
Sorry to bother, but I've updated last night to latest release which includes the PR that I was looking for. I've checked that the correct zwa024.json is deployed.
Unfortunately, the configuration section of the multisensor is still with default param text from aeotec, not being overwritten by the content of the device file.
I've not been able to point out what's wrong
Adding a screenshot to make it more easier to understand and also the debug output of the node
[image]
[My_ZWA024_DebugFile.txt](https://github.com/zwave-js/node-zwave-js/files/7874797/My_ZWA024_DebugFile.txt)
d

AlCalzone, 2022-01-15:
You need to re-interview the device for these changes to be picked up.

paifre, 2022-01-15:
Worked perfectly. Great job :)
Maybe one suggestion while closing: 
Could be interesting to propose multiple downloads via a list with a brief desc.
If I take the example of this device: user manual is quite basic, not so useful, while many interesting infoin the Engineering sheet
[](https://help.aeotec.com/helpdesk/attachments/6124934520)

Many many thanks

closed 2022-01-15 by paifre

AlCalzone, 2022-01-15:
Could you open a new issue for this so we don't forget?