zwave-js/zwave-js #4064: Add value notification event for Multilevel Switch CC

binzhou, 2022-01-10:
**Is your feature request related to a problem? Please describe.**
The Aeotec ZW111 is an in-wall dimmer with a single output but wiring for 2 external switches. It can be configured such that the second switch only sends zwave commands to an association group rather than controlling local load. It supports sending 2 sets of commands: basic and multilevel switch CC.

Currently, we can take advantage of this by associating the device with the controller node and setting it to send only basic commands - these are received by zwave-js and converted to value notifications. However, this doesn't work when set to multievel switch CC. 

If this were supported, then not only could you program automations (eg. in Home Assistant) for on/off events, but also for multilevel start/stop commands which correspond to holding and releasing the external switch.

**Describe the solution you'd like**
1. Support parsing of MultilevelSwitchCCSet and MultilevelSwitchCCStartLevelChange (MultilevelSwitchCCStop already supported as there are no options)
2. Support generating value notifications when the controller receives these commands. 

**Describe alternatives you've considered**
N/A

**Additional context**
Log from when I toggled then held-and-released external switch that had direct association with the controller and was set to send Multilevel Switch CCs
```
Subscribed to Z-Wave JS Log Messages…
2022-01-08T22:45:32.030Z SERIAL « 0x010c0004000d042601ff01da00fd (14 bytes)
2022-01-08T22:45:32.030Z DRIVER Dropping message because it could not be deserialized: MultilevelSwitchCCSet: 
                                  deserialization not implemented (ZW0320)
2022-01-08T22:45:32.031Z SERIAL » [ACK] (0x06)
2022-01-08T22:45:34.329Z SERIAL « 0x010c0004000d0426010001da0002 (14 bytes)
2022-01-08T22:45:34.330Z DRIVER Dropping message because it could not be deserialized: MultilevelSwitchCCSet: 
                                  deserialization not implemented (ZW0320)
2022-01-08T22:45:34.331Z SERIAL » [ACK] (0x06)
2022-01-08T22:45:37.229Z SERIAL « 0x010d0004000d052604200001d80025 (15 bytes)
2022-01-08T22:45:37.230Z DRIVER Dropping message because it could not be deserialized: MultilevelSwitchCCStart
                                  LevelChange: deserialization not implemented (ZW0320)
2022-01-08T22:45:37.231Z SERIAL » [ACK] (0x06)
2022-01-08T22:45:37.829Z SERIAL « 0x010a0004000d022605d80005 (12 bytes)
2022-01-08T22:45:37.830Z SERIAL » [ACK] (0x06)
2022-01-08T22:45:37.831Z DRIVER « [Node 013] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCStopLevelChange]
2022-01-08T22:45:37.833Z CNTRLR « [Node 013] TODO: no handler for application command
```

AlCalzone, 2022-01-10:
Interesting use case, but I can see the value in having this.
Note that Start/Stop would need to be under the `notification` event (not `value notification`), because they are a bit more complex (start/stop, direction and potentially startLevel).

I'll put this on the todo list - if anyone wants to help out, go for it!

AlCalzone, 2022-01-18:
Some pointers:
1. The deserialization must be implemented in the `MultilevelSwitchCCSet` command here:
https://github.com/zwave-js/node-zwave-js/blob/27bb4ee0cce202a93e1479aa4aca4ce5e2589a58/packages/zwave-js/src/lib/commandclass/MultilevelSwitchCC.ts#L584-L589
similar to https://github.com/zwave-js/node-zwave-js/blob/3a7251d111320190f8936b277a723c0b0ae9711f/packages/zwave-js/src/lib/commandclass/BasicCC.ts#L284-L285

2. An `event` value needs to be defined like for the Basic CC here
https://github.com/zwave-js/node-zwave-js/blob/3a7251d111320190f8936b277a723c0b0ae9711f/packages/zwave-js/src/lib/commandclass/BasicCC.ts#L61-L67
and created during the interview:
https://github.com/zwave-js/node-zwave-js/blob/3a7251d111320190f8936b277a723c0b0ae9711f/packages/zwave-js/src/lib/commandclass/BasicCC.ts#L213-L221
I'd gate it behind a compat flag like we do for Basic CC, so this needs to be defined [here](https://github.com/zwave-js/node-zwave-js/blob/7b761d508fd0a85474e1e36156f937d2b7c429c3/packages/config/src/CompatConfig.ts#L1) and [here](https://github.com/zwave-js/node-zwave-js/blob/458a406e65f718fe5491bce29acbbef6555e5677/maintenance/schemas/device-config.json#L282-L284)

3. The Start/Stop level change commands need to implement parsing too (see 1.) and new notifications must be defined for them [here](https://github.com/zwave-js/node-zwave-js/blob/3707c93010ce01dc4c8e28a4eac63cac680f0895/packages/zwave-js/src/lib/node/Types.ts#L85-L91) and emitted like packages/zwave-js/src/lib/node/Node.ts.

reubenbijl, 2022-02-25:
I've just started  having a look at this and have made the changes that you've suggested to the MultilevelSwitchCCSet command, but I'm getting an error where  at node-zwave-js/packages/zwave-js/src/lib/node/Node.ts the command isn't recognised, so I've added a category for the Multilevel Switch but just wanted to check that this was expected?

reubenbijl, 2022-02-25:
@binzhou I've added support for the Start and Stop Level Change Commands now as well, let me know if you get a chance to test it out.

@AlCalzone Where's the best place to get feedback on the commits?

AlCalzone, 2022-02-25:
> Where's the best place to get feedback on the commits?

Just open a PR here, this makes it easiest for me to check out what you did.

j9brown, 2022-03-10:
FWIW, I believe this proposed change will also address a need for Leviton VRCS4 and Leviton VRCZ4 devices (and perhaps others too).  These devices have a rocker switch that functions as a dimmer by sending Multilevel Switch messages to the associated load (or the hub).  The rocker can be pressed to adjust the associated load's level incrementally or held to adjust the level continually until released.

In my previous integration with OpenZwave, I reported these events as "up" / "down" / "stop" similar to what is being proposed here (and I was just investigating how to do it with zwave-js).  So I'm excited about this change.  ;)

PR #4282 (2022-03-14): feat(cc): add value event and notifications for Multilevel Switch CC Set and Start/StopLevelChange
Hi,

Would really appreciate some feedback on this   so I can keep working on it.

fixes: #4064