zwave-js/zwave-js #4070: HS-WX300 switch mode S2 broken

pdobrien3, 2022-01-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?  When including the HS-WX300 while in switch mode, it appears to successfully include but then you cannot control the device.  If included with no encryption, it works fine.  Also, apparently when included with S2 in dimmer mode, the device works fine

What do you observe?

https://community.home-assistant.io/t/homeseer-hs-wx300-in-switch-mode/362425

What did you expect to happen?

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: Homeseer
Model name: HS-WX300
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

zwavejs2mqtt: 6.3.0
zwave-js: 8.10.1
home id: 3479161084
home hex: 0xcf5fc8fc

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

Log files are in the Home Assistant Forum post

pdobrien3, 2022-01-10:
Hoping to pass along this issue to Homeseer for their response

COCottonwood, 2022-01-12:
@pdobrien3 - I updated last night before I went to sleep and at 3:55am Eastern all of my ZWave devices became unavailable. They reconnected 30 seconds later. Now when I got home after work today, any of my "ge_quick_fit_smart_in_wall_paddle_switch" switches that I have listed are no longer in my Dashboard. I saw another post on reddit/homeassistant from PDXSCARGuy that said he had the same problem today, losing all of his "ge_quick_fit_smart_in_wall_paddle_switch" devices listed in his dashboard.

All of their automations are still there, but these devices have reset back to their generic names and entity ids. I can still find them as a device, but under their generic name.  I also can NOT control them.  I can RE-INTERVIEW them, and see that they are communicating in the logs, but the dashboard shows no way to switch them ON or OFF.

These are GE/JASCO 46201 / ZW4008 devices.

blhoward2, 2022-01-12:
You appear to have posted in the wrong issue. Please post it in the correct place or open your own issue.

kpine, 2022-01-12:
@COCottonwood The issue for your problem is https://github.com/home-assistant/core/issues/58874, and it will be fixed in the next HA release. You'll need to rename the entities back to what you want.

COCottonwood, 2022-01-12:
@kpine - I had to ENABLE the device first, then rename it and them I could control it.  Thanks for heads-up.

COCottonwood, 2022-01-13:
New issue opened - https://github.com/home-assistant/core/issues/63984

AlCalzone, 2022-01-13:
@COCottonwood stop posting in unrelated issues please

COCottonwood, 2022-01-13:
@AlCalzone - Sorry, just trying to help.

AlCalzone, 2022-05-04:
@pdobrien3 did you ever hear anything back from Homeseer?

pdobrien3, 2022-05-04:
Yes, here was their response:

While I have not heard back about parameter 3, I do have more information about S2 security. That functionality was tested using independent Z-Wave certification tools as well as HS4 and we are unable to recreate the issue. There are only reports of this issue occurring in Home Assistant systems and I believe our engineers intend to post their detailed findings on the Home Assistant forum.

AlCalzone, 2022-05-04:
> There are only reports of this issue occurring in Home Assistant systems

Might be because there are no other open source drivers doing S2, but let me know when they do. That linked thread hasn't seen any activity in months, wonder when they are going to post their findings.

pdobrien3, 2022-05-04:
I pretty much threw my hands up.  Mostly because I do not have the knowledge to argue with them.  They basically say S2 is a function of the software and not the device. I have the same issue with the 100s, they also can not be included with S2 even though they claim they support it.

AlCalzone, 2022-05-04:
AFAIK S2 on end devices is actually implemented in the Z-Wave SDK, so it ideally they would be right.
But on the other hand there are other devices from some manufacturers which also seem to have sporadic issues with S0 and S2 while devices from other manufacturers don't have these issues at all...

Go figure 🤷🏻‍♂️

pdobrien3, 2022-05-04:
Wouldn’t it be awesome if they saw the benefit of looking at the zwavejs code and providing an opinion.

closed 2023-02-06 by AlCalzone