zwave-js/zwave-js #4072: Eaton RF9540 - no updates from physical presses

rahi6, 2022-01-12:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
No updates from physical presses on the dimmer switch (turn off, turn on, dimmer brightness changes are not updated)

What do you observe?
I have tried to interview the node multiple times, but no group associations are created. Without any group associations when the dimmer is physically pressed on the debug log shows:

2022-01-12T01:18:10.483Z SERIAL « 0x01160049840a1004110426277586707185772b2c727387af (24 bytes)
2022-01-12T01:18:10.486Z SERIAL » [ACK] (0x06)
2022-01-12T01:18:10.489Z DRIVER « [Node 010] [REQ] [ApplicationUpdateRequest]
payload: 0x0a1004110426277586707185772b2c727387
2022-01-12T01:18:10.492Z CNTRLR « [Node 010] Received updated node info

If I add a group association on the dimmer switch with the controller on Group 1 (endpoint 0). The debug log has:

`Phyiscally Switch On

2022-01-12T00:58:20.940Z SERIAL « 0x01160049840a1004110426277586707185772b2c727387af (24 bytes)
2022-01-12T00:58:20.942Z SERIAL » [ACK] (0x06)
2022-01-12T00:58:20.944Z DRIVER « [Node 010] [REQ] [ApplicationUpdateRequest]
payload: 0x0a1004110426277586707185772b2c727387
2022-01-12T00:58:20.947Z CNTRLR « [Node 010] Received updated node info
2022-01-12T00:58:20.951Z SERIAL « 0x01090004000a03200163b9 (11 bytes)
2022-01-12T00:58:20.952Z SERIAL » [ACK] (0x06)
2022-01-12T00:58:20.954Z DRIVER « [Node 010] [REQ] [ApplicationCommand]
└─[BasicCCSet]
target value: 99
2022-01-12T00:58:20.956Z CNTRLR [Node 010] treating BasicCC::Set as a report
2022-01-12T00:58:20.957Z CNTRLR [Node 010] [~] [Basic] currentValue: 0 => 99 [Endpoint 0]
2022-01-11 19:58:20.958 INFO ZWAVE: Node 10: value updated: 32-0-currentValue 0 => 99

Press and release dimmer brightness

2022-01-12T01:02:01.493Z SERIAL « 0x010b0004000a052604406304fa (13 bytes)
2022-01-12T01:02:01.496Z DRIVER Dropping message because it could not be deserialized: MultilevelSwitchCCStart
LevelChange: deserialization not implemented (ZW0320)
2022-01-12T01:02:01.497Z SERIAL » [ACK] (0x06)
2022-01-12T01:02:02.153Z SERIAL « 0x010a0004000a0426014c0490 (12 bytes)
2022-01-12T01:02:02.156Z DRIVER Dropping message because it could not be deserialized: MultilevelSwitchCCSet:
deserialization not implemented (ZW0320)
2022-01-12T01:02:02.157Z SERIAL » [ACK] (0x06)

Phyiscally Switch Off

2022-01-12T01:02:58.405Z SERIAL « 0x01160049840a1004110426277586707185772b2c727387af (24 bytes)
2022-01-12T01:02:58.408Z SERIAL » [ACK] (0x06)
2022-01-12T01:02:58.411Z DRIVER « [Node 010] [REQ] [ApplicationUpdateRequest]
payload: 0x0a1004110426277586707185772b2c727387
2022-01-12T01:02:58.415Z CNTRLR « [Node 010] Received updated node info
2022-01-12T01:02:58.419Z SERIAL « 0x01090004000a03200100da (11 bytes)
2022-01-12T01:02:58.421Z SERIAL » [ACK] (0x06)
2022-01-12T01:02:58.423Z DRIVER « [Node 010] [REQ] [ApplicationCommand]
└─[BasicCCSet]
target value: 0
2022-01-12T01:02:58.427Z CNTRLR [Node 010] treating BasicCC::Set as a report
2022-01-12T01:02:58.428Z CNTRLR [Node 010] [~] [Basic] currentValue: 99 => 0 [Endpoint 0]
2022-01-11 20:02:58.429 INFO ZWAVE: Node 10: value updated: 32-0-currentValue 99 => 0
`

What did you expect to happen?
Instant status update from manual switch presses

Steps to reproduce the behavior:
Physically press the switch and don't get any updates.

### Device information

Manufacturer: Eaton
Model name: RF9540-N
Node ID in your network: <10>

Controller: HomeSeer SmartStick+ Z-Wave Plus

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 6.3.1
`zwavejs2mqtt` branch: 8.10.2

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

openHAB 3.3 (sorry, probably unrelated but this dimmer switch provided instant updates correctly on that platform)

### Attach Driver Logfile

[zwavejs_2022-01-11.log](https://github.com/zwave-js/node-zwave-js/files/7851051/zwavejs_2022-01-11.log)

AlCalzone, 2022-01-12:
Looking at the manual
[image]
and our config file https://devices.zwave-js.io/?jumpTo=0x001a:0x4449:0x0101:0.0, association group 255 is supposed to be auto-assigned. Group 1 seems to be controlling other devices, hence the `Set` commands.

Question is why the association isn't created or working. Please make another driver logfile, this time re-interview the node while doing so. If you can, remove all associations first.

rahi6, 2022-01-12:
Thanks for the response, I appreciate it!

I've added the log file with node being re-interviewed. I removed all associations prior to the re-interview.
[zwavejs_2022-01-12.log](https://github.com/zwave-js/node-zwave-js/files/7857065/zwavejs_2022-01-12.log)

AlCalzone, 2022-01-12:
Ok looks like group 255 isn't reported to exist by the device. This causes some information to be missing thats necessary to decide how to assign the lifeline. Luckily we have config files to rely on, I just need to add a fallback.

rahi6, 2022-01-14:
Hi @AlCalzone unfortunately the fix did not work on the latest version (ZWave JS 8.11.0). I noticed different behavior with different firmware version of the same switch. Switches on firmware 1.1 work perfectly. Switches on firmware 1.2 have the issue still. I have multiples of each version and that's when I noticed the pattern. It seems with the 1.2 firmware version, Group 255 isn't supported and the lifeline needs to be setup on Group 1 (just a hunch).

I have attached the log file with Node 10 (switch firmware 1.2) being re-interviewed and Node 11 (switch 1.1) being re-interviewed. Much appreciated on the work done here!
[zwavejs_2022-01-14.log](https://github.com/zwave-js/node-zwave-js/files/7872959/zwavejs_2022-01-14.log)

AlCalzone, 2022-01-15:
> It seems with the 1.2 firmware version, Group 255 isn't supported and the lifeline needs to be setup on Group 1 (just a hunch).

Looks like this is the case - can you fire up zwavejs2mqtt and try adding that association to verify?

rahi6, 2022-01-15:
I've added the association - still no updates in the value.

Added association:
[image]

Debug log lines after pressing switch on:

2022-01-15T18:45:24.175Z SERIAL « 0x01160049840a1004110426277586707185772b2c727387af                  (24 bytes)
2022-01-15T18:45:24.178Z SERIAL » [ACK]                                                                   (0x06)
2022-01-15T18:45:24.179Z DRIVER « [Node 010] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x0a1004110426277586707185772b2c727387
2022-01-15T18:45:24.184Z CNTRLR « [Node 010] Received updated node info
2022-01-15T18:45:24.189Z SERIAL « 0x01090004000a03200163b9                                            (11 bytes)
2022-01-15T18:45:24.189Z SERIAL » [ACK]                                                                   (0x06)
2022-01-15T18:45:24.190Z DRIVER « [Node 010] [REQ] [ApplicationCommand]
                                  └─[BasicCCSet]
                                      target value: 99
2022-01-15T18:45:24.191Z CNTRLR   [Node 010] treating BasicCC::Set as a report
2022-01-15T18:45:24.191Z CNTRLR   [Node 010] [~] [Basic] currentValue: 0 => 99                      [Endpoint 0]

Value in zwavejs2mqtt not updated:
[image]

AlCalzone, 2022-01-15:
Ok it's sending Basic Set commands, not Multilevel Switch Reports. We need to add a compat flag to work with that.

rahi6, 2022-01-15:
Thanks, how do I add that?

AlCalzone, 2022-01-16:
You can follow [this guide](https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files) using these two files:
- https://raw.githubusercontent.com/zwave-js/node-zwave-js/59fcc6fd2415d7f54c3fcd11243fa1f89c9bb6ad/packages/config/config/devices/0x001a/rf9540-n_0.0_1.1.json
- https://raw.githubusercontent.com/zwave-js/node-zwave-js/59fcc6fd2415d7f54c3fcd11243fa1f89c9bb6ad/packages/config/config/devices/0x001a/rf9540-n_1.2.json

Let me know if this works now (you'll probably have to re-interview the device)

rahi6, 2022-01-17:
the config files work! the on/off presses now get updated instantly. Thank you!

PR #4076 (2022-01-12): fix: add fallback to config file for associations' max node counts
If a config file knows about an association group the node doesn't report, we now take the max node count from the config file instead of returning 0.

fixes: #4072
PR #4096 (2022-01-17): fix(config): correct lifeline for RF9540-N v1.2+ and map Basic Sets
fixes: #4072
closed 2022-01-17 by AlCalzone