zwave-js/zwave-js #4079: Track which device file was the basis of an interview and flag devices with outdated interviews

blhoward2, 2022-01-14:
Hash the device file used during a device interview and compare the hash value of current device files at startup. Also export a function so that UIs can call this function after device files are pulled (zjs2mqtt). Flag devices with outdated interviews so that UIs can display a reinterview flag or prompt on the device. 

Per prior conversations, we don't want to automatically reinterview because a device could be a battery device or there could be an issue, and all information about a device is lost after triggering a reinterview.

kpine, 2022-01-14:
> and all information about a device is lost after triggering a reinterview.

That's been [fixed](https://github.com/zwave-js/node-zwave-js/issues/3776)?

blhoward2, 2022-01-14:
No, it's lost in the driver. We dump the info when you start an interview.

closed 2023-08-16 by zwave-js-bot