zwave-js/zwave-js #4088: Make Log Files Log with Correct Time Zone Instead of UTC

kars85, 2022-01-14:
**Is your feature request related to a problem? Please describe.**
Logging appears to be in UTC. This makes troubleshooting more complicated when inspecting logs as the issue being tracked (for example, lining up when a node goes Unavailable in HA which is in the correct time zone), doesn't align with zwave-js logs.

**Describe the solution you'd like**
Have logging match up with the zwave-js instance's time zone. In my case, the Docker Compose environment variable is set correctly, but zwave-js's logging doesn't respect that environment variable.

**Describe alternatives you've considered**
None that I can think of.

jmgiaever, 2022-01-15:
Hi,

I think there's the `TZ` environment variable. E.g 

```
export TZ=Europe/Oslo
```

AlCalzone, 2022-01-15:
That applies to zwavejs2mqtt only. The driver uses `.toISOString()` which is by definition in UTC.

AlCalzone, 2022-01-15:
Accepting PRs

PR #7800 (2025-05-09): fix: use local time for logging to file
fixes: #4088
closed 2025-05-09 by zwave-js-bot