zwave-js/zwave-js #4092: S2 Inclusion: abort callback can trigger an unhandled promise rejection

marcus-j-davies, 2022-01-16:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Testing the **abort** callback in the .NET wrapper.
I decided to allow the **abort** method to trigger - (I delayed returning the **InclusionGrant**)

In the server, the **abort**  callback also includes rejecting the **validateDSKAndEnterPin** and **grantSecurityClasses** promises.

This, from my investigation, caused the **driver.start()** promise to fail with an unhandled promise rejection.

The server does this:
```javascript
 abort: (): void => {
            grantSecurityClassesPromise?.reject("aborted");
            validateDSKAndEnterPinPromise?.reject("aborted");
            client.sendEvent({
              source: "controller",
              event: "inclusion aborted",
            });
          }
```

The **driver.start()** promise gets.
```log
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
aborted

```

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

N/A

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwave-js.log_2022-01-16.log](https://github.com/zwave-js/node-zwave-js/files/7877085/zwave-js.log_2022-01-16.log)

AlCalzone, 2022-01-16:
Can you confirm the same happening in Node-Red? Both locations where these are called, `catch` rejected promises:
https://github.com/zwave-js/node-zwave-js/blob/85b6e931ea7ab6648c53d26f093829ee0f55c51f/packages/zwave-js/src/lib/controller/Controller.ts#L1891-L1895

https://github.com/zwave-js/node-zwave-js/blob/85b6e931ea7ab6648c53d26f093829ee0f55c51f/packages/zwave-js/src/lib/controller/Controller.ts#L1801-L1808

AlCalzone, 2022-01-16:
Also, this can't be the `start()` promise that's rejected.

marcus-j-davies, 2022-01-17:
> Also, this can't be the start() promise that's rejected.

Yup - don't ask me why I thought that.
I received a situation on the .NET side that should have only been possible within a catch block on driver start - so I will need to look at .NET for that.

However, I think I have pinpointed a possible cause for the unhandled promise. (I couldn't cause the issue in NR BTW).

in **zwave-js-server** it creates  a promise via **createDeferredPromise()**
https://github.com/zwave-js/zwave-js-server/blob/26f81ee968d7f160e6f29860aa5b20390de0922a/src/lib/controller/message_handler.ts#L307 - also one for the DSK  validation.

Later it rejects this instance - but its rejection is not handled - there is a **finally** but no catch - i.e unhandled rejection
https://github.com/zwave-js/zwave-js-server/blob/26f81ee968d7f160e6f29860aa5b20390de0922a/src/lib/controller/message_handler.ts#L308

If my suspicion is correct - this issue may need to be transfer to **zwave-js-server**?

AlCalzone, 2022-01-17:
Ideally the server wouldn't reject the promise itself.