zwave-js/zwave-js #4093: Node info incorrectly extracted from backup after reaching 232 nodes

AlCalzone, 2022-01-16:
First thank you very much for this information and conversion tool.

I was doing some testing between an Aeotec Z-Stick 500 plus and a SiLabs UZB-7.     Restoring to the 700 series stick.

I have added over 232 nodes to the Aeotec stick and I have looped back around and I am up to node 40 as my last added node. 

When converting the Aeotec NVM backup I noticed that only 40 nodes were restored to the 700 series stick.   When I converted the source NVM to json I see that the lastNodeId is set to 40 and only 40 nodes of information are present.   I assume that nvmedit is using the lastNodeId to determine how many nodes to include in the output and does not take into account this situation.

[NVM.zip](https://github.com/zwave-js/node-zwave-js/files/7875546/NVM.zip)

_Originally posted by @dwaleke in https://github.com/zwave-js/node-zwave-js/issues/3906#issuecomment-1013729260_

Madj42, 2022-01-30:
@AlCalzone I'm not sure if this is the same thing I'm seeing but I have 69 nodes and I'm only seeing 51 nodes in my NVM backups.  I'm on v1.2 firmware and this is on both sticks I have (Gen5 non-plus and plus).

AlCalzone, 2022-01-30:
I'll need to see Backups of the source and target stick. You can send them via email if you prefer not to post them here.

AlCalzone, 2022-02-06:
@Madj42 #4199 should fix this. Interestingly your NVM has maxNodeId 72, but nodes up to 92 are defined, so maxNodeId isn't reliable 🤷🏻‍♂️
So we now look at all potential node IDs (<= 232), not just up to lastNodeId/maxNodeId.

Madj42, 2022-02-11:
@AlCalzone Apologize for the late response, work has been hectic.  Is there a way to take a manual backup if I have the tools and node-js downloaded to a computer?  Unfortunately I'm stuck waiting for Frenk to update the add-on for HA since it's on 6.5.0.

AlCalzone, 2022-02-11:
With just Node.js (and the included npm) installed, you can follow the steps described in https://github.com/zwave-js/node-zwave-js/tree/master/packages/nvmedit
Let me know if something is unclear here.

Madj42, 2022-02-11:
@AlCalzone The only issue is that since the previous issue was NVM backup related, I can't create a backup of the 500 series stick without having the driver setup in command mode.  I may just clone frenck's addon and force it to run zwavejs2mqtt 6.5.1 and port my store files over to it later today to get this done.

AlCalzone, 2022-02-11:
The backup that's created is "dumb". The issue with missing nodes appeared when converting it, specifically in the nvm2json step.

Madj42, 2022-02-11:
So I did the conversion and for whatever reason I keep getting:

Error while calling api restoreNVM: Did not find a matching NVM 500 parser implementation! (ZW0280) 

I tried the conversion to json and back again.  Also tried the convert method.  Both gave this.

Madj42, 2022-02-11:
I emailed you the BIN files and command I used.  I verified I was running the latest version of nvmedit.

AlCalzone, 2022-02-13:
@Madj42 I think I finally have it now. Your 700 stick was reporting page sizes of 64kB, which I've seen some do. But this is larger than the maximum of 2kB, so you were getting an NVM file with an invalid size.

Madj42, 2022-02-13:
Thanks for looking into this.  Once it's ready, I'll try it out.

closed 2022-02-14 by AlCalzone