zwave-js/zwave-js #4094: Heltun HE-RS01 missing hold mode duration parameters & hold control mode parameter options

rossharper, 2022-01-16:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

- Device configuration files configuration for parameters 41-45 are missing options for value 4 "Momentary Toggle Switch". 
- Device configuration files are missing configuration parameters 46-50 for Hold Mode Duration (added in Firmware version 1.4)

What do you observe?

- Hold Control Mode cannot be configured as a "Momentary Toggle" as described in the manual (versions 1.0 & 1.1 for all public firmwares 1.2 thru 1.4)
- Hold Mode Duration cannot be configured (feature and parameters added in firmware 1.4 described in manual version 1.1)

What did you expect to happen?

- Hold Control Mode to have an option for "Momentary Toggle Switch" for parameters 41-45 that corresponds to a value of 4.
- Parameters 46-50 ("Hold Mode Duration") to be presented to allow configuration of Hold Mode duration for each switch.

Steps to reproduce the behavior:
  1. Select Heltun Relay Switch Quinto Node
  2. Scroll down to Configuration values and expand
  3. Open value drop down for any of parameters 41-45
  4. Fifth option corresponding to Momentary Toggle Switch is not present
  5. Parameters 46-50 are not listed

### Device information

Manufacturer: Heltun
Model name: Relay Switch Quinto HE-RS01
Node ID in your network: 3

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt
version: 0.33.0
`node-zwave-js` branch: master
`zwavejs2mqtt` branch: master

Home Assistant wave_js integration version: 8.10.2

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

Log file was not captured when issue was presenting. My local configuration now has updated device configuration files that resolve the issues that will be submitted in a PR to fix this issue.

AlCalzone, 2022-01-16:
The PR fixing this is already open, should be merged soon.

closed 2022-01-16 by AlCalzone

rossharper, 2022-01-16:
Thanks. I searched issues and didn't find a matching one, I should have searched PRs also.

I looked at the mentioned PR - it solves the second of the problems I've reported in this issue (new Hold Mode Duration parameters). However, it does not add the missing Hold Control Modes.

I'll update my fork and open a new PR with just the changes for that.

AlCalzone, 2022-01-16:
> However, it does not add the missing Hold Control Modes.
> 
> I'll update my fork and open a new PR with just the changes for that.

Great, thanks! I though I read something about the hold control modes when I reviewed the PR.