zwave-js/zwave-js #4100: Last update removed functionality of Haseman DRS6-AI controller (Z-Uno based)

BebeMischa, 2022-01-17:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Latest update, 8.10.2

What do you observe?

My Haseman DRS6-AI (based on Z-Uno) controller stopped working.

What did you expect to happen?

That it will stay working like before the update

Steps to reproduce the behavior:
That would be quiet difficult. I don't think, you have such device.

It is a 6ch switch with some measure points for temperature and some 0-10V sensors and binary sensors.
The device costs around 200 euro's and is the main controller of my hot tub. So just imagine the problem I have now ;-)

I tried exclusion and inclusion as first aid, but now it just can't be included. It throws errors and gets included, but not ready.

Before this update the device worked just fine, all switches and sensors were correctly imported and worked flawlessly.

  4. See error

```
2022-01-17T17:16:31.186Z CNTRLR   [Node 025] Querying the node info failed

2022-01-17T17:17:51.285Z SERIAL « 0x0107004a3605190098                                                 (9 bytes)
2022-01-17T17:17:51.288Z SERIAL » [ACK]                                                                   (0x06)
2022-01-17T17:17:51.289Z DRIVER « [REQ] [AddNodeToNetwork]
                                    payload: 0x36051900
2022-01-17T17:17:51.292Z SERIAL « 0x01110004101909600d08003105020100bc0004                            (19 bytes)
2022-01-17T17:17:51.293Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2022-01-17T17:17:51.294Z SERIAL » [ACK]                                                                   (0x06)
2022-01-17T17:17:51.335Z SERIAL « 0x0107004a370619009a                                                 (9 bytes)
2022-01-17T17:17:51.336Z SERIAL » [ACK]                                                                   (0x06)
2022-01-17T17:17:51.338Z DRIVER « [REQ] [AddNodeToNetwork]
                                    payload: 0x37061900
2022-01-17T17:17:51.349Z CNTRLR   The inclusion process was stopped
2022-01-17T17:17:53.556Z SERIAL « 0x0112000400190a600d0900310501428001c900a1                          (20 bytes)
2022-01-17T17:17:53.558Z DRIVER   Dropping message because the driver is not ready to handle it yet.
```

I see also, that there is newer release of Z-wave JS, but HA does not have noticed yet, so I can't update.

Thank you very very much for looking in to this, it causes serious troubles here ;-) 
# #

### Device information

Manufacturer: Haseman
Model name: DRS6-AI
Node ID in your network: unknown

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 8.10.2
Home Assistant Supervised core-2021.12.9, supervisor-2021.12.2

### Did you change anything?

yes (please describe)

### If yes, what did you change?

updated to 8.10.2

### Did this work before?

Yes (please describe)

### If yes, where did it work?

All previous vesions

### Attach Driver Logfile

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/7883812/zwave_js.log)

AlCalzone, 2022-01-17:
I'm not sure what's happening and unfortunately the log doesn't have enough context.
It is possible that your stick is simply stuck, which can be resolved with a soft-reset (if enabled) or by re-plugging it.

BebeMischa, 2022-01-18:
Ok thanks anyway. 
Yesterday I did those steps, more times, even restarting the HA server, physically unplugging from power, nothing helped.

Today is did update the HA core to 21.12.10 and after that I can include the device again.
So I suspect, that there was something with the core and those updates fixed that:

2021.12.10
Repository: home-assistant/core · Tag: 2021.12.10 · Commit: 7887f23 · Released by: balloob
•	Avoid removing zwave_js devices for non-ready nodes (@kpine - #59964) (zwave_js docs)
•	Ignore unavailable entities when creating zwave_js device actions list (@raman325 - #64184) (zwave_js docs)

So it looks, problem was not on your side ;-)

Thank you for your time.

kpine, 2022-01-18:
Those HA changes wouldn't have any effect on how the device is included or interviewed by the driver. Sounds like a coincidence.

BebeMischa, 2022-01-18:
Then it will stay a mystery forever :-)

I hope it stays working now again for at least next two years as it did before. I like the Z-wave reliability so much ;-)

BebeMischa, 2022-01-19:
Reason unknown, but problem solved. Closing issue now. Thanks everyone.

closed 2022-01-19 by BebeMischa