zwave-js/zwave-js #4101: Investigate extracting firmware from Leviton updaters

AlCalzone, 2022-01-17:
The updates are available from [Leviton here](https://hubconnecteddecorasmart.zendesk.com/hc/en-us/articles/4407521680155-Can-Leviton-Z-Wave-Products-Firmware-Be-Updated-) 

_Originally posted by @ChrisLizon in https://github.com/zwave-js/node-zwave-js/discussions/4087_

kaijk, 2022-01-17:
Leviton will provide links to the OTA FW files themselves. Contact [support@decorasmartsupport.zendesk.com](url) 

Here are the links they sent me in late November for the then current DZ15S (v1.22) and DZ6HD (v1.23) FW files.  Note, these were more recent versions than those provided in the then current Aeotec executable...  I have installed them both with no issues using zwjs2mqtt.

DZ6HD: [https://decorasmartsupport.zendesk.com/attachments/token/MFT4C61yHdtG43RDv3QkVHANg/?name=DZ6HD_v1_22.ota](url)
DZ15S: [https://decorasmartsupport.zendesk.com/attachments/token/laPYbWOXZISNjsA7Vw8sEESCw/?name=DZ15S_v1_23.ota](url)

AlCalzone, 2023-03-08:
See https://github.com/zwave-js/node-zwave-js/discussions/4087#discussioncomment-3972827 for details - not sure if still necessary.