zwave-js/zwave-js #4102: Eaton RF9542 sends Basic Set commands instead of Multilevel Switch Reports from physical switch presses

rahi6, 2022-01-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

No updates from physical press of the switch. Possibly closely related to #4072 

What do you observe?

Device sending Basic Set commands, not Multilevel Switch Reports. Basic v1 is updated instead of Multilevel Switch v2.

What did you expect to happen?

Multilevel Switch v2 gets updated with physical switch presses.

2022-01-18T03:19:18.943Z SERIAL « 0x0116004984021004120226277586707185772b2c727387a2                  (24 bytes)
2022-01-18T03:19:18.946Z SERIAL » [ACK]                                                                   (0x06)
2022-01-18T03:19:18.948Z DRIVER « [Node 002] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x021004120226277586707185772b2c727387
2022-01-18T03:19:18.950Z CNTRLR « [Node 002] Received updated node info
2022-01-18T03:19:18.957Z SERIAL « 0x01090004000203200163b1                                            (11 bytes)
2022-01-18T03:19:18.958Z SERIAL » [ACK]                                                                   (0x06)
2022-01-18T03:19:18.960Z DRIVER « [Node 002] [REQ] [ApplicationCommand]
                                  └─[BasicCCSet]
                                      target value: 99
2022-01-18T03:19:18.962Z CNTRLR   [Node 002] treating BasicCC::Set as a report
2022-01-18T03:19:18.963Z CNTRLR   [Node 002] [~] [Basic] currentValue: 0 => 99                      [Endpoint 0]

### Device information

Manufacturer:  Eaton
Model name: RF9542
Node ID in your network: <2>

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 6.4.0
`zwavejs2mqtt` branch: 8.11.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I have tried adding:

	"compat": {
		"enableBasicSetMapping": true
	}

to the rf9542 config file as a test (learning from #4072 ) but it did not work.

### Did this work before?

Yes (please describe)

### If yes, where did it work?

openHAB 3.3 (probably unrelated to zwave js)

### Attach Driver Logfile

Driver log with the node being interviewed:
[zwavejs_2022-01-17.log](https://github.com/zwave-js/node-zwave-js/files/7886032/zwavejs_2022-01-17.log)

AlCalzone, 2022-01-18:
> 2022-01-18T03:15:43.312Z CNTRLR   [Node 002] Embedded device config loaded

It didn't seem to pick up the updated config. Did you put your file into the custom config dir and restart z2m?

rahi6, 2022-01-18:
Hey, that was the log when using the default device config file.

See log file below with the node being interviewed with the custom config file.

[zwavejs_2022-01-17_test.log](https://github.com/zwave-js/node-zwave-js/files/7891866/zwavejs_2022-01-17_test.log)

This is the custom config file I tried where I added the compat line from the RF9540 fix. I had to change this to a txt file from json, so it would upload to this message.

[rf9542-z.txt](https://github.com/zwave-js/node-zwave-js/files/7891874/rf9542-z.txt)

This is the debug log lines when manually pressing the RF9542 with the custom config (same behaviour as without the compat line) .

2022-01-18T03:32:51.528Z SERIAL « 0x0116004984021004120226277586707185772b2c727387a2                  (24 bytes)
2022-01-18T03:32:51.531Z SERIAL » [ACK]                                                                   (0x06)
2022-01-18T03:32:51.533Z DRIVER « [Node 002] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x021004120226277586707185772b2c727387
2022-01-18T03:32:51.536Z CNTRLR « [Node 002] Received updated node info
2022-01-18T03:32:51.540Z SERIAL « 0x01090004000203200163b1                                            (11 bytes)
2022-01-18T03:32:51.542Z SERIAL » [ACK]                                                                   (0x06)
2022-01-18T03:32:51.543Z DRIVER « [Node 002] [REQ] [ApplicationCommand]
                                  └─[BasicCCSet]
                                      target value: 99
2022-01-18T03:32:51.545Z CNTRLR   [Node 002] treating BasicCC::Set as a report
2022-01-18T03:32:51.546Z CNTRLR   [Node 002] [~] [Basic] currentValue: 0 => 99                      [Endpoint 0]

AlCalzone, 2022-01-18:
Ok the mapping target is defined by the device class. This one is using one where we don't have a mapping yet (Multilevel Remote Switch).

rahi6, 2022-01-18:
thanks for the info, much appreciated! hopefully the mapping can be added

PR #4382 (2022-03-16): feat: map Basic CC Set to appropriate CCs for Remote Switch devices
fixes: #4102
closed 2022-03-16 by AlCalzone