zwave-js/zwave-js #4116: Firmware Update for Leviton DZ6HD not working [question]

MrSerious0, 2022-01-21:
Hi,

I'm trying to update my Leviton DZ6HD devices from v1.13 to v1.20 using the files supplied by Leviton here: https://hubconnecteddecorasmart.zendesk.com/hc/en-us/articles/4407521680155-Can-Leviton-Z-Wave-Products-Firmware-Be-Updated-
... specifically here: https://filetransfer.leviton.com/link/9LrvU6A5pfP01P9D0LDinj

When selecting DZ6HD_01_20_ota__TargetZwave__.exe, I receive this in the zwavejs2mqtt logs: 

```
_2022-01-21 10:00:15.156 INFO ZWAVE: Calling api beginFirmwareUpdate with args: [
30,
'DZ6HD_01_20_ota__TargetZwave__.exe',
<Buffer 4d 5a 90 00 03 00 00 00 04 00 00 00 ff ff 00 00 b8 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... 719584 more bytes>,
0,
[length]: 4
]
2022-01-21 10:00:15.179 INFO ZWAVE: Unable to extract firmware from file: This does not appear to be a valid Aeotec updater (invalid firmware name)! (ZW0322) beginFirmwareUpdate undefined_
```

Do you have any suggestions?

AlCalzone, 2022-01-21:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly` and attach it here as a file.
While making the log, attempt the update again.

MrSerious0, 2022-01-21:
Here is the requested driver log.  Thanks for your help!

2022-01-21 11:27:38.806 is where I try to attempt the update using the .exe file.  

022-01-21 11:27:49.191 is where I try to attempt the update using the .hex file. 

[zwavejs2mqtt_2022-01-21.log](https://github.com/zwave-js/node-zwave-js/files/7915224/zwavejs2mqtt_2022-01-21.log)

robertsLando, 2022-01-21:
That log is zwavejs2mqtt log, please upload a driver log

MrSerious0, 2022-01-21:
Sorry! Here is the driver log (I believe)
[zwavejs_2022-01-21.log](https://github.com/zwave-js/node-zwave-js/files/7915327/zwavejs_2022-01-21.log)

MrSerious0, 2022-01-22:
Not sure how I missed it earlier, but the firmware (v1.22) linked in issue #4101 worked perfectly.  I was able to use zwavejs2mqtt to update 5 instances of DZ6HD switches using this firmware: https://decorasmartsupport.zendesk.com/attachments/token/MFT4C61yHdtG43RDv3QkVHANg/?name=DZ6HD_v1_22.ota

AlCalzone, 2022-01-23:
> This does not appear to be a valid Aeotec updater (invalid firmware name)! 

The exe file seems to have a format that's a bit different from the Aeotec updater. I'll look into it, probably just a small change.

malmaxx, 2024-11-29:
I'm running into this as well trying to update Leviton devices.  Been a couple years since I last successfully updated Leviton devices so this is a new issue for me.  Any timeline of when this .exe issue will be fixed?

malmaxx, 2024-12-08:
I came across a newer Leviton support article that discusses changes made in 2022 to HA and the OTA process.  This support article provides .OTA files versus the .ZIP file containing .EXE files found in a different Leviton support article that previously worked with the Z-Wave JS OTA process.  I just successfully updated two different devices via both the Z-Wave JS and native HA OTA update interfaces using the .OTA files.

**_Leviton Support - New Home Assistant OTA Instructions:_**
https://hubconnecteddecorasmart.leviton.com/hc/en-us/articles/7151993826075-Updating-Decora-Smart-Z-Wave-Using-Home-Assistant

**_Leviton Support - ZIP file download containing .EXE update files_**
https://hubconnecteddecorasmart.leviton.com/hc/en-us/articles/4407521680155-Can-Leviton-Z-Wave-Products-Firmware-Be-Updated