zwave-js/zwave-js #4118: Yale Connexis Battery Life reduced significantly

jb123567, 2022-01-21:
I have noticed over the past month my Zwave Yale lock is using a set of batteries weekly, whereas it was using a set every 6 months.

The lock is the only device on my network.

Running the latest version on Home Assistant OS using a UZB.

AlCalzone, 2022-01-23:
I'd start by looking at some driver logs and node statistics:
- How often does the lock report on its own?
- How often does the controller send commands to the lock?
- Are there many dropped commands or repeated requests by the lock?
- Anything else that stands out?

jb123567, 2022-01-23:
> I'd start by looking at some driver logs and node statistics:
> 
> - How often does the lock report on its own?
> 
> - How often does the controller send commands to the lock?
> 
> - Are there many dropped commands or repeated requests by the lock?
> 
> - Anything else that stands out?

Thanks for your reply, changed the batteries again last night and already lost 8%. I've enabled "silly" level logging in HA and will have a look through see how often the device is communicating.

Is there a normal you would expect to see? Apologies Zwave is something I know a huge amount about!

AlCalzone, 2022-01-23:
You can post some in a day or two and let me take a look. Just make sure to erase user codes if you set some.

jb123567, 2022-01-24:
> You can post some in a day or two and let me take a look. Just make sure to erase user codes if you set some.
[zwavejs2mqtt_2022-01-23.log](https://github.com/zwave-js/node-zwave-js/files/7929084/zwavejs2mqtt_2022-01-23.log)
[zwavejs2mqtt_2022-01-24.log](https://github.com/zwave-js/node-zwave-js/files/7929085/zwavejs2mqtt_2022-01-24.log)
[zwavejs_2022-01-23.log](https://github.com/zwave-js/node-zwave-js/files/7929086/zwavejs_2022-01-23.log)
[zwavejs_2022-01-24.log](https://github.com/zwave-js/node-zwave-js/files/7929094/zwavejs_2022-01-24.log)

Files attached, hopefully this helps!

AlCalzone, 2022-01-25:
Honestly, there's nothing that stands out. The communication looks fine and there are no excessive messages being received by the controller.

Maybe the lock does try to send more often than we can see. Just to be sure, check out and follow https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues

jb123567, 2022-01-25:
> Honestly, there's nothing that stands out. The communication looks fine and there are no excessive messages being received by the controller.
>
>
>
> Maybe the lock does try to send more often than we [...]

That's frustrating, bizarre that all of a sudden the lock starts using so much battery. I'll try and troubleshoot a bit further with the hardware!

Thank you very much for looking.

closed 2022-02-12 by zwave-js-assistant[bot]