zwave-js/zwave-js #4119: Missing device configuration: Zooz ZEN52

calvolson, 2022-01-22:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Zooz ZEN52

### Manufacturer ID

0x027a

### Product Type

0x0104

### Product ID

0x0202

### Firmware Version

1.11

### Do you have a manual?

[https://cdn.shopify.com/s/files/1/0218/7704/files/zooz-700-series-z-wave-plus-double-relay-zen52-manual.pdf](https://cdn.shopify.com/s/files/1/0218/7704/files/zooz-700-series-z-wave-plus-double-relay-zen52-manual.pdf)

zetlen, 2022-02-03:
I have _almost_ everything I need to contribute this:
- [x] [manual](https://cdn.shopify.com/s/files/1/0218/7704/files/zooz-700-series-z-wave-plus-double-relay-zen52-manual.pdf)
- [x] [mfg's parameter documentation](https://www.support.getzooz.com/kb/article/805-zen52-double-relay-advanced-settings/)
- [ ] [`zwaveAllianceId`](https://products.z-wavealliance.org/Search/Index?regionId=-1&searchText=zen52)

It looks like configs want a `zwaveAllianceId` in the members of the `devices` list. [It doesn't look like one is _required_](https://zwave-js.github.io/node-zwave-js/#/config-files/file-format?id=devices), but as soon as it's available, you'd probably want it updated. So what should we do if the device hasn't appeared in the Z-wave Product Catalog yet?

AlCalzone, 2022-02-04:
If you don't have that ID, its fine. It means we can't easily create a starter config file for manual cleanup though and it has to be written from scratch.

zetlen, 2022-02-04:
Understood. I'm having trouble getting a custom config file to load, though. I've replicated `templates/master-template.json` and `templates/zooz-template.json` in my `store/config` directory, but the logs show no sign of zwave2mqtt finding it.

[image]

Don't wanna commandeer this ticket to discuss this unless it's really quick and obvious. I can clone the repo and step debug `deviceConfigPriorityDir` if need be.

AlCalzone, 2022-02-06:
Just move the zen52 file out of the subdirectory or put the zooz template into a `templates` subfolder of 0x027a. I assume it imports `"templates/zooz_template.json"` which is relative to the current location.

zetlen, 2022-02-07:
Did as you suggested and still can't get it to go, sorry. When I have time this evening I'll step debug it.

AlCalzone, 2022-02-08:
Try restarting zwavejs2mqtt if you haven't already. Also when zwave-js starts up, you should see an error in the driver logfile relatively at the start if there is a problem loading the file.

PR #4262 (2022-04-28): feat(config): add config for Zooz ZEN51 and ZEN52 Smart Relays
Closes #4119.
Fixes: https://github.com/zwave-js/node-zwave-js/issues/4393

- Adds common parameter data for Zooz's new relays to Zooz manufacturer template.
- Adds config file for [ZEN52 Double Relay](https://www.thesmartesthouse.com/products/zooz-700-series-z-wave-plus-double-relay-zen52)
- Adds config file for [ZEN51 Dry Contact Relay](https://www.thesmartesthouse.com/products/zooz-700-series-z-wave-plus-dry-contact-relay-zen51) as well, since it has so many common parameters with the ZEN52.

### notes
- Manuals are linked direc [... cut]
closed 2022-04-28 by AlCalzone