zwave-js/zwave-js #4122: Entire network re-interviewed on restart, S2 secure devices ignore commands until re-joined (intermittent)

jordahlm, 2022-01-23:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Specific root cause unknown, but appears to manifest on process restart.

What do you observe?
Main noticeable symptom is S2 secure nodes ignore all commands. Upon examining the logs, it appears to start with "Restoring the network from cache failed: SyntaxError: Unexpected end of JSON input"
What follows is all nodes are re-interviewed with the appearance of success, but all nodes that use S2 security as opposed to no security will not respond to any commands until a full reset/re-inclusion of the node is performed.

What did you expect to happen?
Network to remain functional when process is restarted.

Steps to reproduce the behavior:
  1. Process restarts. (This time due to cron job that issues "snap restart zwavejs2mqtt" every other day at 3:00 am local time, previous times likely on automatic snap refresh. Cron job was put in place due to lingering stability issues that seem to be fixed upon restart.
  2. Notice nodes that have S2 security fail to respond to any z-wave commands and only respond to physical switch inputs until re-set and re-included. I only have devices with no security and S2 security. All devices were added with the default security options, so devices aquired since S2 was supported ended up with S2 authenticated security. In order to avoid future headaches I rejoined the nodes this time explicitly with no security so that if this happens again it will be largely transparent.
  3. Notice that battery powered nodes need to be woken up to complete their interview before working. I have a battery powered scene controller and a door open/close sensor that would not work at all until manually woken up.
 

### Device information

Manufacturer: Zooz
Model name: Zen 71, Zen 34, XS 41, XS 42
Node ID in your network: 21, 22, 23, 24, 25, 26, 27, 33

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

snap from https://snapcraft.io/zwavejs2mqtt
version: 6.3.1, 6.4.1 was upgraded to during troubleshooting
`node-zwave-js` branch: 8.10.1, 8.11.1
`zwavejs2mqtt` branch: 6.3.1, 6.4.1

This same issue has occurred in the past but I did not troubleshoot as extensively then. I believe it was around 6.1 or 6.2 where I had the same issue with S2 devices not functioning, and in 5.9 where I had no secure devices, but the interview process for the nodes had difficulty completing.

### Did you change anything?

no

### If yes, what did you change?

n/a

### Did this work before?

Yes (please describe)

### If yes, where did it work?

This issue only happens sporadically. The snap restarts on its own periodically when there is a refreshed version, and lately I have implemented a cron job to restart it every other day due to some observed stability issues. I suspect those issues may be due to the 700 series controller meta issue but this issue is different from that.

### Attach Driver Logfile

[zwavejs_2022-01-21.log](https://github.com/zwave-js/node-zwave-js/files/7919772/zwavejs_2022-01-21.log)
 I apologize this is at log level info and not debug. As this isn't readily reproducible I don't have logs running in debug all the time.

AlCalzone, 2022-01-23:
@jmgiaever I think this needs your help. This is a case of the `<homeid>.json` file getting corrupted because of an improper shutdown.

jmgiaever, 2022-01-23:
Hi, I'm actually suspecting that the main issue here is the 700 stick problem. 

The last time it happened to me that the cache got corrupted, I spent many days getting it up running again. 

Many devices didn't interview properly and some not at all. I had to wait for the network to be "interviewed" and then interviewed one by one - and sometimes several times as entities configurations were still missing. 

I also expect the 700 issue to be the reason for instabilities, and not the snap package. 

Having a crontab to restart it is not a good idea. 

E.g can happen at the same time as a snap refresh and possibly corrupt the cache. 

I would recommend doing as I did; convert back to a 500 stick. It's a difference between day and night. 

There's a backup and restore option in Z2M under the advanced options.

jmgiaever, 2022-01-23:
PS! You can copy the cache from the previous revisions. It's located in the directory

```
/var/snap/zwavejs2mqtt/<rev>
```

The subdirectory `current` within this directory is most likely pointing to the latest/highest revision, that you're on. Try copying the cache from the lower revisions into this directory. 

Stop zwavejs2mqtt first.

AlCalzone, 2022-01-23:
@jmgiaever it definitly is a corrupted cache file:
>  "Restoring the network from cache failed: SyntaxError: Unexpected end of JSON input"

jmgiaever, 2022-01-23:
Yes, but using a crontab to restart the app doesn't sound like a good solution to fix his 700 instabilities. 

If the snap is refreshed at the same time, e.g closing down or just starting up, it can cause the cache to crash?

The snap doesn't touch the cache, except that it's copied as a backup between revisions. But that is synced, so a failed backup where content differs would lead to a failed refresh and roll back.

jordahlm, 2022-01-23:
I have disabled the crontab from your feedback. I had some issues that would be resolved by a restart so had put that in place a couple weeks ago as a band-aid. Crontab was simply:
`0 3 */2 * * /usr/bin/snap restart zwavejs2mqtt`

I later discovered the 700 series controller issue and have a 500 series stick on order.

Both of the controller sticks are the Zooz ZST-10, the 700 series currently running and a 500 on order that I'll be migrating to. Let me know if there's anything else I can do to help troubleshoot and contribute. 

Thanks for hopping on this so fast and thanks for developing such a wonderful framework for the home automation community!

AlCalzone, 2022-03-04:
In the upcoming v9 release, the `.json` cache file will be changed to an appendonly `.jsonl` file like the other two. I'm expecting this to fix your problems.

closed 2022-03-04 by AlCalzone

jordahlm, 2022-03-13:
Appreciate the update. For what it's worth, I haven't noticed a recurrence of this issue since updating to the 7.17.1 firmware on my ZST10 700 stick. However, it's possible it has happened and I didn't notice it as all devices were re-interviewed successfully. As part of the troubleshooting of this issue I purposely left all my devices with no security.