zwave-js/zwave-js #4126: NZW31S: Missing Config Parameters

JonLaliberte, 2022-01-24:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Config file in zwave-js: https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0312/nzw31s.json

Just has some missing config options.

Right now the parameters available in the zwave-js repo are 1,2,4,5. There are also 3,8,9. (Options 6 and 7 seem to deal with Association Groups, so I ignored them.)

3: LED Indicator Control (whether the led on the switch is on when the light is on or off)
8: Default value when the switch is turned on (at the switch)
9: Default value when the switch is turned on (via zwave)

Reference: https://github.com/InovelliUSA/OpenZWave/blob/master/config/inovelli/nzw31.xml

### Device information

Manufacturer: INovelli
Model name: NZW31S
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

NA

blhoward2, 2022-01-31:
These parameters aren't in the manual, including references to the manual posted by Inovelli on their forum. How sure are you that these actually exist? Do they exist at every firmware version? Can you contact Inovelli and try to find out?

JonLaliberte, 2022-01-31:
8 and 9 were added on Firmware 1.0.2 (must be 2-3 years ago at least). I was using them with OZW in HA previously (still using them now, just manually).

[inovelli-nzw31.pdf](https://github.com/zwave-js/node-zwave-js/files/7974345/inovelli-nzw31.pdf)

JonLaliberte, 2023-07-31:
Just wanted to note this as well: https://github.com/OpenZWave/open-zwave/blob/master/config/inovelli/nzw31.xml