zwave-js/zwave-js #4130: [bug] Node health check leads to infinite `PowerlevelCCTestNodeSet`/`PowerlevelCCTestNodeGet` loop

LordMike, 2022-01-25:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

### What causes the bug?
Initiating a health check from ZWJS2MQTT, between node 32 and Node 1 (controller). I did get a response on most of the checks I did, but I was testing, so I didn't observe if all my tests led to UI responses.

What do you observe?
I was troubleshooting other issues, when I noticed the logs were very large. Turns out this Set/Get loop is still running today with over 138k messages so far.

What did you expect to happen?
No infinite loops.

Steps to reproduce the behavior:
1. Use Zwavejs2mqtt new `test` image with health test between nodes & controller.

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [X] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: `8.11.2`
`zwavejs2mqtt` branch: `test`

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Updated to `test` to check out healthchecks

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

I have found the following. The node in question that still has Set/Gets is 32:

2022-01-24T21:05:46.329Z: First PowerlevelCCTestNodeSet to Node 32

2022-01-24T21:06:51.958Z: First PowerlevelCCTestNodeGet to Node 32, that repeats every 1s

[zwavejs_2022-01-24.zip](https://github.com/zwave-js/node-zwave-js/files/7936642/zwavejs_2022-01-24.zip)

LordMike, 2022-01-25:
I just rebooted zwavejs2mqtt, to see if I could reproduce it. I noticed that node 32 is my most-far-away node, which [I had issues performing a test on](https://github.com/zwave-js/zwavejs2mqtt/pull/2205#issuecomment-1021023616) - which seems coincidental :).

Doing a new lifeline health check yields this:

```
2022-01-25T19:48:37.461Z CNTRLR   [Node 032] Starting lifeline health check (1 round)...
2022-01-25T19:48:37.461Z CNTRLR » [Node 032] requesting node neighbors...
2022-01-25T19:48:37.469Z SERIAL » 0x010700802001000059                                                 (9 bytes)
2022-01-25T19:48:37.469Z DRIVER » [REQ] [GetRoutingInfo]
                                    remove non-repeaters: true
                                    remove bad links:     false
2022-01-25T19:48:37.471Z SERIAL « [ACK]                                                                   (0x06)
2022-01-25T19:48:37.474Z SERIAL « 0x01200180240000600000020000000000000000000000000000000000000000000 (34 bytes)
                                  018
2022-01-25T19:48:37.474Z SERIAL » [ACK]                                                                   (0x06)
2022-01-25T19:48:37.475Z DRIVER « [RES] [GetRoutingInfo]
                                    node ids: 3, 6, 30, 31, 50
2022-01-25T19:48:37.476Z CNTRLR « [Node 032] node neighbors received: 3, 6, 30, 31, 50
2022-01-25T19:48:37.481Z SERIAL » 0x010800132001002513f3                                              (10 bytes)
2022-01-25T19:48:37.481Z DRIVER » [Node 032] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      19
                                  └─[NoOperationCC]
2022-01-25T19:48:37.482Z SERIAL « [ACK]                                                                   (0x06)
2022-01-25T19:48:37.489Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-01-25T19:48:37.489Z SERIAL » [ACK]                                                                   (0x06)
2022-01-25T19:48:37.489Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-01-25T19:48:37.571Z SERIAL « 0x0107001313000008f0                                                 (9 bytes)
2022-01-25T19:48:37.572Z SERIAL » [ACK]                                                                   (0x06)
2022-01-25T19:48:37.573Z DRIVER « [REQ] [SendData]
                                    callback id:     19
                                    transmit status: OK
2022-01-25T19:48:37.585Z SERIAL » 0x010800132001002514f4                                              (10 bytes)
2022-01-25T19:48:37.585Z DRIVER » [Node 032] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      20
                                  └─[NoOperationCC]
2022-01-25T19:48:37.587Z SERIAL « [ACK]                                                                   (0x06)
2022-01-25T19:48:37.593Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-01-25T19:48:37.593Z SERIAL » [ACK]                                                                   (0x06)
2022-01-25T19:48:37.593Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-01-25T19:48:37.674Z SERIAL « 0x0107001314000009f6                                                 (9 bytes)
2022-01-25T19:48:37.675Z SERIAL » [ACK]                                                                   (0x06)
2022-01-25T19:48:37.676Z DRIVER « [REQ] [SendData]
                                    callback id:     20
                                    transmit status: OK
2022-01-25T19:48:37.684Z SERIAL » 0x010800132001002515f5                                              (10 bytes)
2022-01-25T19:48:37.685Z DRIVER » [Node 032] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      21
                                  └─[NoOperationCC]
2022-01-25T19:48:37.686Z SERIAL « [ACK]                                                                   (0x06)
2022-01-25T19:48:37.692Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-01-25T19:48:37.693Z SERIAL » [ACK]                                                                   (0x06)
2022-01-25T19:48:37.693Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-01-25T19:48:37.774Z SERIAL « 0x0107001315000009f7                                                 (9 bytes)
2022-01-25T19:48:37.775Z SERIAL » [ACK]                                                                   (0x06)
2022-01-25T19:48:37.776Z DRIVER « [REQ] [SendData]
                                    callback id:     21
                                    transmit status: OK
2022-01-25T19:48:37.786Z SERIAL » 0x010800132001002516f6                                              (10 bytes)
2022-01-25T19:48:37.786Z DRIVER » [Node 032] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      22
                                  └─[NoOperationCC]
2022-01-25T19:48:37.787Z SERIAL « [ACK]                                                                   (0x06)
2022-01-25T19:48:37.793Z SERIAL « 0x0104011301e8                                                       (6 bytes)
[... 227 lines cut ...]
[... cut ...]
2022-01-25T19:48:47.198Z SERIAL « 0x010c0004002006730601020000a7                                      (14 bytes)
2022-01-25T19:48:47.199Z SERIAL » [ACK]                                                                   (0x06)
2022-01-25T19:48:47.200Z DRIVER « [Node 032] [REQ] [ApplicationCommand]
                                  └─[PowerlevelCCTestNodeReport]
                                      test node id:        1
                                      status:              In Progress
                                      acknowledged frames: 0

(repeats)
```

AlCalzone, 2022-01-25:
Like I commented in the other issue
```
2022-01-25T19:48:47.200Z DRIVER « [Node 032] [REQ] [ApplicationCommand]
                                  └─[PowerlevelCCTestNodeReport]
                                      test node id:        1
                                      status:              In Progress
                                      acknowledged frames: 0
```
The node is instructed to ping the controller but it never makes any progress and seems to stay at 0 acknowledged frames. The driver expects it to either make some progress or report failure at some point. We should have a safeguard for longer periods of no change.

AlCalzone, 2022-01-25:
I'm going to close this and track it as part of #4075

closed 2022-01-25 by AlCalzone