zwave-js/zwave-js #4132: Missing Reports - Thermostats and Intermittently Schlage Lock

1activegeek, 2022-01-26:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Any updates that should be reported to the console - i.e. change in current room temperature on the thermostat. Additionally in the case of the door locks, the updates seem to be ok when first starting, but will eventually erode it seems. 2 of the exact same thermostat exhibiting the exact same behavior as well, so even less likely a fluke in one of the pairings.

What do you observe?
I can see that no update is made to the thermostat temperatures. When checking Home Assistant, I can see a long history of the temperature never having changed. If I poll or refresh the device using the ZWaveJS2MQTT interface, the value will update almost instantly. I have also validated that ZWaveJS2MQTT does not show an updated value either, so it is assumed ZWJS is not actually receiving these updates at all.

What did you expect to happen?
I expect to see temperature updates be populated in the console after they change. Instant would be nice, but even within a few minutes would be an improvement over never changing.

Steps to reproduce the behavior:
No real steps, as mentioned its a matter of actual missing information being reported - I can't manually reproduce anything. Even trying to get the temp to change and monitor the logs on ZWaveJS is difficult to try and tackle and I've not been able to catch it lately as the temp generally stays relatively the same. (Thanks efficient heating/cooling systems!)

I've also tried the troubleshooting steps for reviewing the logs, I've attempted to re-interview the node, and I've restarted the system many a times. The door lock will usually come back when rebooting, then fade away over time again. Not as concerned with the lock at the moment as it's a fairly unimportant lock - but the thermostats are key for driving certain automations. 

**Something else of note** - not sure if it's just my lack of knowing all the intricate details of these devices and what makes up the configs - but the node JSON has iterations of a "Scene Actuator Configuration" Command Class that I don't see in the spec of the device and has property names of "dimmingDuration" with numbers up to 255. It almost seems like it's being assigned a litany of class features for each individual brightness level of a dimming bulb, which I found quite odd. 

### Device information

Manufacturer: Trane
Model name: XL 624 Thermostat
Node ID in your network: 
Node ID 10 and 37 (there are 2 thermostats, both exhibiting the exact same issue)

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

HomeAssistant OS: core-2021.12.9 / supervisor-2021.12.2
ZWaveJS version: 8.10.2
ZWaveJS2mqtt: 6.3.1
`node-zwave-js` branch:
`zwavejs2mqtt` branch:
[image]

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

These thermostats worked under the old ZWave legacy integration. I had a system that was running that was a bit outdated, and while I did update the HASS version within the last 6 months, I was still running the old integration and hadn't converted over yet. I then decided to upgrade the Stick to the Aeotec Gen5 from Gan4, and move to a new HASS instance and start a bit fresh. So a lot has changed from when it was working, but these didn't require anything special on the old configuration as they don't even use Security. 

### Attach Driver Logfile

Example of Node10 sending a temperature change command successfully. They always seem to work with receiving commands. But I believe I've also noticed that the temperature will stay wrong, even when sending that command as well. 

[Node10 Output for Change Temp.log](https://github.com/zwave-js/node-zwave-js/files/7938731/Node10.Output.for.Change.Temp.log)

Output of the Node Debug Info for Node10: (please change back to JSON extension, GH wouldn't allow JSON)
[node_10.txt](https://github.com/zwave-js/node-zwave-js/files/7938737/node_10.txt)

1activegeek, 2022-01-26:
Thank you in advance to the folks who might be able to help me isolate the issue here. I apologize if there is something blatant I've missed, but I'm happy to try and get any additional details that can help. I wish there were more useful logs to go off, but I can certainly dump any other info you need from the ZWave Configs to the HASS configs, etc.

AlCalzone, 2022-01-26:
There's a difference communication when the controller initiates it (querying values) vs. when the node initiates it (unsolicited reports). In controller-initiated communication, the controller knows a route to the node and the node will reply using that route. In node-initiated communication, the node will use its last known route(s), which may change over time if previous attempts fail.

To me this sounds like the connectivity between the controller and nodes isn't the best and the nodes are forgetting the working routes over time because of intermittent errors. I'm not sure if you've already followed this part of the guide https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting - but if you haven't do that first, so we're not chasing ghosts.

1activegeek, 2022-01-27:
@AlCalzone Thank you, I totally understand and that makes sense. Short version of a background - I'm quite technical in nature and work as a sales engineer, I just don't know ZWave to that level - but if you give me snippets of explanation, I can totally comprehend and interpret pretty easily. All that to say - what you said makes total sense, and I found a way to validate as well that the device is not able to send the reports back. I made manual change at the thermostat of temperature and then again of mode - none of which were observed in the logs in realtime, and used a control of a light switch which did in fact send its messages in that same time period. 

Relative to the troubleshooting, while I can certainly try to move the Stick further - as mentioned this WAS working on an older setup with the older legacy ZWave integration. The placement of my stick is in a USB extension outside of the server rack to the highest point in the basement above the rack - aka away from metal, away from electronics, and away from USB3 devices. I have tried waiting, I have restarted the machine (HASS on a Pi4) - I have not tried unplugging and re-plugging the stick, but I don't see a difference between that and power cycle anyway. Not sure if the soft reset will do anything else in the stick itself that would be worth trying, but I can certainly do that just in case. 

I do know this should not be a 700 series device (Aeotec Gen5 Stick) and it is not security enabled, so those should not be issues relative to the communication. Additionally, all other nodes in the home it seems, are working ok (we can ignore the lock for now as I haven't fully isolated that as an issue with ZWaveJS) and communicate fine.

I've thought about the idea of unpairing and then re-pairing as a final troubleshooting option - not specifically called out, but attributes to the 1st rule of computer troubleshooting in general of restarting it fresh. I just hate the atomic option in this scenario being the troubleshooting step as it means having to muck with HASS configs as well to make it all right after it's done. But if you believe that is a better step before taking other actions that might be available - by all means let me know and I'll do it. This one is mildly important overall to my automation needs, so I really do want to fix this and/or share whatever could be going wrong here as necessary.

AlCalzone, 2022-01-27:
Let's get one thing out of the way:

> as mentioned this WAS working on an older setup with the older legacy ZWave integration

I read this often when communication issues are involved. It is simply impossible that swapping the host software, which simply tells the stick to do things, has any influence on the physical connection from other devices to that stick. 

Under OZW you likely didn't notice because it used to actively poll devices for their status to a much larger extent. Z-Wave JS only does in some very specific situations and definitely not regularly like OZW did. This means even if the devices didn't "call in" their status, they were asked for it.

---

> I've thought about the idea of unpairing and then re-pairing as a final troubleshooting option

Not sure where you got that - re-pairing is only necessary when you've lost the security keys for the device. Or for some device configuration changes that update capabilities.

The whole issue reads like the controller location is generally fine, especially if it is limited to these specific devices. Have you tried healing the network yet like the troubleshooting guide mentions? Maybe they "just" need some good routes to work with.

Otherwise, zwavejs2mqtt is getting tools to measure connection quality [soon](https://github.com/zwave-js/zwavejs2mqtt/pull/2205). Might also be worth a try when they are available.

1activegeek, 2022-01-27:
> I read this often when communication issues are involved. It is simply impossible that swapping the host software, which simply tells the stick to do things, has any influence on the physical connection from other devices to that stick.

That makes sense to me, I just wasn't sure if it's possible there were maybe different driver functions/capabilities that were/n't being enabled under the hood. I figured there shouldn't be (like you said, the hardware is what is actually connecting/driving the communication with the device) but wanted to make the comparison to ensure I provide the best detail.

--- 

> Not sure where you got that - re-pairing is only necessary when you've lost the security keys for the device. Or for some device configuration changes that update capabilities.

Just me thinking of all options to try and "reset" myself to basics and ensure I didn't mix anything up or have some communication issue like the locks. I know I had one lock with that issue initially. Had to re-pair it like 3 times until I figured out what needed to happen to secure include right.

> The whole issue reads like the controller location is generally fine, especially if it is limited to these specific devices. Have you tried healing the network yet like the troubleshooting guide mentions? Maybe they "just" need some good routes to work with.

Healing was my first choice. I figured with 2 devices recently disabled and failed (outside plugs for xmas lights) I would try healing the network so I could be sure none of them were using those plugs as part of their path. I'll try it again, just for giggles to ensure something funky didn't happen last time.

> Otherwise, zwavejs2mqtt is getting tools to measure connection quality [soon](https://github.com/zwave-js/zwavejs2mqtt/pull/2205). Might also be worth a try when they are available.

Ok, I'll be sure to look out for em. Is it something I might be able to find beta or early code to compile on a separate device to test? I also still have my old stick, that I could try using in all this for any extraneous testing. Wouldn't be fun having to unpair then re-pair to a different stick - but I can certainly do that if needed to run any diagnostic or testing needed. 

AND PS - thank you for the thoughtful replies and effort in helping me sort this - I know the OSS Automation community is vast with a lot of specialty, but thankfully they're all quite helpful.

robertsLando, 2022-01-28:
@1activegeek If you want to give that a try use `test` tag on docker instead of `latest`

1activegeek, 2022-01-28:
Sounds good @robertsLando. Do you know offhand if that's easily possible using HassOS? If not, I'll look at creating a VM dummy To test it or I'll see about enabling the local access on HassOS and doing it manually. I just know host access is a complicated setup for HassOS.

robertsLando, 2022-01-28:
@1activegeek I have no clue as I don't use hass but I think it could be possible using portainer someway

1activegeek, 2022-01-28:
I'll see what I can do over the weekend - may be a bit more difficult than anticipated - since I'm using js2mqtt, they bundle the ZWJS inside of their container I believe - so I can't directly alter to use a different container. I'll have to disconnect, re-adjust to use straight ZWJS, then try adjusting the container version.

Looks like it's doable, just a bunch of steps, and need to make sure I make proper backups first! 😊

1activegeek, 2022-02-13:
So I didn't get to try and run the tests using the new health functions using the `latest` tag, since it wasn't as easy or straightforward figuring out which versions were matching up proper relative to HassOS plugins versions, JS2mqtt version, and zwave-js version. That said, it looks like in that timeframe there was already a merge into master, and the latest version I updated to looks to have the health option.

So I ran the test on both thermostats, though it does look like the health always checks THROUGH another node in the network. I can't just click the specific thermostat, and then test that thermostat - I have to select one device (not my target) then choose my target and it will test. So in my case I've got ~40 devices that I could test that connection - not sure I understand why I can't just run a health check from the stick? Maybe I just don't know enough about what it's doing.

All this to say - it all comes out healthy (10/10 pings and low interference dB levels) on all 5 checks each time. I've manually forced a heal 2 or 3 times since then. And I'm still stuck with no reports coming in from the thermostats. I have noticed that these devices DONT have the lifeline configuration that I know some devices do. I wish I had a sample of what their config was on my old setup to compare.

Thoughts?

1activegeek, 2022-02-13:
Is it possible I need to look at using this option: `SERVICE ZWAVE_JS.REFRESH_VALUE` for these devices? I believe I do need to turn this on for a few of my dimmer switches that did use something like this previously as a fix for dimmer switches that would immediately report their state as it started to turn off as being something like 67% vs 1100% (on) and would change to off from the command, but quickly change back to a "dimmed" state of 67 where it caught it during the dim to off (HomeSeer dimmer switches). Is it just possible these thermostats are missing something that will require this also?

robertsLando, 2022-02-14:
@1activegeek about the health check question in order to check the health connection of a node with the controlle simply select the required node in mesh, press check health and select the controller node as target

About your other question it seems more something related to HASS integration and zwavejs, maybe @blhoward2 @AlCalzone could help

AlCalzone, 2022-02-14:
> I have noticed that these devices DONT have the lifeline configuration that I know some devices do. I wish I had a sample of what their config was on my old setup to compare.

The device manual should describe this. If the lifeline association is missing it could be that the devices are not Z-Wave+ and we don't have the information in the config file. Or there was a problem/timeout during the interview, so the lifeline couldn't be set up correctly.

1activegeek, 2022-02-16:
Thanks @robertsLando for the clarification, think I get it now. Makes sense. I ran these checks again with that insight, and it looks ok overall. I've attached the 2 reports for the devices in question. 

@AlCalzone according to the ZWaveAlliance data, I don't believe it does actually have the lifeline support. It is definitely not ZWave+ I know that, but I don't see anything about lifeline on the ZWA spec page: https://products.z-wavealliance.org/products/837?selectedFrequencyId=2 - due to the lack of any Association details where it is ordinarily documented. Sorry if that was just a useless effort to think about, I just remember lifeline being a useful thing long ago when I had battled an issue about auto-updating devices.

So back to no good leads at the moment. Should I look at trying to utilize the Refresh Value option (https://www.home-assistant.io/integrations/zwave_js/#service-zwave_jsrefresh_value) to try and at least get some sort of polling to happen more frequently? 

Alternatively - is there a pointer as to what I would need to find in the old ZWave integration code that maybe I can go diving for to find what might be set differently? As I said the only real lead I have at this moment is that it did in fact work on the OLD ZWave integration (not OpenZWave, previous to that one) successfully. Happy to go code diving if it would help.

closed 2022-03-06 by zwave-js-assistant[bot]

1activegeek, 2022-03-06:
Is there a way in particular I can re-open this issue? There never was a solution, but I'm not sure what I can do next for troubleshooting.