zwave-js/zwave-js #4138: [Bug]:

candrea77, 2022-01-28:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Zwavejs2Mqtt version

6.4.1

### ZwaveJS version

8.11.1

### Describe the bug

Unable to get the battery report from Fibaro FGK101

### To Reproduce

Include a Fibaro FGK101, set Group 3 association with controller

### Expected behavior

Getting battery report once a day

### Additional context

Try clear all association and put them back ... nothing changed.
Problem happening on version 2.5 and 3.2

AlCalzone, 2022-01-28:
You only mentioned which nodes **aren't affected**, which makes it hard to look for potential issues.

There are several repeated commands from nodes which makes me believe the mesh might not be healthy. Try https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting.

If that doesn't help, you should ask the Fibaro support. They should be able to tell you how to configure their devices so that they report properly.

candrea77, 2022-01-28:
Ok,
I've understood that it is quite impossible to investigate this problem.
By my point of view, mesh have not big issue : command are sent and
received immediately by any device

Thanks anyway for trying help me

Regards

Crippa Andrea

closed 2022-02-15 by zwave-js-assistant[bot]