zwave-js/zwave-js #4139: WT00Z5-1 Switch Missing

prairieit, 2022-01-29:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Including the device

What do you observe?
Once including, the device does not have a light. or a switch. service to control its on/off state. It has a diagnostic Node Status only.

What did you expect to happen?
The device should have a light. service to control its on/off state

Steps to reproduce the behavior:
  1. Include device
  2. Controllable light does not exist.
  3. Re-interview
  4. Controllable light still does not exist.

### Device information

Manufacturer: Linear (Nortek Security Control LLC)
Model name: WT00Z5-1
Node ID in your network: 48

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Driver Version: 8.10.2
Server Version: 1.14.1

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

The device previously worked flawlessly on a Vera (non HA) device.

### Attach Driver Logfile

log here

AlCalzone, 2022-01-29:
> ### Attach Driver Logfile
> 
> log here

We ask for the log for a reason. Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly`. Re-interview the device and when done, attach the log here as a file.

prairieit, 2022-01-30:
I dont use zwavejs2mqtt, just the Z-Wave JS HA add-on.... are these the same logs?
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/7965315/zwave_js.log)

prairieit, 2022-01-31:
@AlCalzone was that the proper log file?

AlCalzone, 2022-02-01:
Yes. But from the information there, the device is a wall controller (controlling other devices) and not a switch:
```
[Node 048] received response for protocol info:
basic device class:    Routing Slave
generic device class:  Wall Controller
specific device class: Basic Wall Controller
node type:             Routing End Node

...

supported CCs:
· Z-Wave Plus Info
· Central Scene
· Version
· Manufacturer Specific
· Powerlevel
· Device Reset Locally
· Association
· Association Group Information
· Configuration
· Firmware Update Meta Data
controlled CCs:
· Central Scene
```

It also doesn't have any endpoints or `Manufacturer Proprietary CC`, which could expose additional functionality.

The [OpenZWave config file](https://github.com/OpenZWave/open-zwave/blob/master/config/linear/WT00Z5-1.xml) also doesn't add any CCs to the list, so maybe the device isn't telling everything.

Try excluding, factory resetting and then including again.