zwave-js/zwave-js #4141: Missing device configuration: AIBase

prairieit, 2022-01-29:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

HA-ZW-5SABC

### Manufacturer ID

0x0384

### Product Type

0x0200

### Product ID

0x0009

### Firmware Version

2.1

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/2943?selectedFrequencyId=2

### Do you have a manual?

https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/2943/AIBASE%20Z-Wave%20Multi-Sensor%20UserGuide%2020180606.pdf

AlCalzone, 2022-01-29:
@zwave-js-bot import config 2943 from zwa

AlCalzone, 2022-01-29:
The bot has created a PR but the file only serves as a starter template. It needs to be cleaned up to conform with our style guide, see above.

For anyone who wants to help out:
- fork the repo if you haven't already
- edit branch `import-config-1643451012` in your fork
- create a pull request with your branch `import-config-1643451012` as the source and our branch `import-config-1643451012` as the target

PR #4143 (2022-06-22): feat(config): add AIBase HA-ZW-5SABC [DRAFT]
fixes: #4141

**TODO:**
- [x]	Change PR title to be more specific
closed 2022-06-22 by AlCalzone