zwave-js/zwave-js #4148: Fantem/Oomi Multisensor 6 (Aeotec ZW100 clone) Not Recognized - No Config Parameters

drumrboy44, 2022-01-31:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I'm hoping this is a relatively easy fix based on this description. Essentially, I have what I believe are several Oomi/Fantem Multisensor 6's (Fantem FT100 in the ZwaveJS db), which is a clone of the Aeon/Aeotec Multisensor 6 (ZW100 in the ZwaveJS db). I say "what I believe" because there is no branding on the boxes I received. I am unable to adjust the configuration parameters in Zwavejs2mqtt because the device is coming up as Unknown Device, though it identifies the device as a Fantem Multilevel Sensor.

From reviewing the debug notes, it looks like the Product ID/Product Code combination (which are 0x0102: 0x0064) are not listed as a device in the FT100 file, and I'm assuming that is why zwavejs2mqtt lists the device as an unknown product. 

I looked at the JSON config files for the FT100 device, and it looks like perhaps the only solution needed is to add this new product ID/ Product code to the FT100 db file, but I'm obviously not an expert. 

### Device information

Manufacturer: Fantem/Oomi (0x016a)
Model name: FT100 (0x0102)
Node ID in your network:  (2, 4, 11, 12).

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [X] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 
`node-zwave-js` branch:  6.4.1
`zwavejs2mqtt` branch: 8.11.1

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

N/A

AlCalzone, 2022-01-31:
@zwave-js-bot add fingerprint 0x0102:0x0064 to 0x016a:0x0002:0x0064

closed 2022-01-31 by AlCalzone

drumrboy44, 2022-01-31:
Damn, you guys work fast, thank you!

PR #4149 (2022-01-31): fix(config): add fingerprint `0x0102:0x0064` to "Fantem FT100"
fixes: #4148